qmk/qmk_firmware #17077: [Bug] Anne Pro 2 Firmware fails to build

morphiak, 2022-05-13:
## Describe the Bug
Building Anne Pro2 firmware fails with 
...
contrib/os/hal/ports/HT32/HT32F523x2/hal_lld.c: In function 'ht32_clock_init':
lib/chibios-contrib/os/hal/ports/HT32/HT32F523x2/hal_lld.c:56:5: error: this 'while' clause does not guard... [-Werror=misleading-indentation]
   56 |     while (PWRCU->BAKTEST != 0x27);
...
cc1: all warnings being treated as errors
...

Resolved by editing line 56 to  
while (PWRCU->BAKTEST != 0x27){};

filterpaper, 2022-05-18:
Both keyboards compiled file on default:
```c
Linking: .build/annepro2_c15_default.elf                                                            [OK]
Creating binary load file for flashing: .build/annepro2_c15_default.bin                             [OK]
Creating load file for flashing: .build/annepro2_c15_default.hex                                    [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  33652	      0	  33652	   8374	annepro2_c15_default.bin

Copying annepro2_c15_default.bin to qmk_firmware folder                                             [OK]
(Firmware size check does not yet support cortex-m0plus; skipping)
```
```c
Linking: .build/annepro2_c18_default.elf                                                            [OK]
Creating binary load file for flashing: .build/annepro2_c18_default.bin                             [OK]
Creating load file for flashing: .build/annepro2_c18_default.hex                                    [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  33728	      0	  33728	   83c0	annepro2_c18_default.bin

Copying annepro2_c18_default.bin to qmk_firmware folder                                             [OK]
(Firmware size check does not yet support cortex-m0plus; skipping)
```

Did you try with a fresh clone of `qmk_firmware` and confirmed that your git sub-modules are up-to-date with `make git-submodules`?

azisaka, 2022-06-12:
I had the same problem with the open anne pro fork, old code.

closed 2022-08-29 by tzarc