qmk/qmk_firmware #17079: [Bug] Dactyl Manuform 5x6: Malfunctioning in latest QMK Firmware 0.16.4

Vasilis, 2022-05-14:
I am comparing QMK Firmware versions **0.13.38** vs. **0.16.4** on the [Ergohaven Dactyl Manuform 5x6](https://ergohaven.xyz/keyboards/tproduct/339856256-557749895691-dactyl-manuform-5x6).

**In the former version everything works fine**. In latest QMK version, the secondary part of this split keyboard will not work. I couldn't find any major changes in `handwired/dactyl_manuform/5x6` since 0.13.38 to explain the failure in 0.16.4... :/ 
Anyone can pinpoint what code change may have caused the failing behavior?

**Issue**
No matter what I have tried I cannot make both left and right sides of my Ergohaven 5x6 manuform work. Whichever side I connect, it operates as the master and the other side does not type anything. I am testing with the [default](https://github.com/qmk/qmk_firmware/tree/master/keyboards/handwired/dactyl_manuform/5x6/keymaps/default) keymap.
I have tried `MASTER_LEFT`, `EE_HANDS` (with :avrdude-split-left and :avrdude-split-right), resetting EEPROM with default image, ... but no luck.

precondition, 2022-05-15:
Have you tried flashing the new firmware to the slave half? Version 0.14 changed how the split transport protocol works so a master half with version >=0.14 won't be compatible with a slave half with a version <0.14 (https://docs.qmk.fm/#/ChangeLog/20210828?id=split-transport-improvements)

Vasilis, 2022-05-15:
Yeah, I always flash both halves either same firmware.

brunooomelo, 2022-05-16:
and now, what can we do to make it work?

harikulendran, 2022-06-21:
I see that the ergohaven manuforms have type C out, this could mean they're using quikpro micro C's which have a slightly different voltage output than standard pro micro clones which I found causes issues with detecting the master side of the keyboard. Adding the following definitions in the config solves this issue for me, hopefully can be of help to you too:
```
#define SPLIT_USB_DETECT
#define SPLIT_USB_TIMEOUT 2500
```