qmk/qmk_firmware #17115: RP2040 PR branch support

tzarc, 2022-05-16:
Placeholder for issues/discussion with the RP2040 branch.

Jpe230, 2022-05-17:
Meta-question: Would it be viable to open a discord channel specific to RP2040 to enable a more informal discussion?

freznel10, 2022-05-17:
I'm encountering some primary/secondary issues. Have setup a split hand pin and I know it works since the OLEDs show what's supposed to be displayed on the right and left sides. Same goes with the split pointers--they behave according to what I've setup for the right and left pointers no matter which side I connect. But no matter what I do, when I connect to left split, the matrix is inverted.

doesntfazer, 2022-05-18:
> > When trying to compile a via(l) keymap I created, I am getting this error. I turned off basically everything in rules.mk. I am aware that there isn't an EEPROM or EEPROM emulation yet, but I thought it meant that it wouldn't persist, but you could still build it.
> > Compiling: quantum/dynamic_keymap.c quantum/dynamic_keymap.c:88:1: error: static assertion failed: "Dynamic keymaps are configured to use more EEPROM than is available."
> > 88 | _Static_assert((DYNAMIC_KEYMAP_EEPROM_MAX_ADDR) - (DYNAMIC_KEYMAP_MACRO_EEPROM_ADDR) >= 100, "Dynamic keymaps are configured to use more EEPROM than is available.");
> > | ^~~~~~~~~~~~~~
> > [ERRORS]
> > |
> > |
> > |
> > make[1]: *** [builddefs/common_rules.mk:452: .build/obj_crowboard_via/quantum/dynamic_keymap.o] Error 1
> > make: *** [Makefile:418: crowboard:via] Error 1
> > Make finished with errors
> 
> Try increasing `TRANSIENT_EEPROM_SIZE` and enable the fake EEPROM driver if you haven't: `EEPROM_DRIVER = transient`

That ended up working. Thank you.

jaygreco, 2022-05-30:
I'm seeing some issues with a custom RP2040 board that uses [this](https://datasheet.lcsc.com/lcsc/1912111437_BOYAMICRO-BY25Q32BSTIG-T_C394868.pdf) 4MB flash chip. I can flash the rp2040 oneky UF2 but it doesn't boot to the USB stack, whereas I can get the board I have with a Winbond  25Q128JVSQ to boot no problem with the same UF2. The BY25Q32BSTIG board works using the Ardiono RP2040 with the following settings, so I know it's okay hardware-wise:
[image]

BTW, beyond stoked to retire the legacy Bit-C in favor of an RP2040 flavor 🙏

KarlK90, 2022-05-31:
> Out of curiosity, does anyone know why the `generic_03h` and `w25x10cl` bootloader options aren't present in the QMK bootcode?

They are now with the latest commits to the rp2040 branch just select one of the defines, the fallback is always a `W25Q080` flash chip.

```c
#define RP2040_FLASH_AT25SF128A
#define RP2040_FLASH_GD25Q64CS
#define RP2040_FLASH_W25X10CL
#define RP2040_FLASH_GENERIC_03H
#define RP2040_FLASH_IS25LP080
```

masiullah, 2022-06-20:
I took the most recent code for compiling the firmware and getting the below error 

_Compiling: quantum/keymap_introspection.c                                                          quantum/keymap_introspection.c: In function 'keymap_layer_count':
quantum/keymap_introspection.c:9:45: error: 'keymaps' undeclared (first use in this function)
    9 | #define NUM_KEYMAP_LAYERS ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))_

I can think it might be due to the  change https://github.com/qmk/qmk_firmware/pull/17229

When I remove the code at line 398 in build_keyboard.mk then it works. Is there something I am missing that I need to do to make this work without build_keyboard changes. I am using userspace keymaps and overriding it in the keymap using USER_NAME := 

line 398 in build_keyboard.mk
$(QUANTUM_DIR)/keymap_introspection.c \

YCF, 2022-07-13:
How to make the ADC working?
I'm adding the joystick to my keyboard.

tzarc, 2022-07-13:
> How to make the ADC working? I'm adding the joystick to my keyboard.

Unsupported on RP2040 at the moment.

YCF, 2022-07-13:
> > How to make the ADC working? I'm adding the joystick to my keyboard.
> 
> Unsupported on RP2040 at the moment.

ETA？

KarlK90, 2022-07-13:
No ETA.

zvecr, 2022-08-28:
Supported merged in 2022q3

closed 2022-08-28 by zvecr