qmk/qmk_firmware #17128: [Bug] Xcode Command Line Tools 13.4 breaks make for qmk compile

jackmcconnell, 2022-05-17:
## Describe the Bug

The latest version of Xcode Command Line Tools is 13.4, released on 16/05/2022.
`qmk compile` worked without issue prior to this. After upgrading from 13.3.1 to 13.4 via System Preferences, `qmk compile` fails. 
Downloading and re-installing 13.3.1 to downgrade allows `qmk compile` to work as expected again.

## System Information

**Keyboard:** 
GMMK Pro ISO

**Operating system:** 
macOS Monteray 12.4 (Also fails with Xcode Command Line Tools is 13.4 on macOS Monteray 12.3)

**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /Users/Jack/qmk_firmware
Ψ Detected macOS 12.4.
Ψ Git branch: master
Ψ Repo version: 0.16.9
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

zvecr, 2022-05-18:
```
% pkgutil --pkg-info=com.apple.pkg.CLTools_Executables | grep version

version: 13.4.0.0.1.1651278267
```

qmk doctor output:
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /Users/zvecr/qmk_firmware
Ψ Detected macOS 12.4.
Ψ Git branch: master
Ψ Repo version: 0.16.9
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

builds fine for me
```
Compiling: platforms/chibios/syscall-fallbacks.c                                                    [OK]
Compiling: platforms/chibios/wait.c                                                                 [OK]
Linking: .build/gmmk_pro_iso_default.elf                                                            [OK]
Creating binary load file for flashing: .build/gmmk_pro_iso_default.bin                             [OK]
Creating load file for flashing: .build/gmmk_pro_iso_default.hex                                    [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  33604	      0	  33604	   8344	gmmk_pro_iso_default.bin

Copying gmmk_pro_iso_default.bin to qmk_firmware folder                                             [OK]
(Firmware size check does not yet support STM32F303; skipping)
```

jackmcconnell, 2022-05-18:
Sure. 
So i've tested this directly before and after upgrading. The result is always the same, even after completely removing qmk from Homebrew and deleted the qmk_firmware folder and then re-installed. 

The version of Xcode Command Line Tools that allows `qmk compile` to run without errors is 13.3.1.0.1.1648687083.

Updating to version 13.4.0.0.1.1651278267 from System Preferences > Software Update in macOS Monteray 12.4 causes `qmk compile` to fail with the following errors: 

`
~/qmk_firmware (master*) » sudo qmk compile -kb gmmk/pro/iso -km default       
Ψ Compiling keymap with gmake --jobs=1 gmmk/pro/iso:default

Making gmmk/pro/iso with keymap default

⚠ "git describe --abbrev=6 --dirty --always --tags" returned error code 128
⚠ "git describe --abbrev=6 --dirty --always --tags" returned error code 128
⚠ "git describe --abbrev=6 --dirty --always --tags" returned error code 128
builddefs/build_keyboard.mk:49: *** multiple target patterns.  Stop.
gmake: *** [Makefile:413: gmmk/pro/iso:default] Error 1
Make finished with errors
`

zvecr, 2022-05-18:
> "git describe --abbrev=6 --dirty --always --tags" returned error code 128

This usually happens when the folder it is inspecting is not a valid git repo. The following make errors are due to this initial failure, rather than an issue with make itself.

If i had to guess, the use of `sudo` is picking up a different qmk_firmware folder or trashing the environment.

sigprof, 2022-05-18:
Or, more likely, Xcode 13.4 contains a Git version with a fix for [CVE-2022-24765](https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2022-24765): https://support.apple.com/en-us/HT213261

The updated Git version refuses to work with a repository that is owned by a different user; this includes the `sudo` case, when Git is running as root and tries to access the repository owned by the normal user. There is the [safe.directory](https://git-scm.com/docs/git-config/2.35.2#Documentation/git-config.txt-safedirectory) config option that can be used to work around that error (note that the option must be specified in the system or global config file to take effect).

fauxpark, 2022-05-18:
Shouldn't the Brew-installed git take precedence?

jackmcconnell, 2022-05-18:
@fauxpark I've used Git for a long time prior to using qmk, whick I only installed 2 weeks ago. I would assume the version I originally installed would be used. I can't see Homebrew installing it's own version of git if there was already a version on the system. I think git on my system was installed as part of Xcode Command Line tools.  

@sigprof This sounds like the most plausible explanation. Especially as it works perfectly with or without sudo before updating to Xcode Command Line Tools 13.4 and doesn't work after, and I can replicate it on demand.

@zvecr Just tested this but it outputs the correct QMK_HOME and QMK_FIRMWARE folders before and after, whether using `sudo` or not.

fauxpark, 2022-05-18:
I could have sworn the `qmk` formula depended on git, but I guess not. Perhaps it would be a good idea for it to?

Either way, you should not really need to use `sudo` to compile QMK; in fact I would say it's discouraged precisely due to the increased chance of issues like this.

jackmcconnell, 2022-05-18:
@sigprof Have tried adding: 
```
[safe]
	directory = /Users/Jack/qmk_firmware
```
to .gitconfig but the error is still the same when running Xcode Command Line Tools 13.4. 

If I run `make gmmk/pro/iso:default VERBOSE=true` using 13.4, I see the following, which may help clarify the error a bit: 
```
QMK Firmware 0.16.9
# Check if we have the CMP tool installed
cmp .//Makefile .//Makefile >/dev/null 2>&1; if [ $? -gt 0 ]; then printf "\033[31;01mError:\033[0m\033[1m cmp command not found, please install diffutils\n\033[0m"; exit 1; fi;
# Ensure that qmk works.
if ! qmk hello 1> /dev/null 2>&1; then printf "\033[31;01mERROR:\033[0m Cannot run \"qmk hello\"!\n\n Please run \033[1mqmk setup\033[0m to install all the dependencies QMK requires.\n\n"; exit 1; fi
# Check if the submodules are dirty, and display a warning if they are
if [ ! -e lib/chibios ]; then git submodule sync lib/chibios && git submodule update --depth 50 --init lib/chibios; fi
if [ ! -e lib/chibios-contrib ]; then git submodule sync lib/chibios-contrib && git submodule update --depth 50 --init lib/chibios-contrib; fi
if [ ! -e lib/lufa ]; then git submodule sync lib/lufa && git submodule update --depth 50 --init lib/lufa; fi
if [ ! -e lib/vusb ]; then git submodule sync lib/vusb && git submodule update --depth 50 --init lib/vusb; fi
if [ ! -e lib/printf ]; then git submodule sync lib/printf && git submodule update --depth 50 --init lib/printf; fi
git submodule status --recursive 2>/dev/null | \
	while IFS= read -r x; do \
		case "$x" in \
			\ *) ;; \
			*) printf "\033[33;01mWARNING:\033[0m Some git submodules are out of date or modified.\n Please consider running \033[1mmake git-submodule\033[0m.\n\n";break;; \
		esac \
	done
rm -f .build/error_occurred > /dev/null 2>&1
make: *** [gmmk/pro/iso:default] Error 1

```

jackmcconnell, 2022-05-18:
@sigprof Thanks. Have just tried that but it gives the same error. As mentioned, I've completely wiped and re-installed qmk, including re-cloning the qmk_firmware directory so I'm not sure it's nessecerily a permissions issue. Again, this all worked perfectly before Xcode Command Line Tools 13.4.

jackmcconnell, 2022-05-18:
Huzzah! That worked. Running `make gmmk/pro/iso:default` compiled correctly. So why was `sudo` breaking the build?

sigprof, 2022-05-18:
There are multiple ways for unneeded `sudo` to break things:

1) The program running under `sudo` may not get the same environment (e.g., `$PATH` may be completely different), so it might see different versions of some tools, or complain about missing tools.

2) When the build process is run with root privileges, the files and directories that are created during the build will be owned by root, and then running a build as the normal user may run into permission errors because of those leftover files. (This was apparently your problem — the previous invocation of `sudo make ...` created the `.build` directory as root, and then the make process running as the normal user could not remove the `.build/error_occurred` file.)

3) With that CVE-2022-24765 fix Git won't be able to access the repository owned by the normal user when running under `sudo`. If you want to allow that access, you would need to run something like `sudo git config --global --add safe.directory /path/to/repo` (in this particular case `sudo` is needed, because the config setting needs to be done to the root's global Git config).

jackmcconnell, 2022-05-19:
Thanks for clarifying - that's really useful. Sounds like it was a combination of me using `sudo` and the CVE-2022-24765 fix introduced in Xcode Command Line Tools 13.4.

closed 2022-05-19 by jackmcconnell