qmk/qmk_firmware #17154: Regarding some questions about bluetooth dual-mode

EasonQian1, 2022-05-19:
I want to make a dual-mode keyboard using ATmega32u4 and nRF51822 using the spi interface scheme. How do I switch modes and set up bluetooth search? Is there an example for reference.

fauxpark, 2022-05-19:
You use the output select keycodes to switch between auto, Bluetooth and USB. As for searching, that's handled by the module itself.

EasonQian1, 2022-05-19:
<html>
<body>

OUT_AUTO | Automatically switch between USB and Bluetooth
-- | --
OUT_USB | USB only
OUT_BT | Bluetooth only

</body>
</html>
ok i found this, thanks a lot. Regarding the Bluetooth module, take MDBT40 as an example to connect another device in the connected state. Do I need to debug a separate instruction for this module?

EasonQian1, 2022-05-19:
Ok, I see. Thanks for your patience

EasonQian1, 2022-05-21:
@fauxpark There is one more question. Regarding this automatic hibernation, does it also depend on the module?

fauxpark, 2022-05-21:
I don't know what you mean.

EasonQian1, 2022-05-21:
> I don't know what you mean.

What I mean is that the keyboard goes to sleep for a certain period of time without motion. How should this be set?

fauxpark, 2022-05-21:
I don't think there is any support for that.

EasonQian1, 2022-05-21:
> I don't think there is any support for that.

Ok, I see. Thanks for your patience.

closed 2022-05-21 by EasonQian1