qmk/qmk_firmware #17170: [Bug] AnnePro2 LED Profile Set isn't working as expected.

riazufila, 2022-05-20:
## Describe the Bug

`ap2_led_set_profile(7);` doesn't set the LED profile. Changing the argument from 7 to 1 or 6 doesn't change the keyboard's default LED profile.

## System Information

**Keyboard: AnnePro2** 
**Revision (if applicable): C18** 
**Operating system: Arch Linux** 
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /home/riazufila/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: riazufila-keymap
Ψ Repo version: 0.16.9
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.1.0
Ψ Found avr-gcc version 12.1.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

Jpe230, 2022-05-24:
Are you building it using the custom rgb matrix?

riazufila, 2022-05-24:
@Jpe230 Yes I did.

```
# Custom RGB matrix handling
RGB_MATRIX_ENABLE = yes
RGB_MATRIX_DRIVER = custom
```

Jpe230, 2022-05-24:
> @Jpe230 Yes I did.
> 
> 
> 
> ```
> 
> # Custom RGB matrix handling
> 
> RGB_MATRIX_ENABLE = yes
> 
> RGB_MATRIX_DRIVER = custom
> 
> ```

That explains it! the profiles are meant to be used without the custom RGB matrix.

The custom RGB matrix uses QMK animations and it is overriding the profiles defined in Shine.

Try erasing both lines, it should allow it to use the Shine profiles :)

riazufila, 2022-05-25:
@Jpe230 Thanks! Erasing both the lines did allow setting up default LED profiles.

However, now I cant seem to make ```KC_AP_RGB_MOD, KC_AP_RGB_TOG, KC_AP_RGB_VAD, and KC_AP_RGB_VAI``` to function properly at all. I cant cycle though LED profile, toggle the LED, or set LED brightness anymore as I used to.

closed 2022-08-29 by tzarc