qmk/qmk_firmware #17197: [Feature Request] add a new bootloader support of stm32f103?

zhaqian12, 2022-05-23:
Add a new bootloader option of stm32f103.(UF2)

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
Stm32duino is the default bootloader for QMK STM32F103,but because it would wait before entering the app, so its experience was not good.I found an open source bootloader project [uf2-stm32f103](https://github.com/zhaqian12/uf2-stm32f103).After testing, I think it works well so far.It takes up 16 kB of flash.So,is it necessary to add an option for this bootloader?If yes,maybe I can make a contribution.

drashna, 2022-10-07:
Most likely, this would need a new ld script for the controller, and probably a new board folder, for the different config. 

You can look at the blackpill uf2 support to see how to add that, I think.

zvecr, 2023-01-15:
#19594

PR #19594 (2023-01-15): Add mmoskal/uf2-stm32f103 bootloader support

## Description

Seems a bunch of ymdk keyboards are now using [mmoskal/uf2-stm32f103](https://github.com/mmoskal/uf2-stm32f103#switching-to-the-bootloader) but by specifying (and duplicating) a ton of crap at the keyboard level.

Releases have filename as `uf2boot-BLUEPILL.bin` but QMKs name for the bootloader could be changed to more match the repo name?

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Document [... cut]
closed 2023-01-15 by zvecr