qmk/qmk_firmware #17201: [Bug] ErgoDox EZ fails to compile

ifohancroft, 2022-05-24:
## Describe the Bug

ErgoDox EZ fails to compile.

I made sure that I am using the latest upstream and that I have tried the default keymap as well. The output I'm pasting is for my keymap but I did a `vimdiff` on the errors and compiling the default map gives the exact same errors.

## System Information

**Keyboard:** Falba.Tech ErgoDox
**Revision (if applicable):** 
**Operating system:** Manjaro Linux 21.2.6
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /home/ifohancroft/Projects/Keyboards/Firmware/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/ifohancroft/Projects/Keyboards/Firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/ifohancroft/Projects/Keyboards/Firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32' boards. Run 'sudo cp /home/ifohancroft/Projects/Keyboards/Firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/ifohancroft/Projects/Keyboards/Firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/ifohancroft/Projects/Keyboards/Firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'massdrop' boards. Run 'sudo cp /home/ifohancroft/Projects/Keyboards/Firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/ifohancroft/Projects/Keyboards/Firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/ifohancroft/Projects/Keyboards/Firmware/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: ifohancroft-compile-here
Ψ Repo version: 0.16.9
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.1.0
Ψ Found avr-gcc version 12.1.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

## Additional Context

Here's the error log:

```
Ψ Compiling keymap with make --jobs=1 ergodox_ez:ifohancroft

QMK Firmware 0.16.9
Making ergodox_ez with keymap ifohancroft

avr-gcc (GCC) 12.1.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  19032	      0	  19032	   4a58	ergodox_ez_ifohancroft.hex

Compiling: keyboards/ergodox_ez/matrix.c                                                           In file included from platforms/gpio.h:21,
                 from quantum/quantum.h:46,
                 from keyboards/ergodox_ez/ergodox_ez.h:23,
                 from keyboards/ergodox_ez/matrix.c:35:
keyboards/ergodox_ez/matrix.c: In function 'init_cols':
platforms/avr/gpio.h:26:49: error: array subscript 0 is outside array bounds of 'volatile uint8_t[0]' {aka 'volatile unsigned char[]'} [-Werror=array-bounds]
   26 | #define setPinInputHigh(pin) (DDRx_ADDRESS(pin) &= ~_BV((pin)&0xF), PORTx_ADDRESS(pin) |= _BV((pin)&0xF))
      |                                                 ^~
keyboards/ergodox_ez/matrix.c:133:5: note: in expansion of macro 'setPinInputHigh'
  133 |     setPinInputHigh(F0);
      |     ^~~~~~~~~~~~~~~
platforms/avr/gpio.h:26:49: error: array subscript 0 is outside array bounds of 'volatile uint8_t[0]' {aka 'volatile unsigned char[]'} [-Werror=array-bounds]
[... 155 lines cut ...]
  211 |                 writePinLow(B0);
      |                 ^~~~~~~~~~~
In file included from /usr/avr/include/avr/io.h:99,
                 from platforms/avr/pin_defs.h:18,
                 from platforms/pin_defs.h:22,
                 from quantum/config_common.h:20,
                 from ./keyboards/ergodox_ez/config.h:23,
                 from <command-line>:
In function 'read_cols',
    inlined from 'store_raw_matrix_row' at keyboards/ergodox_ez/matrix.c:73:32:
keyboards/ergodox_ez/matrix.c:164:19: error: array subscript 0 is outside array bounds of 'volatile uint8_t[0]' {aka 'volatile unsigned char[]'} [-Werror=array-bounds]
  164 |         return ~((PINF & 0x03) | ((PINF & 0xF0) >> 2));
      |                   ^~~~
keyboards/ergodox_ez/matrix.c:164:36: error: array subscript 0 is outside array bounds of 'volatile uint8_t[0]' {aka 'volatile unsigned char[]'} [-Werror=array-bounds]
  164 |         return ~((PINF & 0x03) | ((PINF & 0xF0) >> 2));
      |                                    ^~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:456: .build/obj_ergodox_ez_ifohancroft/matrix.o] Error 1
Make finished with errors
make: *** [Makefile:413: ergodox_ez:ifohancroft] Error 1
```

zvecr, 2022-05-24:
<https://github.com/qmk/qmk_firmware/issues/17064>

closed 2022-05-24 by zvecr

ifohancroft, 2022-05-24:
Sorry, I totally missed that.