qmk/qmk_firmware #17206: [Bug] Split Keyboard Off-Hand does not work when SPLIT_POINTING_ENABLE and VIA _ENABLE=yes

foureight84, 2022-05-24:
It seems that the off-hand half does not respond when enabling VIA with SPLIT_POINTING_ENABLE.

## Describe the Bug

The keyboard in question is the Sofle V1 with an Elite-C on the left and Promicro on the right. The right side also has a Pimoroni Trackball attached. When enabling VIA support, plugging the USB cable into the left side results in the right side becoming unresponsive. I can see the active light on the Pro-micro but no key presses, no trackball movement, and OLED is off. 

Plugging the USB cable into the right side results in both halves working properly.

With everything in the code remaining the same, removing `SPLIT_POINTING_ENABLE` from `rules.mk` while enabling VIA support results in both halves being responsive. Of course, the trackball does not work when the USB cable is plugged into the left side as data transport is not enabled.

Firmware configurations are as followed:
EE_HANDS
SPLIT_POINTING_ENABLE
POINTING_DEVICE_RIGHT
POINTING_DEVICE_ROTATION_90
POINTING_DEVICE_TASK_THROTTLE_MS 8
VIA_ENABLE = yes

QMK fork being referenced:
https://github.com/foureight84/qmk_firmware/tree/sofle_foureight84_v2/keyboards/sofle/keymaps/foureight84 (VIA is currently not enabled due bug mentioned above)

## System Information

**Keyboard:** Sofle
**Revision (if applicable):** V1
**Operating system:** Windows 11
**`qmk doctor` output:** 

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

thinkJD, 2022-05-29:
Hello, 
thanks for your great work on this Sofle mod! 

I upgraded my Sofle with your trackball mod. My Keyboard ist mostly the same except for the controllers. I use two pro micros. 

I patched my keymap with your code but the trackball is not working at all. Anyways, this could be a unrelated problem. But i have the same Issue with the unresponsive right side. 

For me it‘s independent from VIA support. If SPLIT_POINTING is enabled and the right side is plugged in both keyboard sides working correctly. If I plug the left side in the right side is unresponsive. 

If I disable SPLIT_POINTING it works wherever I plug the USB cable. 

I tried different EE_HANDS settings without luck.

Later I flashed your keymap without modifications but it has the same effect. No working trackball and works only if the right side is plugged in.

foureight84, 2022-05-29:
@thinkJD 

Do you have 
```
#define POINTING_DEVICE_ROTATION_90 // depends on your trackball orientation, this is set according to my post.
#define SPLIT_POINTING_ENABLE
#define POINTING_DEVICE_RIGHT
```

I would double check your config.h and rules.mk. But currently, I am having issues with this myself on the latest master branch although I have talked to Dasky on discord and he's not having issues when compiling against my fork. I'm still in the process of debugging this but I corrupted my only promicro's bootloader. I'm waiting for a new one to arrive to try.

defined in your config.h?

foureight84, 2022-05-29:
I want to quickly follow up on this:
I haven't had any luck restoring my pro micro with the Atmel Caterina bootloader. By that, I am able to flash QMK on the board but the trackball never works. Maybe I've set the fuse incorrectly?

But currently, I've burned the atmel-dfu (https://github.com/qmk/qmk_firmware/blob/master/util/bootloader_atmega32u4_1.0.0.hex) and set the following fuse bits:
`-U lfuse:w:0x5E:m -U hfuse:w:0x99:m -U efuse:w:0xC3:m`
Which now has both sides working properly with the trackball. However, the same issue persists, if VIA is enabled the rightt side is non responsive (no oled, keypresses, and trackball).

daskygit, 2022-05-29:
Worked through this a little on Discord, it appears the bug only occurs on a old branch, rebasing to 0.17.0 solved the issue.

After running a reverse bisect this commit seems to have fixed it.  7712a286dccea029785976311433cf8673594f6f

foureight84, 2022-05-30:
Confirmed work as per @daskygit comment. Thank you!

@thinkJD Please feel free to open a ticket on my repo instead if you are having issues with the new push I've made. If there's an issue, it's most likely on my end.

closed 2022-05-30 by foureight84