qmk/qmk_firmware #17217: [Feature Request] Improve retro shift usability

jose-elias-alvarez, 2022-05-27:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

First, I apologize if any information here is incorrect - I'm still very new to QMK and am still wrapping my head around its features.

I've been trying the `RETRO_SHIFT` feature to combine `AUTO_SHIFT` with home row mods. I love the idea of it and am excited about the possibility of moving Shift to a less-used key (or eliminating it altogether) but I'm having a hard time getting used to the quirks described in the documentation:

> Tap Hold Configurations work a little differently when using Retro Shift. Referencing TAPPING_TERM makes little sense, as holding longer would result in shifting one of the keys.
> 
> IGNORE_MOD_TAP_INTERRUPT changes only rolling from a mod tap (releasing it first), sending both keys instead of the modifier on the second. Its effects on nested presses are ignored.
> 
> As nested taps were changed to act as though PERMISSIVE_HOLD is set unless only IGNORE_MOD_TAP_INTERRUPT is (outside of Retro Shift), and Retro Shift ignores IGNORE_MOD_TAP_INTERRUPT, PERMISSIVE_HOLD has no effect on Mod Taps.
> 
> Nested taps will always act as though the TAPPING_TERM was exceeded for both Mod and Layer Tap keys.

In my experience, `IGNORE_MOD_TAP_INTERRUPT` and `TAPPING_TERM` are crucial to reducing the rate of misfires when using home row mods. The way their behavior changes when `RETRO_SHIFT` is enabled massively increases my misfire rate, since (if my understanding is not too far off the mark) *any* nested tap of a home row mod + a normal key results in the normal key getting modified, regardless of other settings.

I've tried to adjust my typing style but haven't been able to compensate for it, even with hours of practice. I'd also venture that I'm not the only one having issues with it, since as far as I can tell, [there are currently no layouts using the feature](https://github.com/qmk/qmk_firmware/search?q=RETRO_SHIFT) in the repository.

Ideally, there would be a better way to handle nested taps when `RETRO_SHIFT` is enabled, either by preserving existing settings or via new settings. I'm happy to investigate potential improvements, but given the way it's described in the documentation, I wonder if these are quirks of the existing implementation or if the limitations are somehow inherent to the feature. Outside of modifying my firmware, I haven't worked with C before, so I wouldn't want to go too deep into a rabbit hole if what I'm looking for is impossible in the first place.

Again, I apologize for any inaccuracies or misunderstandings here, and feel free to close this if the feature is considered to be working as expected.

IsaacElenbaas, 2022-07-04:
Your understanding is correct:
`IGNORE_MOD_TAP_INTERRUPT`'s effects for nested presses (sending both keys) are ignored
`PERMISSIVE_HOLD`'s normal reversal of `IMTI` in the above scenario is unnecessary (and isn't implemented anyway ?)
See [scenario 5](https://cdn.discordapp.com/attachments/451392568345755648/747388373781708930/tap-holds-qmk.pdf)
(#15741 cleans up that atrocious documentation wording)

Ideally, I would have added an option outside of Auto Shift to provide the "always pretend `TAPPING_TERM` is exceeded for nested taps" behavior so that configurations are consistent. The Retro Shift PR was already massive and horrible (and not quite working, I'm about to start working on the last two issues I and others have run across in the last few months and a few others are already in #15889), but maybe that is something we should look into now in a separate PR.

The Tap Hold configurations are reimplimented [here](https://github.com/qmk/qmk_firmware/blob/8f585153c470b07bb0c529ff49b39ef45f68d37e/quantum/process_keycode/process_auto_shift.c#L393-L429) for Retro Shift, but this effect (and only this one) has [a tweak](https://github.com/qmk/qmk_firmware/blob/8f585153c470b07bb0c529ff49b39ef45f68d37e/quantum/action_tapping.c#L179-L219) in `action_tapping.c` because `process_tapping` normally sends the two tapping keycodes and things further down the pipeline don't even know about the mod tap key. I believe commenting out the latter (or changing it to use a new setting) should be all that is required for Retro Shift, though I have not tried it.

(backstory, can be skipped)
The reason I didn't think to look further into that is because of typing habits Auto Shift without Retro Shift encouraged. Pressing another key interrupts your current hold, so when holding a key you have to wait for `AUTO_SHIFT_TIMEOUT` to elapse (and back then longer, things didn't send once it elapsed without a release/interrupt). This made press/releases for individual keys usually separate and led to rolling typically being only done when tapping two keys in quick succession and actually rolling (press 1->press 2->release 1->release 2).
The issue is Retro Shift should allow one to start a hold, start tapping another key, release the hold (now past the tapping term, but not when the interrupt happened), release the tap, and get a shifted and non shifted tap. This new typing style freedom doesn't enforce the above habits and can lead to this issue. Note that that isn't the case anyway yet though, #15889 fixes that behavior and one of the issues I need to tackle is that behavior when using two mod taps instead of one and a normal key. Currently you will always get two unshifted taps.

I don't think this is a major blocker for most people as you guess, though - I know five people other than me were using Retro Shift on their main boards before it was even merged. I have not bothered to merge my keymaps to upstream since they started getting wacky because I doubt anyone will try them and learning them would be a journey even if they did, surely others don't for similar reasons or just because it is unnecessary work.
Sorry for the wall of text, I started elaborating on things I will need to reference later.

jose-elias-alvarez, 2022-07-06:
Thank you for the detailed breakdown! You obviously have far more context on this than I ever could, but please let me know if there's anything I can do to help.

IsaacElenbaas, 2022-07-09:
One issue I just realized will exist is [sceniarios 5 and 6](https://cdn.discordapp.com/attachments/451392568345755648/747388373781708930/tap-holds-qmk.pdf). If we respect `IMTI`'s behavior in 5, and `PH`'s undoing of it, the tap key's release being delayed in either scenario will cause it to be shifted if `TAPPING_TERM` is greater than `AUTO_SHIFT_TIMEOUT` (which it likely is for most people). Supporting `IMTI` in scenario 5 prevents us from always using `PH`'s behavior in 6 in the code I linked above.

closed 2022-11-08 by github-actions[bot]