qmk/qmk_firmware #17245: RAW_ENABLE - raw_hid_receive is never called - ZSA Moonlander

martinmeinke, 2022-05-29:
Hello everyone,

recently I have become a proud owner of a ZSA moonlander keyboard.
I have a project in mind that would require me to control the LED's from the host computer, so I naturally turned to [Raw HID](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_rawhid.md) and tried to implement the host<-->device communication.

I enabled 
```
RAW_ENABLE = yes
CONSOLE_ENABLE = yes
```

and added some debug statements in the `raw_hid_receive` which I am monitoring them with `qmk console`:

```cpp
void raw_hid_receive( uint8_t *data, uint8_t length )
{
    print("raw_hid_receive");
}
```

I can see debug statements originating from `process_record_user`, so the debugging seems to work in general.

On the Host, I tried reading / writing with the python and node-js HID libraries.
After enabling `RAW_HID`, I seem to be receiving some messages originating from keypress/release, so it seems like the connection is working in general - but I have not been able to trigger the `raw_hid_receive` function by writing to the hid device from the host.

I had to set up a udev rule to change the default permissions on the `/dev/hidraw*` devices like so:
`KERNEL=="hidraw*", ATTRS{busnum}=="1", ATTRS{idVendor}=="3297", ATTRS{idProduct}=="1969", MODE="0666"`

Any hint on how I might proceed to debug this issue is highly appreciated.

drashna, 2022-06-21:
Is this using the zsa repo, or the qmk repo?

martinmeinke, 2022-06-22:
Hi @drashna - I tried both, neither of them with success.

martinmeinke, 2022-08-13:
Ok - I dug a little deeper and have some further insights.

For all of the experiments I performed so far I used [pyhidapi](https://github.com/apmorton/pyhidapi)
Today, I decided to continue my experiments and downloaded [hidapitester](https://github.com/todbot/hidapitester) which allows me to specify a `usage page` (whatever that is).

I am now able to receive data in my `raw_hid_receive` using:
`./hidapitester --vidpid 3297/1969 --usagePage ff60 --open --length 8 --send-output 0,0   --close`

I can also manually look for the /dev/hidraw device with the matching usage page and write like this
`echo -en "0\x00\x00\x00\x00\x00\x00\x00" > /dev/hidraw6`

```
void raw_hid_receive(uint8_t *data, uint8_t length) {
    // this stuff prints in the console
    print("I received some stuff\n");
    raw_hid_send(data, length);
}
```

However trying to receive any response from the keyboard using either `pyhidapi`/`hidapitester`/`cat /dev/hidraw6` have been unsuccessful.

Not sure how to continue from here - I'll keep you posted in case I discover something else.

drashna, 2022-08-14:
Very odd.  Because, I can confirm that VIA, which uses raw HID will work with the moonlander. So I'm not quite sure.

As for the console, make sure you use xprintf or the like, not just print.

closed 2022-12-14 by github-actions[bot]

Makonede, 2025-02-02:
having the same issue with my framework laptop 16, and nothing seems to be working to get the function to be called. every hid report i send is treated as an "`led_task`" according to the console, with some strange exceptions like `0x16` as the first byte turning on the caps lock indicator, so clearly they're getting received, but i don't know why `raw_hid_receive` isn't seeing them

here's the [fork](https://github.com/Makonede/qmk_firmware) i have

tzarc, 2025-02-03:
> having the same issue with my framework laptop 16, and nothing seems to be working to get the function to be called. every hid report i send is treated as an "`led_task`" according to the console, [...]

Unrelated; you’re probably opening the wrong endpoint.