qmk/qmk_firmware #17259: Nix-Shell does not pull in the Python library Pillow [Bug]

IQubic, 2022-05-30:
## Describe the Bug
I use NixOS. I'm attempting to use the `shell.nix` file to create a development environment for working on compiling and flashing keyboard firmware. I can successfully run `nix-shell` to enter a development environment. This does in fact give me a shell with the qmk cli application on $PATH. However, attempting to run any command with the CLI tool I get the following error:
```
Could not find module pillow!
Would you like to install the required Python modules? [y/n]
```
I've tried saying `y` to this, but that doesn't work.

## System Information
**Keyboard:** N/A
**Revision (if applicable):** N/A
**Operating system:** NixOS 21.05
**`qmk doctor` output:** Not able to run `qmk doctor` due to the aforementioned issues.

**Any keyboard related software installed?** 
N/A

tzarc, 2022-05-30:
Can you confirm #17243 fixes the issue?

IQubic, 2022-05-31:
How can I test out this fix on my own fork? I'm a bit new to Git.

IQubic, 2022-05-31:
Thanks. That seems to work for me.

closed 2022-05-31 by IQubic