qmk/qmk_firmware #17281: [Bug] Mod-tap keys on momentary layers don't work as expected

fxkuehl, 2022-06-01:
## Describe the Bug

My keymap uses layers with LT(layer, key) keys, but this issue also applies to MO(layer) layer keys. When I quickly press a "normal" key on a layer, it does not matter whether I release the LT key or the other key first. I get the key from the layer, not the key from the base layer.

If I do the same thing with an MT(mod, key) on the layer and release the layer key first, I get the key from the base layer instead, because the tap of the MT key is only processed when I release the MT key. By that time the layer state already switched back to the base layer.

This problem affects layers, but not modifiers. For example using Shift with a MT(mod, key), and releasing Shift first still gives me the shifted key. The release of the modifiers is delayed until the end of tapping the MT key. 

It would be more intuitive if layers and modifiers had the same, consistent behaviour with respect to how they affect MT keys.

## System Information

**Keyboard:** Keebio Iris
**Revision (if applicable):** rev4
**Operating system:** Linux
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /home/felix/src/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: kolibri
Ψ Repo version: 0.16.9
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

## Additional Context

I have a patch that implements the desired behaviour that I would be happy to contribute.

bschwehn, 2022-06-04:
that is brilliant, I was bothered by an insistent behaviour between MT and LT keys for a while and started debugging yesterday, initially thinking that achordion was causing the issue (https://github.com/getreuer/qmk-keymap/issues/13).

Applying this patch fixes my issue, at least in an initial test. 
Thank you so much, this made my day!

KarlK90, 2022-11-28:
With the merge of #17282 this issue is fixed. Thanks again @fxkuehl

closed 2022-11-28 by KarlK90