qmk/qmk_firmware #17303: Problems found in the MAC system

EasonQian1, 2022-06-04:
Entering sleep mode under MAC system will automatically wake up the trigger button.what caused it?

mechmerlin, 2022-06-05:
Zsolid and Sesame from ClickClack reached out to me to check this out. 

I just flashed my Capsule 65 with both the via and default firmware files and tested it on my M1 and Intel macOS machines.
I am unable to reproduce this with QMK I updated just a few minutes ago. 

M1 is running Monterey 12.4
Intel is running 10.15.7

This is the output of my QMK Doctor

Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /Users/mechmerlin/dev/qmk_firmware
Ψ Detected macOS 12.4.
Ψ Git branch: master
Ψ Repo version: 0.17.0
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go

EasonQian1, 2022-06-05:
Thanks, maybe it is caused by the peripheral power supply of mcu

EasonQian1, 2022-06-05:
[image]

EasonQian1, 2022-06-05:
> Zsolid and Sesame from ClickClack reached out to me to check this out.
> 
> I just flashed my Capsule 65 with both the via and default firmware files and tested it on my M1 and Intel macOS machines. I am unable to reproduce this with QMK I updated just a few minutes ago.
> 
> My Capsule 65 is connected to my laptop directly thru Apple USB C to C cable.
> 
> I am testing this by going to the Apple logo on the top left and selecting sleep and watching what happens. Keyboard is not triggering random key press to exit sleep mode. I am able to exit sleep mode by manually pressing a random key.
> 
> M1 is running Monterey 12.4 Intel is running 10.15.7
> 
> This is the output of my QMK Doctor
> 
> Ψ QMK Doctor is checking your environment. Ψ CLI version: 1.1.0 Ψ QMK home: /Users/mechmerlin/dev/qmk_firmware Ψ Detected macOS 12.4. Ψ Git branch: master Ψ Repo version: 0.17.0 Ψ CLI installed in virtualenv. Ψ All dependencies are installed. Ψ Found arm-none-eabi-gcc version 8.3.1 Ψ Found avr-gcc version 8.4.0 Ψ Found avrdude version 7.0 Ψ Found dfu-util version 0.11 Ψ Found dfu-programmer version 0.7.2 Ψ Submodules are up to date. Ψ QMK is ready to go

I compiled with angle 17.0 and still have this problem.Could it be caused by the peripheral circuit of the MCU?please help me check

closed 2022-06-08 by EasonQian1