qmk/qmk_firmware #17313: [Feature Request] Retro tapping timeout

antosha417, 2022-06-05:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
In the [docs](https://docs.qmk.fm/#/feature_auto_shift?id=retro-shift) I see that `RETRO_SHIFT` can be configured with a timeout.
>If `RETRO_SHIFT` is defined to a value, hold times greater than that value will not produce a tap on release. This enables modifiers to be held for combining with mouse clicks without generating taps on release. For example: `#define RETRO_SHIFT 500`

I think adding the same behaviour to `RETRO_TAPPING` would be a great with exactly the same reasoning as stands for `RETRO_SHIFT`. What do you think about that?

antosha417, 2022-06-06:
I managed to achieve desired behaviour with `TAPPING_TERM_PER_KEY` and `HOLD_ON_OTHER_KEY_PRESS_PER_KEY`.

This way when I hold mod tap key longer then it's tapping term and release it no keycodes are send
 to the computer. And as soon as I press and hold it I get hold action. So tapping term acts like timeout for tapping action and I get configure it to be as big or as little as I like.

I think you can safely close the issue.

codebynils, 2022-07-14:
Please, do not close this issue. It is a necessary feature. 

First of all, I use the Moonlander fork, so I cannot customise the keyboard in such a detail. And all of this settings seems to have side effects, so only a specific configuration works for me. 

I enabled Retro Tapping and it makes typing much better. But there are huge problems:

- If I use it with the mouse it does not recognise the mouse clicks. For example if I want to select many files I hold shift key which is configured as "held on 'a' character" and click on the last file in the list and it selects all files as wanted. But then I release the a/shift key an a is sent and the selection is revoked. So I have to handle hundred files one by one.
This happens with the programmed mouse keys of the keyboard (what i think is a bug on its own) just as with an external mouse (what is logical).
- Often I have the situation that I start a special key combo and stop in this process because I want to do something else first. So the hold key is down an by releasing it fires the key. I have no chance to stop it. But outside of the text document a key press can damage things. A hour ago I moved six emails to the archive folder in thunderbird, because the a key stroke seems to do this. And I have many situations in the last weeks where something unexpected happen in a program and I do not know what because the unintended key stroke triggered something.

So this timeout is probably not hard to implement and would make Retro Tapping much more enjoyable.

closed 2022-11-13 by github-actions[bot]

JohnRigoni, 2024-04-18:
I also wanted this feature! Here is a working [proof of concept](https://github.com/JohnRigoni/qmk_firmware). Theses changes can be applied to the ZSA firmware as well!

JohnRigoni, 2025-04-11:
Now that the retro tapping keyroll fix has been merged https://github.com/qmk/qmk_firmware/pull/23641, I have submitted a pull to add this retro tapping timeout feature https://github.com/qmk/qmk_firmware/pull/25138.