qmk/qmk_firmware #17360: [Bug] Splash and Reactive RGB Matrix Modes Will Not Trigger Slave from Master

stephaje, 2022-06-11:
## Describe the Bug

On my keebio/iris/rev6 keyboard, when using any reactive or splash RGB matrix mode, effects which should trigger the second half of the board only work from slave to master. i.e. Pressing keys on the slave may light up the master keyboard, but not the other way around.

## System Information

**Keyboard: keebio/iris
**Revision (if applicable): rev6
**Operating system: Windows
**`qmk doctor` output:** 
```
[x ~]$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: C:/Users/x/qmk_firmware
Ψ Detected Windows 10 (10.0.19044).
Ψ Git branch: master
Ψ Repo version: 0.17.0
⚠ Git has unstashed/uncommitted changes.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

daskygit, 2022-06-11:
Try add `#define SPLIT_TRANSPORT_MIRROR` to your config.h and flash both sides again.

stephaje, 2022-06-11:
Thank you @daskygit, that did resolve the issue. The use of this is outlined [here](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md), but unfortunately I did not find this while searching for a solution myself.

closed 2022-06-11 by stephaje