qmk/qmk_firmware #17381: Symbols "<" and ">" in mac OS with Latin American keyboard

DLopezGo90, 2022-06-13:
When typing symbol "<" on my mac keyboard (using Latin American keyboard, not US International) the web https://config.qmk.fm/#/test (test keyboard section) tells me keycode KC_NUBS

[image]

But when putting keycode KC_NUBS in the keymap of my Corne kbd, it types symbol "|" and not symbol "<" (what I want) in mac OS Latin American keyboard 

I think it is an issue. If it is not, what keycode KC can I use to type "<" using my Corne with a mac and Latin American keyboard

amrsoll, 2022-06-16:
QMK configurator does not take the OS keyboard layout into account. It only gives you the keycodes it can read directly from the keyboard. Could it be that your OS (MacOS) might be misinterpreting the keys? Can you confirm That the configurator shows the same keycode from your normal keyboard compared to your QMK keyboard?

amrsoll, 2022-06-21:
Ah now I think that I understand better. This is probably related to a limitation of shifted symbols in QMK.

See documentation here https://docs.qmk.fm/#/keycodes_us_ansi_shifted

DLopezGo90, 2022-06-23:
Sorry I didn't get it

@GMBermeo @amrsoll 

What keycode should I use to type "<" in mac OS Latin American layout? KC_LT?

and symbol ">"? KC_GT?

GMBermeo, 2022-06-23:
> Sorry I didn't get it
> 
> @GMBermeo @amrsoll 
> 
> What keycode should I use to type "<" in mac OS Latin American layout? KC_LT?
> 
> and symbol ">"? KC_GT?

Yes, try use KC_LT and KC_GT and let us know if it works for you 😁
And the layout doesn't matter in this case because QMK overwrites any kind of OS configuration.

DLopezGo90, 2022-06-24:
@GMBermeo bro what keycode can be used to type "~" in mac using US International layout?  and the symbol `

precondition, 2022-06-24:
Are you sure that MacOS considers your QMK keyboard to be ISO? If it thinks it is an ANSI keyboard, ISO keys like `KC_NUBS` and `KC_NUHS` tend to produce unexpected results.

GMBermeo, 2022-06-24:
> @GMBermeo bro what keycode can be used to type "~" in mac using US International layout? and the symbol `
> 
> The layout does matter. That's why https://github.com/qmk/qmk_firmware/tree/master/quantum/keymap_extras exists
> 
> For example KC_MINS types symbol - using US Int, and symbol ' using Latinamerican layout (OS Windows)

Key | Aliases | Description
-- | -- | --
KC_TILDE | KC_TILD | ~
KC_GRAVE | KC_GRV | ` and ~

[Take a look at the layout that I use.](https://github.com/GMBermeo/qmk_bermeo/tree/master/keyboards/crkbd/keymaps/bermeo)

In this documentation you send, for example, for the Brazilian layout every key starts with BR_, the same for the Spanish starting every key with ES_. You can absolutely ignores this and use the KC mapping.

[Here you can find more direct keycodes](https://github.com/qmk/qmk_firmware/blob/master/docs/keycodes.md).

DLopezGo90, 2022-06-27:
@GMBermeo nope, these direct keycodes are not working

For example KC_MINUS:

[image]

OS Windows Latinamerican layout --> KC_MINUS types '

OS Windows US International layout --> KC_MINUS types -

What am I doing wrong? why for you the layout doesn't matter and for me it does?

DLopezGo90, 2022-06-27:
> QMK configurator does not take the OS keyboard layout into account. It only gives you the keycodes it can read directly from the keyboard. Could it be that your OS (MacOS) might be misinterpreting [...]

It looks like OS keyboard layout matters

[image]

[image]

OS Windows Latinamerican layout --> KC_MINUS types '

OS Windows US International layout --> KC_MINUS types -

keyboard-magpie, 2022-08-03:
Appropriate support provided earlier in the issue, closing for now.

closed 2022-08-03 by keyboard-magpie

cNoveron, 2022-11-24:
Hey, wait, I'm not entirely sure this issue has been solved. @DLopezGo90 What was your solution to the problem?

DLopezGo90, 2022-12-01:
Not solved. I can't type symbol "<" in macOS with latin american layout. I have to use US International layout

fauxpark, 2022-12-01:
You need to set the keyboard type to ISO, as precondition mentioned earlier. macOS keeps track of this per-keyboard, not globally, so you can have an ANSI keyboard and an ISO keyboard both connected which will both do the correct thing when you type on each of them.

andersong9012, 2022-12-01:
@DLopezGo90 I had this same issue with an external windows keyboard. The workaround I found is to type Shift + Alt +3 for ‹ and Shift + Alt + 4 for ›. I know it's not the best solution but it allows me to type these two symbols without leaving my external keyboard.

ejerico, 2023-03-07:
This worked! I'm using a macOS with latin american layout too and Shift + Alt + 3 / 4 worked. Thank you!

ejerico, 2023-03-07:
This worked to me but with a slight difference! Shift + Option + 3 / 4. MacOS with latin american layout. Thank you!

etiennecelery, 2023-09-22:
This is still unsolved. I have the same problem, can't type the "‹" symbol with ISO keyboard on Mac.

fauxpark, 2023-09-22:
It is solved. Please see my previous comment.

etiennecelery, 2023-09-22:
[image]

The problem appears to be that VIA in order to generate the "<" symbol, it combines "shift" + ","

I even tried using macros with the same result, the output is ";" every time

fauxpark, 2023-09-22:
That's a completely different thing. VIA only shows the US legends for the keycodes. For the key next to left shift you want to use `KC_NUBS`.

IvanovskyOrtega, 2024-04-12:
Hi everyone! I was facing the same issue with my Corne kbd with no success but switching to a US layout in MacOS settings to use "<" and ">".

Finally I found a solution and I hope it can be useful for someone else. I don't know why the preinstalled Latinamerican keyboard layout doesn't work as expected, installing the following ones: https://github.com/neosergio/Latam-Keyboard (instructions in the repo's README) and selecting it as default in Keoyboard Input sources settings makes "<" and ">" available again.

The "Latin American B" layout worked perfect for me. Tested in my Corne and a Logitech MX Mechanical Mini.