qmk/qmk_firmware #17413: [Bug] CL_(TOGG|SWAP|CTRL) doesn't work on MacOS 12.4

oscarcarlsson, 2022-06-17:
Discovered the magic key codes and thought I should use the CL_(TOGG|SWAP|CTRL) keys to make this board more user friendly for others (so I can toggle if caps lock is a control key or not). However, these don't seem to work on MacOS 12.4 - caps lock continues to be caps lock, control continues to be ctrl and so forth.

I couldn't find any documentation if this is supported or not, or any old bugs regarding this.

## Describe the Bug
CL_(TOGG|SWAP|CTRL) keys don't seem to work on MacOS.

## System Information

**Keyboard:** kbdfans/kbd19x
**Revision (if applicable):** N/A
**Operating system:** MacOS 12.4
**`qmk doctor` output:** 
```
qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /Users/oscar/git/own/qmk_firmware
Ψ Detected macOS 12.4.
Ψ Git branch: oscarcarlsson
Ψ Repo version: 0.17.0
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

**Any keyboard related software installed?** 
None of autohotkey, karabiner or similar.

## Additional Context
N/A

oscarcarlsson, 2022-06-17:
The keymap can be found here: https://git.sr.ht/~monotux/qmk_firmware/tree/oscarcarlsson/item/keyboards/kbdfans/kbd19x/keymaps/oscarcarlsson/keymap.c

The AG_TOGG magic key works fine, if it matters. Really handy function!

fauxpark, 2022-06-17:
OS shouldn't matter here, as the keycodes themselves aren't actually sent to the host. All they do is change the behaviour internally. If your repo is up to date, it should just work.

oscarcarlsson, 2022-06-17:
I must be doing something wrong, then. 

I cloned the qmk repo earlier today, and branched off from 545923b0a0319f9d7d918e39f61594d8d5b64086 (the latest commit I can see now).

AG_TOGG works fine, but none of the CL_ stuff.

fauxpark, 2022-09-26:
@oscarcarlsson is this still an issue?

oscarcarlsson, 2022-09-26:
No, I just forgot to close the issue. Thanks for reminding me!

closed 2022-09-26 by oscarcarlsson