qmk/qmk_firmware #17473: [Feature Request] support for keychron q6

NittanyLion, 2022-06-24:
## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

The Keychron Q6 is advertised as a QMK keyboard but not on the list of supported keyboards here.  Support would be awesome!

lokher, 2022-06-25:
Please refer to `playground` branch of Keychron fork: https://github.com/Keychron/qmk_firmware/tree/playground/keyboards/keychron

NittanyLion, 2022-06-25:
Awesome, thanks!

lokher, 2022-06-25:
And here is Keychron Discord server: https://discord.com/invite/HAYbRrTsjN Both sharing and questions are welcome.

closed 2022-08-29 by tzarc

milkman45, 2022-10-14:
Hi - Is there an updated version of the Q6 file? I am getting the following error

keyboards/keychron/q6/q6_ansi_stm32l432_ec11/q6_ansi_stm32l432_ec11.c:195:5: error: implicit declaration of function 'encoder_insert_state' [-Werror=implicit-function-declaration]
  195 |     encoder_insert_state(0);
      |     ^~~~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:359: .build/obj_keychron_q6_q6_ansi_stm32l432_ec11_default/keyboards/keychron/q6/q6_ansi_stm32l432_ec11/q6_ansi_stm32l432_ec11.o] Error 1
Make finished with errors
make: *** [Makefile:414: keychron/q6/q6_ansi_stm32l432_ec11:default] Error 1

adophoxia, 2022-10-14:
> Hi - Is there an updated version of the Q6 file? I am getting the following error
>
> keyboards/keychron/q6/q6_ansi_stm32l432_ec11/q6_ansi_stm32l432_ec11.c:195:5: error: implicit declaration of [...]

Dealt with this issue before. It's just a matter of removing that one line with `encoder_insert_state(0)`. IIRC, it's because of encoder mapping support that's implemented on the keyboard (or keymap) itself.

milkman45, 2022-10-14:
Awesome, thank you. Think this is why i am getting the following error? Sorry I am learning how to program my keyboard on the fly.

Linking: .build/keychron_q6_q6_ansi_stm32l432_ec11_default.elf                                      [ERRORS]
 |
 | arm-none-eabi-gcc.exe: error: .build/obj_keychron_q6_q6_ansi_stm32l432_ec11_default/factory/major/test.o: No such file or directory
 |
make[1]: *** [builddefs/common_rules.mk:267: .build/keychron_q6_q6_ansi_stm32l432_ec11_default.elf] Error 1
Make finished with errors
make: *** [Makefile:414: keychron/q6/q6_ansi_stm32l432_ec11:default] Error 1

adophoxia, 2022-10-14:
> Awesome, thank you. Think this is why i am getting the following error? This is happening when trying to set my keymap as the default. Sorry I am learning how to program my keyboard on the fly.
> 
> Linking: .build/keychron_q6_q6_ansi_stm32l432_ec11_default.elf                                      [ERRORS]
>  |
>  | arm-none-eabi-gcc.exe: error: .build/obj_keychron_q6_q6_ansi_stm32l432_ec11_default/factory/major/test.o: No such file or directory
>  |
> make[1]: *** [builddefs/common_rules.mk:267: .build/keychron_q6_q6_ansi_stm32l432_ec11_default.elf] Error 1
> Make finished with errors
> make: *** [Makefile:414: keychron/q6/q6_ansi_stm32l432_ec11:default] Error 1

It's trying to find a file that doesn't exist. I'll assume that you copied only the Q6 folder. (There was a `factory` folder in the `playground` branch.) In this case, can you check the `rules.mk` in both keymap folders and see if `SRC += factory/major/test.c` is present. If it is, delete this since that's what's causing the error.

milkman45, 2022-10-14:
That worked, thank you. You are a life savior.