qmk/qmk_firmware #17477: [Feature Request] CH579M chip based keyboard support

ryanm101, 2022-06-25:
Is it likely that support for [mini keyboard](https://www.aliexpress.com/item/1005004231035046.html?spm=a2g0o.order_list.0.0.598b1802MpHjLq) Could be added?
datasheet here:
[CH579M.pdf](https://github.com/qmk/qmk_firmware/files/8985300/CH579M.pdf)

tzarc, 2022-06-26:
You'd need to upstream the MCU support into https://github.com/ChibiOS/ChibiOS-Contrib first.

closed 2022-08-29 by tzarc

adrianszymanski89, 2026-09-25:
Hi @zvecr why was this issue closed? I can see that the general demand for this feature is quite large. I myself would love to see this feature finished and implemented.

ryanm101, 2026-09-25:
Probably because if you see my other prs they while they work they need a lot of cleanup. Because i broke my hardware and the new hardware i ordered didnt have the same chip so I was unable to continue.