qmk/qmk_firmware #17546: [Bug] When turning on Caps Word, my keyboard types a number for some reason

Anomalocaridid, 2022-07-02:
Whenever I activate Caps Word, my keyboard immediately types a zero or a nine. I activate Caps Word by pressing both of my shift keys, which are also Space Cadet shift keys.

## Describe the Bug

Whenever I activate Caps Word, my keyboard immediately types a zero or a nine.

## System Information

**Keyboard:** Ergodox EZ
**Revision (if applicable):** N/A
**Operating system:** EndeavourOS Sway edition
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /home/anomalocaris/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: personal-keymap
Ψ Repo version: 0.14.23
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.1.0
Ψ Found avr-gcc version 8.3.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context
I have updated my fork of QMK earlier today, about 5 hours ago.

drashna, 2022-07-05:
Do you have a link to your keymap?

Anomalocaridid, 2022-07-05:
Here's a link to my keymap: https://github.com/Anomalocaridid/qmk_firmware/tree/personal-keymap/keyboards/ergodox_ez/keymaps/anomalocaridid

mflagg2814, 2022-09-27:
I see the same behavior on a GMMK2. I'm also using Space Cadet shift.

Anomalocaridid, 2022-11-17:
After a while of trying to use Caps Word and work around this issue on and off, I have a hypothesis as to why this issue occurs.

If I hold the second shift key for a short while before releasing it, Caps Word activates without typing a number.

In addition, I noticed that whether a zero or nine is output depends on which shift key I press second, getting a nine from the left shift and zero from the right shift.

I think that what is happening is that Caps Word gets activated, then Space Cadet Shift types a parenthesis, which gets "de-shifted" into zero or nine.

So maybe the implementation of Caps Word just needs to account for tapping a Space Cadet Shift key?

getreuer, 2022-11-28:
It would be good to fix this. Space Cadet is interesting: although it basically acts like a mod-tap, the implementation is completely separate ([code link](https://github.com/qmk/qmk_firmware/blob/d27b852fccf6165730b5a013e49ae69ab3124c56/quantum/process_keycode/process_space_cadet.c#L85)). And even when tapping the key, the Shift mod is immediately applied on key down, unlike normal mod-tap keys.

It sounds like how this bug is happening is:

1. The left Space Cadet key is pressed. The left Shift mod is applied.
2. The right Space Cadet key is pressed. The right Shift mod is applied, and the left Space Cadet key is now considered held (because of the `sc_last` variable in the handler).
3. Because left + right Shift mods are on, Caps Word activates. When Caps Word activates, it clears the shift mods ([here](https://github.com/qmk/qmk_firmware/blob/d27b852fccf6165730b5a013e49ae69ab3124c56/quantum/caps_word.c#L52)).
4. The Space Cadet keys are released before `TAPPING_TERM` expires. The Space Cadet handler considers the right Space Cadet key as being tapped, so it types `0`. This would normally type `)` because of the Shift mod, but Caps Word cleared it.

Similarly, `9` is typed if the right Space Cadet key went down first.

Anomalocaridid, 2024-03-23:
Just updated my firmware for the first time in a while. The problem appears to be fixed for me.

closed 2024-03-23 by Anomalocaridid