qmk/qmk_firmware #17550: [Bug] Nothing happens when I try to manipulate mouse report

Anomalocaridid, 2022-07-03:
In the keymap for my Ploopy Classic Trackball, I have a custom key that is intended to make scrolling go horizontally rather than vertically. However, when I try to set `mouse_report.h` and `mouse_report.v,` in `pointing_device_task_user(),` it appears to have no effect and my mouse continues to scroll vertically after pressing the key.

## Describe the Bug
Nothing seems to happen when I try to set `mouse_report.h` and `mouse_report.v` in `pointing_device_task_user()`.

## System Information

**Keyboard:** Ploopy Classic Trackball
**Revision (if applicable):** rev. 1.005
**Operating system:** EndeavourOS Sway Edition
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /home/anomalocaris/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: personal-keymap
Ψ Repo version: 0.14.23
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.1.0
Ψ Found avr-gcc version 8.3.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context
Here is a link to my [keymap](https://github.com/Anomalocaridid/qmk_firmware/blob/44a214724ce7fcacef0cee5ac9ff45c25aaa91da/keyboards/ploopyco/trackball/keymaps/anomalocaridid/keymap.c#L70). The code that implements the custom keycode, `HORI_SCROLL` is located at lines 70-91. `pointing_device_task_user()`, where I attempt to manipulate the mouse report is located at lines 83-91.

drashna, 2022-07-05:
Are you returning said mouse report? 

And could you post the code that you're using?

drashna, 2022-08-14:
`process_record_keymap` isn't supported by default, so there is that. 

But as for pointing_device_task_user, that does appear to be getting called.  And the code looks like it should work.   

Are you using `PLOOPY_SAFE_RANGE` rather than `SAFE_RANGE` for your keycode enum?

Anomalocaridid, 2022-10-16:
Sorry about that and sorry for taking so long to respond. I completely forgot that I use some custom stuff I defined in userspace:
```C
__attribute__((weak)) bool process_record_keymap(uint16_t keycode, keyrecord_t *record) {
    return true;
}

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case VRSN:
            if (record->event.pressed) {
                send_version();
            }
            return false;
        case FLASH:
            if (record->event.pressed) {
                send_flash_command();
            }
            return false;
    }

    return process_record_keymap(keycode, record);
}
```

For my keycode enum, I have this in my keymap:
```C
enum custom_keycodes_keymap {
    HORI_SCROLL = KEYMAP_SAFE_RANGE,
};
```

And this in userspace:
```C
enum custom_keycodes_user {
#if defined(KEYBOARD_ergodox_ez)
    VRSN = EZ_SAFE_RANGE,
#elif defined(KEYBOARD_ploopyco_trackball_rev1_005)
    VRSN = PLOOPY_SAFE_RANGE,
#else
    VRSN = SAFE_RANGE,
#endif
    FLASH, // Type qmk command for flashing firmware
    KEYMAP_SAFE_RANGE,
};
```

Here are links to my keymap and userspace in case I forgot anything else that is pertinent:
- https://github.com/Anomalocaridid/qmk_firmware/tree/personal-keymap/keyboards/ploopyco/trackball/keymaps/anomalocaridid
- https://github.com/Anomalocaridid/qmk_firmware/tree/personal-keymap/users/anomalocaridid

Rebasing onto the newest commits did not seem to fix anything.

Anomalocaridid, 2025-07-28:
For posterity, apparently `encoder_update_*` callbacks are already deprecated: https://docs.qmk.fm/ChangeLog/20250525#deprecation-of-encoder-update-kb-user