qmk/qmk_firmware #17576: [Bug]  bytesdfu-util: dfuse_download: libusb_control_transfer returned -1 (LIBUSB_ERROR_IO) dfu-util: Error during download

toniz4, 2022-07-06:
## Describe the Bug

When trying to flash the cantor keyboard firmware on a stm32 blackpill i get the following error:

```
~/src/qmk_firmware-valilla (master *) λ make cantor:default:dfu-util
QMK Firmware 0.17.5
Making cantor with keymap default and target dfu-util

arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   24600       0   24600    6018 cantor_default.bin

Size after:
   text    data     bss     dec     hex filename
      0   24600       0   24600    6018 cantor_default.bin

Copying cantor_default.bin to qmk_firmware folder                                                   [OK]
dfu-util: Cannot open DFU device 0a5c:21e6
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuDNLOAD-IDLE, status = 0
aborting previous incomplete transfer
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 56888
Erase           [=========================] 100%        56888 bytes
Erase    done.
Download        [==============           ]  57%        32768 bytesdfu-util: dfuse_download: libusb_control_transfer returned -1 (LIBUSB_ERROR_IO)
dfu-util: Error during download
make[1]: *** [platforms/chibios/flash.mk:41: dfu-util] Error 74
Make finished with errors
make: *** [Makefile:413: cantor:default:dfu-util] Error 1
```

## System Information

**Keyboard:** custom (the cantor firmware is compatible)
**Revision (if applicable):** 
**Operating system:** NixOS 22.05
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /home/cassio/src/qmk_firmware-valilla
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.17.5
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.2.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

The problem also happens in qmk-toolbox running on windows, possibly is a problem with my blackpill or/and with dfu-util, but I want to be sure.

siraphobk, 2022-07-19:
Hi. I have got the same problem. My OS is Ubuntu 22.04. 

However, after the flash failed, my Blackpill just bricked and cannot enter DFU mode anymore. Have you found this problem too? If so, do you know how to fix it?

toniz4, 2022-07-20:
> Hi. I have got the same problem. My OS is Ubuntu 22.04.
> 
> However, after the flash failed, my Blackpill just bricked and cannot enter DFU mode anymore. Have you found this problem too? If so, do you know how to fix it?

After chatting in the qmk discord server I found some people that had the same problem, but most of them didn't find a solution. Tried a couple things that Drashana suggested with no luck.

I think this is a problem with the mcu, I asked for a full refund in my order and I got it.

But I still can enter the dfu mode, did you tried heating up the mcu with a blow dryer?

And one thing I observed, the other guy that was having the same problem as me had the same batch number on the mcu, 117.

[image]

siraphobk, 2022-07-24:
My batch number is 145. Doesn't really know if that's the case. However, I managed to get the board into DFU mode. Last time I did not press the right sequence of BOOT0 and RST.

Here are the things I tried.

1. Flashing with Arduino IDE. I tried to flash a blinking program using DFU mode but it did not succeed because the computer couldn't read the MCU memory.
2. Flashing using STLink. Flash succeeded with no problem but my computer couldn't recognize the keyboard. I tried `lsusb` and my keyboard was not found.
   ```
   16:38:58 : Memory Programming ...
   16:38:58 : Opening and parsing file: f401_blackpill_default.bin
   16:38:58 :   File          : f401_blackpill_default.bin
   16:38:58 :   Size          : 54.25 KB 
   16:38:58 :   Address       : 0x08000000 
   16:38:58 : Erasing memory corresponding to segment 0:
   16:38:58 : Erasing internal memory sectors [0 3]
   16:38:59 : Download in Progress:
   16:38:59 : File download complete
   16:38:59 : Time elapsed during download operation: 00:00:01.334
   ```

[image]
Hooking up the board with STM32F4Disc's STLink

drashna, 2022-08-14:
Are these from WeAct's store on ali? or another seller?  If it's another seller, it could be fake chips.

toniz4, 2022-08-14:
> Are these from WeAct's store on ali? or another seller? If it's another seller, it could be fake chips.

Mine are from another seller. 99% Sure they were fake chips. Managed to get a refund. Bou

Closing because this is most likely a problem with hardware.

closed 2022-08-14 by toniz4

Schievel1, 2022-12-22:
I had the same problem, bath 117. I flashed the tinyuf2 bootloader with stlink and after that they worked.

rah2501, 2023-11-04:
I came across the same error with a Blackpill.  For anyone who follows, I successfully flashed a binary compiled by Arduino using a Raspberry Pi Debug Probe and openocd like so:

```
openocd -f "interface/cmsis-dap.cfg" -f "target/stm32f4x.cfg" -c "program Blink.ino.bin verify reset exit 0x08000000"
```