qmk/qmk_firmware #17585: [Feature Request] Hi-res scrolling

b-, 2022-07-07:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

Windows and Linux support a "resolution modifier" report for scrolling, which allows finer scrolling instead of just "ticks." Specifically, one can do smooth per-pixel scrolling. It also seems to use fewer resources, possibly because of the larger descriptors.

Unfortunately, it seems that there's virtually no public code or documentation available. The following is everything I've managed to gather in terms of documentation and examples:

  - [wheel.docx](https://download.microsoft.com/download/b/d/1/bd1f7ef4-7d72-419e-bc5c-9f79ad7bb66e/wheel.docx) (_Enhanced Wheel Support in Windows_)
  - [our_descriptor.cc](https://github.com/jfedor2/hid-remapper/blob/master/firmware/src/our_descriptor.cc) (from https://github.com/jfedor2/hid-remapper)
  - [HID: input: use the Resolution Multiplier for high-resolution scrolling](https://patchwork.kernel.org/project/linux-input/patch/20181205004228.10714-5-peter.hutterer@who-t.net/) (from the Linux kernel)

This almost certainly requires MOUSE_EXTENDED_REPORT to be enabled based on my testing.

I started mocking up something here: https://github.com/b-/qmk_firmware/tree/resolution_multiplier

However this is the first time I've ever gotten this close to USB or to HID descriptors.

I haven't tested my above code yet, either, though I will as soon as I get the chance.

Anyway, I'm posting this here to help me keep track of this, and of my progress (or lack thereof). I expect to edit this post frequently.

pandages, 2022-07-08:
I think this is a really cool concept.  Is it for using a rotary encoder as a mouse-style scrollwheel?  Could it potentially be mapped to any keys?

drashna, 2022-07-08:
Taking a look at "wheel.doc", I'm not sure how practical this is.  Namely, this appears to need custom drivers to support it (namely for registry settings). And those dirvers need to be signed, or the overall security of the system degraded (driver signing disabled, which requires secure boot to be disabled).  

It's definitely possible to implement, but I'm not sure that it is worth the effort here.

b-, 2022-07-08:
I can say with confidence this works without drivers other than what's bundled with Windows and Linux. 

[hid-remapper](https://github.com/jfedor2/hid-remapper) allows one to set a custom scroll multiplier and it works without drivers just fine (tested on Windows 10 and 11, and on KDE Plasma). [This](https://github.com/jfedor2/hid-remapper/blob/master/firmware/src/our_descriptor.cc) what I'm stealing the HID collection tree from, so if I can simply do what it's doing this would work just fine.

b-, 2022-07-08:
> I think this is a really cool concept.  Is it for using a rotary encoder as a mouse-style scrollwheel?  Could it potentially be mapped to any keys?

QMK already has scroll wheel support via [mouse keys](https://docs.qmk.fm/#/feature_mouse_keys). You can just add `MOUSEKEY_ENABLE = yes` to your `rules.mk` and use `KC_MS_WH_UP` and `KC_MS_WH_DOWN` as keycodes for mouse wheel scrolling. 

My goal here is to add a resolution multiplier for wheel scrolling, as that will allow finer precision than an ordinary scroll wheel allows for (and this works without drivers on Windows Vista and up, or Linux). 

My motivation is for DRAG_SCROLL on my [Ploopy Classic](https://ploopy.co/classic-trackball/) to work at a finer precision.

Another idea is to put some kind of tiny trackball on top of a mouse, much like Apple's discontinued [Mighty Mouse](https://en.m.wikipedia.org/wiki/Apple_Mighty_Mouse).

I mean, you can already do all of these things with QMK as is, but scrolling be "jumpy" compared to something like a good multitouch trackpad. This actually allows for providing such smooth scrolling!

b-, 2022-07-08:
Microsoft's own mice that took advantage of this had "notchless" scroll wheels whose encoders were much more sensitive than what's standard today.

Example: https://www.cnet.com/reviews/microsoft-wireless-laser-mouse-7000-2008-review/

b-, 2022-08-31:
I bought a [Microsoft Surface Precision Mouse](https://www.microsoft.com/en-us/d/surface-precision-mouse/8qc5p0d8ddjt?activetab=pivot:overviewtab) to see how Microsoft implements hi-res scrolling, and gathered some data here. This is on a Windows 11 machine, with the mouse attached over USB.

First, I installed the relevant drivers for the mouse, and then disabled "flick" scrolling in the new settings panel for the mouse. I also inverted the scroll wheel for preference, but I don't think that's particularly relevant.

Then I unplugged the mouse, and opened the USB port in Wireshark (with the other devices filtered out on the USBPcap level).

Once the capture began, I plugged in the mouse, holding it upside down to prevent any tracking activity. Then I played with the scroll wheel, scrolling down a little bit, then up a little bit, then down a little bit again (varying the speed as I went along). I added comments to the packet capture to mark the end of the initialization handshake, and each scroll direction change.

What I saw that I found particularly curious is that this mouse actually doesn't send a `count` descriptor in its wheel reports. It instead varies the resolution multiplier with the direction and magnitude of the wheel motion, and that's the only information about the wheel. Still, I'd think it's enough.

Although I also should note that when I tried using the mouse on my Pinebook Pro in KDE Plasma (Wayland), neither Firefox nor Chrome nor native Qt apps had any hi-res scrolling support out of the box. I didn't try testing further. 

The spec says the mouse really shouldn't send hi-res scrolling information until the host sends a command to switch into a fancy enhanced mode, so perhaps libinput doesn't have support for this mouse yet to send that command?

I also tried in Chromium and Firefox on Windows Subsystem for Linux, in Wayland, and hi-res scrolling was supported. I think WSL uses libinput as well (but of course Windows is abstracting the input packets), and I know for certain that libinput does normally support hi-res scrolling where allowed.

Attached is my wireshark capture.
[hrscroll-surface_precision_mouse-with-comments.zip](https://github.com/qmk/qmk_firmware/files/9457365/hrscroll-surface_precision_mouse-with-comments.zip)

Alabastard-64, 2022-09-06:
So I found [this](https://www.reddit.com/r/Trackballs/comments/qwbg12/a_trackball_with_two_balls/) reddit thread where someone made a trackball with the extended mouse report. a tiny bit more detail [here](https://www.reddit.com/r/Trackballs/comments/qwbg12/comment/hl4jiwi/).
looking at his [github](https://github.com/jfedor2/two-ball-trackball) you can see he breaks out the HID report (with comments no less!) [here](https://github.com/jfedor2/two-ball-trackball/blob/master/firmware/src/trackball.cc)
looking at the HID report: line 140 to 160 seems to detail the scroll wheel info.
**NOTE: while he does have a config utility it is just for changing settings I do not believe this requires a custom driver**

Alabastard-64, 2022-09-06:
you could try reaching out to him on reddit or github to see if he can give you more info but it seems that he had the same info you found but must have tinkered until he got it working so he may be able to help here

jfedor2, 2022-09-06:
Hi guys. I think I should be able to help.

As you've already noticed there's some stuff in the report descriptor that's required for this to work. I'd recommend you start with the exact same descriptor HID Remapper uses (the relevant top level collection, you can change the report ID) and only start tweaking it to your needs once you confirm hi-res scroll is working. There's some stuff that's not part of the spec that Windows expects anyway, like the resolution multiplier using 2 bits even if 1 would be enough. Also the logical collection semantics might not be obvious if you're new to HID report descriptors.

The important part is the feature report, which is different from input/output reports. It's a sort of configuration variable that the host can read and write. I don't know if QMK currently uses feature reports for anything.

In our case the feature report will contain two values - the multipliers for horizontal and vertical scroll. In theory there can be multiple levels of the multiplier, but since in practice the operating systems choose the highest value anyway, I think it makes sense to only have two levels, 1x (legacy behavior) and the maximum, which is 120x. This is what it looks like in the report descriptor:

```
Logical Minimum (0)
Logical Maximum (1)
Physical Minimum (1)
Physical Maximum (120)
```

What this means is that if the relevant bits in the feature report are set to 0 (which should be the default), the multiplier will be 1x. If they are set to 1, it will be 120x. When it's 120x, it means 120 units of scroll correspond to 1 normal/legacy scroll "tick", giving you higher resolution.

To recap you need:

1. The relevant parts in the report descriptor.
2. The ability to receive and send feature reports.
3. Adjust what you send in the normal input reports in the horizontal/vertical scroll fields, depending on the values set by the host in the feature report.

An operating system that doesn't know about high resolution scroll will not touch the resolution multiplier in the feature report and will expect 1 unit=1 legacy tick. An operating system that knows about high resolution scroll will write 1 (or whatever the logical maximum is) to the resolution multiplier values in the feature report and will then expect 120 units=1 legacy tick (or whatever the physical maximum is).

It's also a good idea to detect USB mounting event and reset the resolution multiplier to the default of 1x, otherwise you run into issues when you reboot from an OS that knows about hi-res scroll to one that doesn't.

I'm not familiar with QMK source code, but I'll be happy to answer any questions you might have about HID so please don't hesitate to ask for clarification on any of this.

Alabastard-64, 2022-12-07:
Posting to keep this active for now as a reminder for myself.
I am planning on getting around to trying to add this into QMK unless someone else is on it already such as @fauxpark but I have a few other things to get too first.

Alabastard-64, 2023-01-03:
Okay just a quick update on my progress.
1. I have taken @fauxpark 's code and made some modifications and also included vusb but not ChibiOS yet (once it works I will port it over)
2. I have also added in Extended (16bit) Mouse Wheel reports
   * Working great and reduces code size a bit with my `pointing_modes` included.
   * I currently have not added anything for the boot protocol for mouse wheel and AC pan as I am pretty sure these are not supported by boot anyways _Please let me know if I am wrong on this!_
4. I have gotten the code to compile but there is some very strange behavior with scrolling so far:
  * Similar to what @b- has reported I am having constant positive scroll output while pointing device is active.
 
Will need to look into this a bit more but I think I'm close to getting this working

b-, 2023-01-03:
Hey, thanks for working on this! A lot has gotten in the way of me making more progress, and it's hard to debug this sort of thing when you consider that everything I know about USB and HID I learned from working on this :) but I'll see if I can poke at your branch at some point anyway.

Just a random thought: what if you swap the wheel and AC pan entries in the descriptor? I wouldn't expect that to fix anything, but if it changes the axis upon which we get rogue outputs it might point us in the right direction? (Pun not intended, I promise!)

Alabastard-64, 2023-01-04:
Hmmm after having a think about it, I may know what the issue is: (get
report code in lufa.c and the resolution_multiplier_t data struct).
Unfortunately though I will not be able to test for a day or so.

Also I will triple check that I have properly set up the mouse descriptor
as unfortunately it is a forest of #ifdefs now...

Alabastard-64, 2023-01-04:
Well I was right about the issue.

I now am no longer getting any errors with the Hires code enabled but I am pretty sure it is not working correctly as it doesn't seem to behave any differently while scrolling to the other board I have without Hires enabled.

Alabastard-64, 2023-01-04:
@b- if you have wireshark set up already could you check if [this](Alabastard-64/qmk_firmware/feature_hires_scroll) is working?

Alabastard-64, 2023-01-04:
BOOM! got it working! @b-

We were missing a crucial aspect of the HID report from the MS doc that seemed useless.

Currently because changing the multiplier requires changing the HID report it needs to be set at compile time.
Now I just need to adjust the pointing mode behavior when it is active to drop the divisor to bump up the sensitivity.

I do have some bad news however...
Seems that different apps behave very differently in response to the Hires scrolling. basically all apps fall into 3 distinct groups
1. Full Support of Hires scrolling  (Firefox, discord, etc.)
   * Works great! full smooth single pixel scrolling!
2. No support but can detects Hires scrolling (Windows file Explorer and most apps using default windows interfaces)
   * Detects Hires scrolling but does not support sub line scrolling so instead waits until `ticks > MOUSE_WHEEL_MULTIPLIER` to move the mouse setting of lines per tick `DEFAULT = 3`
   * This is a good thing tbh as it will respond well to changes in sensitivity due to hires scrolling being enabled (see below).
3. Absolutely no support whatsoever (Libre Office, GIMP etc.)
   * Will behave as if hires scrolling is not enabled at all:
      * Does not support sub line scrolling and in fact will just do the windows default of `1 tick = 3 lines` 
      * **This will respond badly to increased sensitivity to manage hires scrolling**

So overall great that this now works but it unfortunately has some minor drawbacks with some apps...
Managing a comfortable sensitivity may be challenging for some apps but when it works it is actually pretty awesome!

b-, 2023-01-04:
Wow! That's awesome!! Thank you!

I expected it to be about as picky as you're seeing -- that's about what I've experienced with @jfedor2/hid-remapper

Most mouse driver software that implements hires scrolling seems to turn it on and off based on what window your input is going into. Logitech is horribly picky about this and has a very limited whitelist of .exe files that it enables hires scrolling on. Microsoft ~~Intellipoint~~ _Mouse and Keyboard Center_ lets you pick which apps let you do this, though.

Honestly part of my interest in a "hardware" hires scroll thing like this is to help put pressure on Microsoft to better implement their spec :)

Alabastard-64, 2023-01-05:
Yeah I could see how logi would be that way about it not wanting a customer to have one of their apps suddenly become wildly oversensitive on scroll as soon as they switch to their hardware.

I think for QMK having it as an option with a warning in docs and handling the sensitivity in my pointing modes automatically would be good and makes QMK mouse support more feature complete IMO.

One additional caveat:
I currently have not gotten it working when the mouse HID report is on it's own separate interface so that aspect still needs to be worked out before I will be able to take the PR out of draft just need to figure out where that is breaking.

Alabastard-64, 2023-01-08:
Alright I have everything working at least on U4 chips and it is ready to test on U2 and ARM chips once that PR is merged we can close this request

Weerasak1239, 2023-03-14:
119e1708825ef8fce037be3116fae39e035d438d

Weerasak1239, 2023-03-14:
Snooker1239

iicurtis, 2023-09-30:
I don't really understand HID setup, so I am hoping someone can help clarify some things. Looking at the Microsoft `wheel.docx`, they give the following collection hierarchy:

```
Mouse Application Collection
    Mouse Logical Collection
        Pointer Physical Collection
            Buttons (Input Report 0x01)
            X (Input Report 0x01)
            Y (Input Report 0x01)
            Logical Collection
                Resolution Multiplier (Feature Report 0x02)
                Wheel (Input Report 0x01)
            End Logical Collection
            Logical Collection
                Resolution Multiplier (Feature Report 0x02)
                AC Pan (Input Report 0x01)
            End Logical Collection
        End Physical Collection
    End Logical Collection
End Application Collection
```

However, the current QMK hierarchy is noticeably different. Instead of **Application -> Logical -> Physical**, the QMK hierarchy goes **Application -> Physical**. 

```
Mouse Application Collection
    Pointer Physical Collection
        Buttons (Input Report 0x01)
        X (Input Report 0x01)
        Y (Input Report 0x01)
        Wheel (Input Report 0x01)
        AC Pan (Input Report 0x01)
    End Physical Collection
End Application Collection
```

I attempted to modify [usb_descriptor.c](https://github.com/iicurtis/qmk_firmware/commit/06d6c109b39cbdb79893658bd7163606670ddde4) to add an additional logical map and the corresponding Resolution Multiplier collections, but I am now realizing this was a bit too naive. My mouse breaks after updating the HID report structure, which probably needs to be reflected elsewhere. 

- Do we need the additional logical map or can I add directly to the physical collection?
- Once the logical collections are added, what else needs to be updated?

BanchouBoo, 2023-09-30:
Someone was working on a PR for this feature but I can't seem to find it anymore, what happened to it?

DejayRezme, 2023-12-26:
I just wanted to add my user experience with an apple magic mouse I used a while at work:

**It was the best scrolling experience I've ever had!** It is a touch mouse of course with a smooth trackpad on top. But you could very smoothly scroll the window vertically 1:1 and it felt like the perfect mouse wheel. No indents or "3 lines at a time". Just smooth fast responsive scrolling. And you could "fling" the window content for fast scrolling and then stop it again like it was a weighted wheel. I still miss that mouse. It's one of those little things that you don't know you want until you try it (like a third dedicated mouse button for your ring finger).

So this could also be implemented with two of those "Cirque GlidePoint Circle Trackpads" one for mouse pointer and one for scrolling.

thejevans, 2024-02-20:
@b- I just got a Ploopy Adept hoping to have precision/hi-res scrolling, and I'm saddened to see it's not an option. I'm fairly green with QMK, but this is a priority for me, so I'm happy to help any way I can. Are you stuck on anything in particular?

b-, 2024-02-20:
> @b- I just got a Ploopy Adept hoping to have precision/hi-res scrolling, and I'm saddened to see it's not an option. I'm fairly green with QMK, but this is a priority for me, so I'm happy to help any way I can. Are you stuck on anything in particular?

As you may have noticed I haven't touched this in a long while. However, I would love help solving this once and for all!

I want to warn you, though, I was kind of stuck on everything! I understand USB HID descriptors and the like at a cursory level, having learned virtually everything I know about it from trying to solve this. I also am not very experienced in C/C++, either. 

That said, the main blockers (at least for me) were that I was getting "weird" cursor movement and it appeared that some input buffer on Windows was getting filled without being properly flushed. This happened just from moving the cursor around while hidpi was enabled in the firmware. (I think I mostly used a patch from someone else's comment in this thread or something like that, but I was encountering similar issues when I tried to crudely craft the descriptor and reports myself). 

I assume that some of the messages being generated were malformed, and I don't really remember where I left off with that. I have a _hunch_ that the descriptor is what's malformed, rather than the reports, but I'm not sure and I don't really have any backups of what I was trying besides anything remaining on some lingering branch at https://github.com/b-/qmk_firmware

I know I was looking at Wireshark messages, and I even purchased Microsoft mouse that does support hidpi scrolling to record and compare its own descriptor and reports, but I don't remember where I went from there. 

(It doesn't help that I lost my job around the time I stopped working on this and have still been unable to find full time employment since then. But I suppose that gives me more time now to potentially pick it back up...)

If you come up with anything interesting please ping me, either here and/or on the QMK Discord server (my Discord username is @bri9). I'd be more than excited to test with my old Ploopy Classic. 

Oh, and I'm sure I already mentioned it more than once in this issue thread, but @jfedor2 has an excellent repository for a different HID device firmware, [hid-remapper](https://github.com/jfedor2/hid-remapper), with a working and customizable scroll multiplier.

Good luck!

evan-goode, 2024-02-20:
Hi, I'm also interested in getting this working eventually, though I don't have much time to work on it right now. Like @b- I also tried implementing this feature a couple years ago, but I can't remember what I was stuck on.

I recall these articles by Peter Hutterer (libinput developer) were helpful to me:

- https://who-t.blogspot.com/2021/08/libinput-and-high-resolution-wheel.html
- https://who-t.blogspot.com/2018/12/high-resolution-wheel-scrolling-on.html

Sorry for the noise!

BanchouBoo, 2024-06-27:
> Someone was working on a PR for this feature but I can't seem to find it anymore, what happened to it?

Realized I still had a local copy of the PR so I checked the log to find the username of the person working on it, some light searching led me to this post on reddit. @drashna did anything ever come of this? If desired I could upload it, but wouldn't want to do so without getting permission first.

[image]

justinjradi, 2024-07-04:
Hey. I'm not familiar with QMK, but I just implemented this using both ST's USB Library and TinyUSB. It seems to be working fine for the hosts I've been able to test on (Windows 10 and 11) but I'm still kinda confused. Here's my report descriptor for reference:

```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        //   Usage (Mouse)
0xA1, 0x02,        //   Collection (Logical)
0x85, 0x01,        //     Report ID (1)
0x09, 0x01,        //     Usage (Pointer)
0xA1, 0x00,        //     Collection (Physical)
0x05, 0x09,        //       Usage Page (Button)
0x19, 0x01,        //       Usage Minimum (0x01)
0x29, 0x05,        //       Usage Maximum (0x05)
0x15, 0x00,        //       Logical Minimum (0)
0x25, 0x01,        //       Logical Maximum (1)
0x95, 0x05,        //       Report Count (5)
0x75, 0x01,        //       Report Size (1)
0x81, 0x02,        //       Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //       Report Count (1)
0x75, 0x03,        //       Report Size (3)
0x81, 0x01,        //       Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //       Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //       Usage (X)
0x09, 0x31,        //       Usage (Y)
0x95, 0x02,        //       Report Count (2)
0x75, 0x08,        //       Report Size (8)
0x15, 0x81,        //       Logical Minimum (-127)
0x25, 0x7F,        //       Logical Maximum (127)
0x81, 0x06,        //       Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xA1, 0x02,        //       Collection (Logical)
0x85, 0x02,        //         Report ID (2)
0x09, 0x48,        //         Usage (Resolution Multiplier)
0x95, 0x01,        //         Report Count (1)
0x75, 0x02,        //         Report Size (2)
0x15, 0x00,        //         Logical Minimum (0)
0x25, 0x01,        //         Logical Maximum (1)
0x35, 0x01,        //         Physical Minimum (1)
0x45, 0x78,        //         Physical Maximum (120)
0xB1, 0x02,        //         Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x85, 0x01,        //         Report ID (1)
0x09, 0x38,        //         Usage (Wheel)
0x35, 0x00,        //         Physical Minimum (0)
0x45, 0x00,        //         Physical Maximum (0)
0x95, 0x01,        //         Report Count (1)
0x75, 0x08,        //         Report Size (8)
0x15, 0x81,        //         Logical Minimum (-127)
0x25, 0x7F,        //         Logical Maximum (127)
0x81, 0x06,        //         Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //       End Collection
0xA1, 0x02,        //       Collection (Logical)
0x85, 0x02,        //         Report ID (2)
0x09, 0x48,        //         Usage (Resolution Multiplier)
0x95, 0x01,        //         Report Count (1)
0x75, 0x02,        //         Report Size (2)
0x15, 0x00,        //         Logical Minimum (0)
0x25, 0x01,        //         Logical Maximum (1)
0x35, 0x01,        //         Physical Minimum (1)
0x45, 0x78,        //         Physical Maximum (120)
0xB1, 0x02,        //         Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x35, 0x00,        //         Physical Minimum (0)
0x45, 0x00,        //         Physical Maximum (0)
0x75, 0x04,        //         Report Size (4)
0xB1, 0x01,        //         Feature (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x85, 0x01,        //         Report ID (1)
0x05, 0x0C,        //         Usage Page (Consumer)
0x0A, 0x38, 0x02,  //         Usage (AC Pan)
0x95, 0x01,        //         Report Count (1)
0x75, 0x08,        //         Report Size (8)
[... 10 lines cut ...]
```

Other than making sure that you follow the correct collection hierarchy in the report descriptor, the key to getting it to work is to use the feature reports correctly. Again, here's the format of the feature report:

```
typedef struct __attribute__((packed))
{
	uint8_t report_ID;
	uint8_t scroll_resolution : 2;
	uint8_t pan_resolution : 2;
	uint8_t padding : 4;
} ResMultiplierReport;
```

Since the resolution multiplier usages have a logical range of 0 to 1, I interpret them as really just saying whether the wheels are in high-resolution mode. I haven't played around with them too much, but I think the physical maximums are the actual resolution.

Here's the important part: In addition to preparing for a Set_Report request with the feature report, you also need to configure your device to respond to a Get_Report request for the feature report. The Get_Report request preceeds the Set_Report request. I think this is totally undocumented and I only figured it out by analyzing the Wireshark capture that Bri shared. I highly recommend everyone take a look at the capture themselves.

What I'm still confused about is why? If there was just one Get_Report request, I'd think it's the host just asking for the capabilities of the device, and then the Set_Report requests that follow are just configuring the device. But for some reason every Set_Report request seems to be preceded by a Get_Report request. In all my testing so far, all the Set_Report requests I've seen just echo what my device sent in response to the Get_Report request. How many times have I said Get_Report and  [... cut ...]
Right now, I just have the feature report I respond with hard coded with both scroll_resolution and pan_resolution set. But if and when the host sets values other than what I initially reported, am I supposed to echo that or just respond with the same thing?

A couple more notes:
- I tried adding AC Zoom in the same manner, but it didn't work. I don't think Microsoft's mouse driver supports it :(
- It looks like Wireshark has a bug where it interprets the wheel and AC pan data as resolution multipliers, so don't be confused if you see that.

morgannewellsun, 2024-08-14:
Hi everyone. I'm not sure what the progress on this is, but I wanted to report that, with the help of the descriptor from @justinjradi, I was able to get a simple smooth scrolling example working. Please see [this commit](https://github.com/morgannewellsun/qmk_firmware/commit/8c7e992761c143cb0c45b0cec094fa2912f14669) for the details.

A few things I'd like to look into further:
- Clean things up and add a `#DEFINE` to turn the entire smooth scrolling feature on and off.
- Currently, smooth scrolling is set to a constant 1:120 resolution, and this is hard-coded into the USB HID description as an on/off switch. Is it possible to modify the description to allow this resolution to be adjusted dynamically by the keyboard?
- Provide some function(s) that lets user/kb code control smooth scrolling behavior.
- Integrate smooth scrolling better into existing pointing device code, so that end-users can turn on the feature without their scrolling sensitivity being slashed by a factor of 120.
- Add support to platforms other than ChibiOS.

I am a complete novice when it comes to the nitty-gritty of QMK, and I've also never touched USB HID before this in my life. So any support in polishing this feature up would be immensely appreciated. Maybe we can also push for this to be integrated into the official QMK repo? Although I'm not sure how that really works, I've never contributed to an open source project before.

filyp, 2024-08-19:
@morgannewellsun I pulled your changes and confirm they work wonderfully, you're a wizard

IsaacElenbaas, 2024-08-20:
Thanks so much! Great timing - I got my first trackball without a scroll wheel recently and only had to survive two or three weeks without smooth scrolling.

A few things:
* Note that that commit won't apply cleanly to the current QMK master, for others planning to use it. 0f341dfd is the latest upstream commit on @morgannewellsun's branch. (I think I can bear my trackball being a few months back from the latest and greatest haha)
* Even after adding buffer variable overflow protection and playing with lower DPI and such I was having up and down wiggling if I tried to scroll quickly. evtest showed reports with pretty random signs in the int8 range; limiting to 8 bits to feed the report values fixed it. Might be a Linux thing.
* For anyone with a Ploopy device, you will definitely want `#define PLOOPY_DRAGSCROLL_SEMAPHORE 0` with this (by default they eat all but every fourth scroll report).

morgannewellsun, 2024-08-20:
@IsaacElenbaas When you're talking about the wiggling issues when scrolling quickly, are you using 8 bits or 16 bits for the wheel reports?

filyp, 2024-08-20:
One issue I still have (on ploopy trackball) is that scrolling only starts after a threshold of 60 is crossed. So the start of scrolling always looks like:
```
event6   POINTER_SCROLL_WHEEL    +3008.498s	vert 7.50/60.0* horiz 0.00/0.0 (wheel)
event6   POINTER_SCROLL_WHEEL    +3008.499s	vert 0.12/1.0* horiz 0.00/0.0 (wheel)
event6   POINTER_SCROLL_WHEEL    +3008.500s	vert 0.12/1.0* horiz 0.00/0.0 (wheel)
...
```

morgannewellsun, 2024-08-20:
@filyp can you confirm that this isn't due to your drag scroll implementation?

filyp, 2024-08-20:
@morgannewellsun this happens also without drag scroll, using normal mouse wheel. I experimented with different values for `HID_RI_PHYSICAL_MAXIMUM(8, 0x78)` (which controls some kind of sensitivity) and the threshold of activation was always 60.

So you don't have that issue? Could be something in how ploopy handles scrolling, then. In practice it isn't a big issue now, after I set a high scrolling DPI + low scroll speed in the system.

morgannewellsun, 2024-08-20:
> @morgannewellsun I was filling the struct's int16 but didn't make any changes to the report. The Ploopy was already using int16 temp variables as did your comment, so I read your edit as "oops yeah you can use 16 bits" assuming your temp variables were int8s before. Checking the edit history that's not the case, I'll poke at increasing the report size.

Yeah, in my [commit](https://github.com/morgannewellsun/qmk_firmware/commit/8c7e992761c143cb0c45b0cec094fa2912f14669), `usb_descriptor.c`, take a look at lines 177-179 (vertical) and 200-202 (horizontal). Just change both to:
```
                HID_RI_REPORT_SIZE(8, 0x10),
                HID_RI_LOGICAL_MINIMUM(16, -32767),
                HID_RI_LOGICAL_MAXIMUM(16,  32767),
```

IsaacElenbaas, 2024-08-20:
That is working, thank you again! A writeup ready before I even get back from work to look myself.

I threw in some "acceleration" (`#define SCALE(a) (((a >= 0) ? 1 : -1)*((a*a)/100+abs(a)))`) @ 300DPI and it feels great! Have to calculate in an int32 and crop still with the 60Hz reports (my system works without limiting them, but nice to accumulate for better speed measurement).

evan-goode, 2024-08-22:
Thanks @morgannewellsun! I applied your commit https://github.com/morgannewellsun/qmk_firmware/commit/8c7e992761c143cb0c45b0cec094fa2912f14669, the 60hz throttling (https://github.com/qmk/qmk_firmware/issues/17585#issuecomment-2287758137), and https://github.com/qmk/qmk_firmware/issues/17585#issuecomment-2298792294 to my Ploopy config https://github.com/evan-goode/qmk_firmware/tree/evan-goode/ploopy and it seems to work, in Windows at least. The `int16_t` report size is necessary if you don't want "wiggling" when scrolling quickly.

@IsaacElenbaas is there anything additional you had to do to get this working on Linux? I see frequent `REL_WHEEL_HI_RES` events in `evtest` but the events logged by `libinput debug-events` are less frequent and look like:

```
 event11  POINTER_SCROLL_WHEEL    +10.523s	vert 7.50/60.0* horiz 0.00/0.0 (wheel)
 event11  POINTER_SCROLL_WHEEL    +10.523s	vert 0.00/0.0 horiz -2.62/-21.0* (wheel)
```

Seems like libinput is waiting for the evdev events to accumulate before dispatching its `POINTER_SCROLL_WHEEL`.

diggit, 2024-08-22:
@evan-goode I can confirm this libinput behavior, it's same with non QMK device.

IsaacElenbaas, 2024-08-23:
@evan-goode I jinxed it by invoking the hardcoded deadzones :sob: Here's a bash script to patch libinput if you don't want it to buffer the start of a scroll - just follow the prompts and hit q if meson brings up a menu. The script needs Pacman, only tested on Arch.

[unsmoothen.txt](https://github.com/user-attachments/files/16726005/unsmoothen.txt)

You only need to patch libinput, but I left the kernel patch in there if anyone wants to use it for trackpads/tablets like I do. You don't need to restart X like it says to have the scroll deadzone removed, I assume not Wayland either.

morgannewellsun, 2024-08-30:
Update: I discovered that all you need to do to enable high-resolution scrolling is to modify the HID report descriptor. 

None of the other stuff involving feature report datastructures, report IDs, responding to HID get and set, etc... none of it is actually necessary to get high-resolution scrolling. It seems like that stuff is only needed if you want the _keyboard_ to know whether or not the OS supports high-resolution scrolling. To be honest, I have no idea what I'm doing when it comes to the USB stuff in QMK, so I'm not going to bother with adding that functionality.

Other than that... I cleaned things up, added some documentation and a couple example implementations for high-resolution drag scrolling, and opened a PR (my first ever 🤪). It branches off [0.26.0](https://github.com/qmk/qmk_firmware/releases/tag/0.26.0) so it should be much more up-to-date.

jfedor2, 2024-08-30:
> It seems like that stuff is only needed if you want the _keyboard_ to know whether or not the OS supports high-resolution scrolling.

The keyboard needs to know that though.

Otherwise it will scroll 120x times too fast if the OS didn't enable high-resolution scrolling. (Or whatever it thinks the multiplier is.)

morgannewellsun, 2024-08-30:
> The keyboard needs to know that though.
>
> Otherwise it will scroll 120x times too fast if the OS didn't enable high-resolution scrolling. (Or whatever it thinks the multiplier is.)

I totally agree, but unfortunately with the newest version of QMK, there seem to have been some changes to the way USB reports are handled... the code looks nothing like the stuff that I was hacking on, and to be honest I simply don't know how to work with any of it. Also, I'm pretty sure that implementing the ability for the keyboard to be aware of the OS's selection regarding high-resolution scrolling would need to be done independently for each platform, which is a ton of work, not to mention testing. It would also make this whole feature much more delicate and prone to breaking in the future. For my personal use, I'm OK with having the keyboard blindly assume that the OS has enabled high-resolution scrolling - maybe I'll have a key to toggle high-resolution scrolling on/off if it's really necessary.

morgannewellsun, 2024-09-01:
Keyboard awareness of host OS's choice regarding hires scrolling has been added to ChibiOS, LUFA, and VUSB, although only ChibiOS has been tested.

chrishoage, 2024-09-02:
I've been playing with the pull request this weekend and it works beautifully on both Linux and Windows. 

I'm really struggling to get it to function on macOS. The best I can tell is that macOS just does not support high resolution scrolling in the same way that Linux and Windows does. 

Can anyone confirm this? 

Or is this user error on my part and there is something necessary needed in order to make it work?

morgannewellsun, 2024-09-02:
> I'm really struggling to get it to function on macOS. The best I can tell is that macOS just does not support high resolution scrolling in the same way that Linux and Windows does.

I don't have any macOS devices, so I can't do any testing, but I'd be interested in learning more about what the issue is. 

When `POINTING_DEVICE_HIRES_SCROLL_ENABLE` is defined, is there no difference in behavior compared to when it's undefined? Or is there weird/different/erroneous behavior?

Also, if you'd be willing to do a quick test: with `POINTING_DEVICE_HIRES_SCROLL_ENABLE` defined and the keyboard connected to macOS, what does `is_hires_scroll_on()` return? (I think this function is can be used by including `quantum.h`.)

chrishoage, 2024-09-02:
> Also, if you'd be willing to do a quick test: with POINTING_DEVICE_HIRES_SCROLL_ENABLE defined and the keyboard connected to macOS, what does is_hires_scroll_on() return? (I think this function can be used by including quantum.h.)

Sorry, I realized after I wrote my message that it was ambiguous. I edited my comment right before you commented.

To summarize:  The os does not report "hires_scroll_on". My question is if there is something more that is required on macOS in order to enable this? (Or if it is just one more thing apple chooses not to implement)

morgannewellsun, 2024-09-02:
> To summarize: The os does not report "hires_scroll_on". My question is if there is something more that is required on macOS in order to enable this? (Or if it is just one more thing apple chooses not to implement)

I'm not 100% sure, but my guess is that it's just Apple not playing nice. If `is_hires_scroll_on()` is returning `false`, that means either (1) the OS is not sending resolution multiplier feature reports at all or (2) the OS is sending resolution multiplier feature reports that select the non-hires mode. 

I don't have any experience using macOS, but my gut feeling is that this isn't the kind of thing that Apple would let users tinker with via a setting... 

I'm not sure if there's third-party software that might be able to fix things - I've seen [MOS](https://mos.caldis.me/) mentioned in a few places. They claim to offer "smooth scrolling" but it seems like they use that term to refer to interpolation of inputs from a normal scroll wheel. But it also seems like there are some other settings provided - maybe it could help?

chrishoage, 2024-09-02:
> I don't have any experience using macOS

I'd like to be clear my post wasn't directed at you specifically, but more anyone in this thread (or coming from google in the future) had any ideas. 

I do greatly appreciate your help! 

> I'm not sure if there's third-party software that might be able to fix things - I've seen [MOS](https://mos.caldis.me/) mentioned in a few places.

I have tried:

- mos
  - Offers "smooth scrolling" not "hires" scrolling. E.g. animates between scroll values smoothly
- BetterMouse 
   - Does appear to support "hires" scrolling, but only for specific mice (e.g. [Logitech MX mice](https://better-mouse.com/#jp-carousel-2448) but I don't have one of these to understand how it actually behaves). 
 - SteerMouse
   - Seemed the most promising at first as it offered the ability to set fractional scroll units, but this broke many applications (or at least VS Code, stopped trying after that one). 

At this point my assumption is that qmk may need a driver (or something similar, in the way that BetterMouse may work) in order to get this behavior. 

All these applications just offer "smoothing/animation" over the "by-line" scroll wheel events, which is distinct from the hires scrolling Windows and Linux offers. 

My work is making me switch away from Linux to macOS and I'm remembering all the reasons I ditched it 5 years ago.

morgannewellsun, 2024-09-03:
@chrishoage By the way, what platform (chibiOS, LUFA, VUSB) did you test? Thanks!

chrishoage, 2024-09-03:
chibiOS on a Ploopy Adept

morgannewellsun, 2024-09-04:
I took a look at the latest `IOHIDFamily` from apple. I was a bit confused - some parts of the code, like `IOHIDPointing`, seem to support a resolution for the wheel, while other parts, like `IOHIDPointingDevice` don't. Not sure what to make of it.

Anyways, my current PR implements `SetRequest` (where the host tells the keyboard what the resolution multiplier is) but not `GetRequest` (where the keyboard tells the host what the resolution multiplier is). Maybe if I can implement that, it would make things work on macOS? But I'm kind of intimidated by the prospect of modifying the code that handles this stuff - it seems kind of hard to refactor, especially since QMK, to the best of my knowledge, didn't send or receive feature reports at all prior to this PR - only input and output reports.

justinjradi, 2024-09-04:
> Anyways, my current PR implements `SetRequest` (where the host tells the keyboard what the resolution multiplier is) but not `GetRequest` (where the keyboard tells the host what the resolution [...]

I think that george-norton started implementing getting/setting feature reports in https://github.com/george-norton/qmk_firmware/tree/multitouch_experiment (see #23243) if that's of any help. I should have clarified that the get_report stuff I was trying was just based soley off of my observations of the behavior of a Windows host. However I'm still confused and maybe just misinterpreted the packets since it looks like you've been able to get things working without it.

I wish I were able to see the report descriptor of a magic mouse and/or the traffic between a magic mouse and a Mac to see how viewport navigation is implemented. However, I don't own a Mac either.

chrishoage, 2024-09-04:
>  But I'm kind of intimidated by the prospect of modifying the code that handles this stuff - it seems kind of hard to refactor, especially since QMK, to the best of my knowledge, didn't send or receive feature reports at all prior to this PR - only input and output reports.

I don't think the lack of macOS support should hold up the open PR, it is great as is, and any improvements to macOS can be made later.

morgannewellsun, 2024-09-05:
@chrishoage If you end up having some time to do some testing, would you be able to try the following?

1. I added a bodge that should enable the keyboard to respond to `HID_REQ_GetReport`. You can find this on my [`develop-hires-scrolling-macos-fix`](https://github.com/morgannewellsun/qmk_firmware/tree/develop-hires-scrolling-macos-fix) branch. This might fix things for macOS...?
2. If (1) doesn't work: On my [`develop-hires-scrolling`](https://github.com/morgannewellsun/qmk_firmware/tree/develop-hires-scrolling) branch, try setting `POINTING_DEVICE_HIRES_SCROLL_EXPONENT` and seeing if that does anything on macOS. The "generic mouse descriptor" in apple's `IOHIDPointingDevice` has an unit exponent, so maybe this does something?

chrishoage, 2024-09-05:
> If (1) doesn't work: On my [develop-hires-scrolling](https://github.com/morgannewellsun/qmk_firmware/tree/develop-hires-scrolling) branch, try setting POINTING_DEVICE_HIRES_SCROLL_EXPONENT and [...]

Actually I had to add `POINTING_DEVICE_HIRES_SCROLL_EXPONENT` before any of this as when I didn't I get a compile error. 

When I compile with out it I get this error 

```
Compiling: tmk_core/protocol/usb_descriptor.c                                                      In file included from tmk_core/protocol/chibios/lufa_utils/LUFA/Drivers/USB/USB.h:38,
                 from tmk_core/protocol/usb_descriptor.h:46,
                 from tmk_core/protocol/usb_descriptor.c:41:
tmk_core/protocol/usb_descriptor.c:167:44: error: 'POINTING_DEVICE_HIRES_SCROLL_MULTIPLIER' undeclared here (not in a function); did you mean 'POINTING_DEVICE_HIRES_SCROLL_ENABLE'?
  167 |                 HID_RI_PHYSICAL_MAXIMUM(8, POINTING_DEVICE_HIRES_SCROLL_MULTIPLIER),
      |                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./lib/lufa/LUFA/Drivers/USB/Class/Common/HIDReportData.h:67:76: note: in definition of macro '_HID_RI_ENCODE_8'
   67 |                         #define _HID_RI_ENCODE_8(Data)                  , (Data & 0xFF)
      |                                                                            ^~~~
tmk_core/protocol/chibios/lufa_utils/LUFA/Drivers/USB/USB.h:22:31: note: in expansion of macro 'CONCAT'
   22 | #define CONCAT_EXPANDED(x, y) CONCAT(x, y)
      |                               ^~~~~~
./lib/lufa/LUFA/Drivers/USB/Class/Common/HIDReportData.h:70:73: note: in expansion of macro 'CONCAT_EXPANDED'
   70 |                         #define _HID_RI_ENCODE(DataBits, ...)           CONCAT_EXPANDED(_HID_RI_ENCODE_, DataBits(__VA_ARGS__))
      |                                                                         ^~~~~~~~~~~~~~~
./lib/lufa/LUFA/Drivers/USB/Class/Common/HIDReportData.h:72:115: note: in expansion of macro '_HID_RI_ENCODE'
   72 |                         #define _HID_RI_ENTRY(Type, Tag, DataBits, ...) (Type | Tag | HID_RI_DATA_BITS(DataBits)) _HID_RI_ENCODE(DataBits, (__VA_ARGS__))
      |                                                                                                                   ^~~~~~~~~~~~~~
./lib/lufa/LUFA/Drivers/USB/Class/Common/HIDReportData.h:110:73: note: in expansion of macro '_HID_RI_ENTRY'
  110 |                         #define HID_RI_PHYSICAL_MAXIMUM(DataBits, ...)  _HID_RI_ENTRY(HID_RI_TYPE_GLOBAL, 0x40, DataBits, __VA_ARGS__)
      |                                                                         ^~~~~~~~~~~~~
tmk_core/protocol/usb_descriptor.c:167:17: note: in expansion of macro 'HID_RI_PHYSICAL_MAXIMUM'
  167 |                 HID_RI_PHYSICAL_MAXIMUM(8, POINTING_DEVICE_HIRES_SCROLL_MULTIPLIER),
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
tmk_core/protocol/usb_descriptor.c:168:41: error: 'POINTING_DEVICE_HIRES_SCROLL_EXPONENT' undeclared here (not in a function); did you mean 'POINTING_DEVICE_HIRES_SCROLL_ENABLE'?
  168 |                 HID_RI_UNIT_EXPONENT(8, POINTING_DEVICE_HIRES_SCROLL_EXPONENT),
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./lib/lufa/LUFA/Drivers/USB/Class/Common/HIDReportData.h:67:76: note: in definition of macro '_HID_RI_ENCODE_8'
   67 |                         #define _HID_RI_ENCODE_8(Data)                  , (Data & 0xFF)
      |                                                                            ^~~~
tmk_core/protocol/chibios/lufa_utils/LUFA/Drivers/USB/USB.h:22:31: note: in expansion of macro 'CONCAT'
   22 | #define CONCAT_EXPANDED(x, y) CONCAT(x, y)
      |                               ^~~~~~
./lib/lufa/LUFA/Drivers/USB/Class/Common/HIDReportData.h:70:73: note: in expansion of macro 'CONCAT_EXPANDED'
   70 |                         #define _HID_RI_ENCODE(DataBits, ...)           CONCAT_EXPANDED(_HID_RI_ENCODE_, DataBits(__VA_ARGS__))
      |                                                                         ^~~~~~~~~~~~~~~
./lib/lufa/LUFA/Drivers/USB/Class/Common/HIDReportData.h:72:115: note: in expansion of macro '_HID_RI_ENCODE'
   72 |                         #define _HID_RI_ENTRY(Type, Tag, DataBits, ...) (Type | Tag | HID_RI_DATA_BITS(DataBits)) _HID_RI_ENCODE(DataBits, (__VA_ARGS__))
      |                                                                                                                   ^~~~~~~~~~~~~~
./lib/lufa/LUFA/Drivers/USB/Class/Common/HIDReportData.h:111:73: note: in expansion of macro '_HID_RI_ENTRY'
  111 |                         #define HID_RI_UNIT_EXPONENT(DataBits, ...)     _HID_RI_ENTRY(HID_RI_TYPE_GLOBAL, 0x50, DataBits, __VA_ARGS__)
      |                                                                         ^~~~~~~~~~~~~
tmk_core/protocol/usb_descriptor.c:168:17: note: in expansion of macro 'HID_RI_UNIT_EXPONENT'
  168 |                 HID_RI_UNIT_EXPONENT(8, POINTING_DEVICE_HIRES_SCROLL_EXPONENT),
      |                 ^~~~~~~~~~~~~~~~~~~~
 [ERRORS]
 |
 |
 |
make: *** [builddefs/common_rules.mk:373: .build/obj_ploopyco_madromys_rev1_001_chrishoage/usb_descriptor.o] Error 1
```

The only way for me to get it to compile has been 

```
#define POINTING_DEVICE_HIRES_SCROLL_MULTIPLIER 120
#define POINTING_DEVICE_HIRES_SCROLL_EXPONENT 1 
```

Omitting both lead to the error above. 

> I added a bodge that should enable the keyboard to respond to HID_REQ_GetReport. You can find this on my [develop-hires-scrolling-macos-fix](https://github.com/morgannewellsun/qmk_firmware/tree/develop-hires-scrolling-macos-fix) branch. This might fix things for macOS...?

I'll cherry pick this to my scratch branch I've been hacking on things and try it out!

morgannewellsun, 2024-09-05:
Regarding the undefined macro errors - I realized that I put the preprocessor directives for the default values of those macros in the wrong .h file. This is located in commit [225fc629926a164a617ebd098ab3d612f16c2cf2](https://github.com/morgannewellsun/qmk_firmware/commit/225fc629926a164a617ebd098ab3d612f16c2cf2)

chrishoage, 2024-09-05:
The best I can tell `develop-hires-scrolling-macos-fix` regressed all hires scrolling. 

Please forgive my testing methodology, I'm hesitant to install qmk on my work computer in order to use `qmk console` 

In ploopyco.c I've taken your hires dragscroll example and done this (with `hires_scroll_task_kb` being your dragscroll example with small modifications and `drag_scroll_task_kb` being the original ploopy dragscroll)

```
report_mouse_t pointing_device_task_kb(report_mouse_t mouse_report) {
    if (is_drag_scroll) {
        if (is_hires_scroll_on()) {
            return hires_scroll_task_kb(mouse_report);
        } else {
            // return drag_scroll_task_kb(mouse_report);
        }
    }

    return pointing_device_task_user(mouse_report);
}
```

with `develop-hires-scrolling-macos-fix` and the fallback dragscroll I get no dragscroll on Linux or macOS (don't have the chance to test windows today) 

with a fresh pull of `develop-hires-scrolling` and the above code I get dragscroll on Linux (indicating `is_hires_scroll_on`) and no dragscroll on macos (indicating `is_hires_scroll_on` is returning false) 

I'm happy to test anything further you wish!

morgannewellsun, 2024-09-05:
I was able to replicate the issue on Windows. I'm not sure how that slipped past me - I think I failed to flash the keyboard I was testing with, but thought I flashed it...

Anyways, I have rewritten history on the [`develop-hires-scrolling-macos-fix`](https://github.com/morgannewellsun/qmk_firmware/tree/develop-hires-scrolling-macos-fix) branch on my fork. The new bodge accomplishes the same thing as the earlier attempt (implement code to respond to HID_REQ_GetRequest for the wheel resolution feature report, so we can see if that's what macOS wants from us). But this time, I've properly confirmed that it works, at least on Windows. 

_**However, it also only works with ChibiOS devices where `MOUSE_SHARED_EP` is set.**_ 

If that's the case for your device, I'd really appreciate if you could give it a try when you get a chance.

[4 more replies]