qmk/qmk_firmware #17589: [Bug] led_min/led_max issue with rgb matrix raindrops effect

lokher, 2022-07-07:
## Describe the Bug
led_min and led_max  as parameter of rgb_matrix_indicators_advanced_kb()/rgb_matrix_indicators_advanced_user()  are always same at each invoke when running raindrops effect. I dig into the code, look like Line 31~35 never have chance to run since this function always return false at Line 28, which lead to no increment of iter
https://github.com/qmk/qmk_firmware/blob/e7931289918221081cbe2a7ea5df27a5d86324db/quantum/rgb_matrix/animations/raindrops_anim.h#L22-L36 

In addition, RGB_MATRIX_INDICATOR_SET_COLOR can only set first 1/5~1/4 part of LEDs.

## System Information

**Keyboard:** 
**Revision (if applicable):** 
**Operating system:** 
**`qmk doctor` output:** 
```
(Paste output here)
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

filterpaper, 2022-07-13:
Line 31~35 is ran once on init to light up every LED with random hues.

Line 23~29 will run on every cycle, and on interval tick, randomly change one LED to another random color.

Checked this effect again and it does change one LED on every tick. However the effect can be too subtle if user set saturation (in eeprom) is too high.

lokher, 2022-07-14:
I think you didn't get my point, let me describe more details. 
Rgb matrix doesn't render whole keyboard leds in a single loop , it renders `RGB_MATRIX_LED_PROCESS_LIMIT` number leds,  which define in rgb_matrix.h:

https://github.com/qmk/qmk_firmware/blob/99e9e1b8e70b7b6a0cc03a6672f2512b472fb650/quantum/rgb_matrix/rgb_matrix.h#L49-L51

Generally, if there are 100 leds in the keyboard, rgb matrix renders led 0 to 20 in the 1s tloop, `led_min` is set to 0 and `led_max is` set to 20 , then increase `params->iter` ,  then renders led 21 to 40 in the 2nd loop, `led_min` is set to 21 and `led_max is` set to 40, then 3rd loop.... 

`params->iter`  increase in the diffinition of `RGB_MATRIX_USE_LIMITS`

https://github.com/qmk/qmk_firmware/blob/99e9e1b8e70b7b6a0cc03a6672f2512b472fb650/quantum/rgb_matrix/rgb_matrix.h#L63-L66

`RGB_MATRIX_USE_LIMITS(led_min, led_max)` is run in other rgb matrix effect, but not in raindrop effect, so that `led_min` and `led_max`  is always 0 and 20 on 100 leds keyboard.

If you use   `RGB_MATRIX_INDICATOR_SET_COLOR` to set the led,  it work only when target index is between `led_min` and `led_max` , setting led which index is greater than 20 won't work.

https://github.com/qmk/qmk_firmware/blob/99e9e1b8e70b7b6a0cc03a6672f2512b472fb650/quantum/rgb_matrix/rgb_matrix.h#L83-L86

filterpaper, 2022-07-14:
I missed the part about breaking `rgb_matrix_indicators_advanced_user()`. Can you test the linked PR and confirm it is working?

PR #17678 (2022-07-14): [Core] Fix LED limit loop

## Description

* Ensure `RGB_MATRIX_USE_LIMITS` is called on every iteration
* Use the smaller `random8()` function

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #17589

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have rea [... cut]
PR #17679 (2022-07-14): Add LED limits call

## Description

Add `RGB_MATRIX_USE_LIMITS` to avoid breaking `rgb_matrix_indicators_advanced_*()`

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #17589

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist**  [... cut]
closed 2023-11-01 by zvecr