qmk/qmk_firmware #17605: STM32F103 Bluepill USB disconnects and not detected after flashing

djask, 2022-07-09:
Have a issue with a standard STM32F103 bluepill and would like to double-check if it is a problem with the chip or with the code (as I do not have any other bluepill boards on hand to test).

Basically I can run stm32duino on it flashed from UART and also able to upload .bin files. However, I have tried multiple provided bluepill test firmwares (like the handwired/onekey) and also compiled myself and they do not show up as usb HID after the flash. I don't think it is a problem with the pinout and I also measured the 1.5kohm resistor as expected value.

Is there anything else that needs to be specified in chibios configs when building?

Appreciate any help.

fauxpark, 2022-07-09:
The STM32duino bootloader will not boot straight into the application code directly after flashing, you need to replug it.

djask, 2022-07-09:
I have tried unplugging/replugging the USB and it gets recognized for about 1 second ( I think this is the stmduino32 part) then disconnects immediately after and no USB device shows up.

fauxpark, 2022-07-09:
Both the jumpers should be set to 0.
Also, what bootloader file did you flash?

djask, 2022-07-09:
Yes, I have both jumpers on 0. Flashed the generic_boot20_pc13 file from the repo below, tried pretty much all the pc13 loaders.
https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/bootloader_only_binaries/generic_boot20_pc13.bin

The only firmware which I have gotten detected after flash is the keyberon one so I believe the chip is still somewhat working, but I just can't get anything qmk based to be detected in OS after flashing.

fauxpark, 2022-07-09:
Did you follow this guide? https://docs.qmk.fm/#/isp_flashing_guide?id=flashing-stm32duino-bootloader

djask, 2022-07-09:
I flashed the bootloader with serial UART as I do not have a STLink. Does this make a difference when loading applications?

fauxpark, 2022-07-09:
I don't know, the method above is what worked for me (which is why I wrote it).

sigprof, 2022-07-09:
Exactly what MCU does your bluepill have (there should be some markings visible on the chip, but sometimes the text is hard to see)?

By default QMK assumes the STM32F103C**B**T6 chip (128K flash), but that code also usually works with STM32F103C**8**T6 (nominally 64K flash), because genuine STM32F103x8 chips tend to actually have 128K flash; but some xxx32F103C8 clone chips may have only the documented 64K flash and don't work without reconfiguring the flash size in the EEPROM emulation code. And now there are some “bluepills” with STM32F103C6T6 chips (32K flash) — if you got one, you will need to do even more low-level reconfiguration (see #15297).

djask, 2022-07-09:
That's what is confusing me because I got this supposedly listed as a C6T6, but it shows up as a high density chip with 256k flash and the marking on it is a C8T6. 

I'm trying the instructions in the linked issue, with following in rules.mk (not sure if I still add BOOTLOADER = xxx as you said to remove it when adding the other information)
```
OPT_DEFS += -DBOOTLOADER_STM32DUINO
MCU_LDSCRIPT = STM32F103x6_stm32duino_bootloader
BOARD = STM32_F103_STM32DUINO
STM32_BOOTLOADER_ADDRESS = 0x80000000
DFU_ARGS = -d 1EAF:0003 -a 2 -R
DFU_SUFFIX_ARGS = -v 1EAF -p 0003

MCU = STM32F103
BOOTLOADER = stm32duino
```

Also had some GPIO errors in compiling.
```
Compiling: platforms/chibios/boards/STM32_F103_STM32DUINO/board/board.c                            platforms/chibios/boards/STM32_F103_STM32DUINO/board/board.c:31:3: error: extra brace group at end of initializer
   31 |   {VAL_GPIOEODR, VAL_GPIOECRL, VAL_GPIOECRH},
      |   ^
platforms/chibios/boards/STM32_F103_STM32DUINO/board/board.c:31:3: note: (near initialization for 'pal_default_config')
platforms/chibios/boards/STM32_F103_STM32DUINO/board/board.c:31:3: error: excess elements in struct initializer [-Werror]
platforms/chibios/boards/STM32_F103_STM32DUINO/board/board.c:31:3: note: (near initialization for 'pal_default_config')
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
```

Temporarily fixed it by commenting this out, but don't think I should be.
`//  {VAL_GPIOEODR, VAL_GPIOECRL, VAL_GPIOECRH},`

So now I have a compiled file for C6T6 ( I think) but it still doesn't work when I try to flash it. No USB HID device detected still.  The firmware is smaller by around 20 bytes though.

sigprof, 2022-07-10:
> That's what is confusing me because I got this supposedly listed as a C6T6, but it shows up as a high density chip with 256k flash and the marking on it is a C8T6.

Where exactly do you see that “high density chip with 256k flash”? A real STM32F103C6 chip should have 32K flash and be classified as “low density”, while STM32F103C8 (64K flash) and STM32F103CB (128K flash) should be classified as “medium density”. There is also STM32F103C4 (16K flash, also “low density”), and that is the exhaustive list of genuine STM32F103 chips in the LQFP48 package — although there are also “high density” and “XL density” chips in that line, they are available only in some larger packages. However, some other manufacturers produce somewhat compatible clones that might have larger flash sizes in LQFP48.

Here is what `st-info --probe` prints for the chip marked “STM32F103C8T6” that I happen to have:
```
  flash:      131072 (pagesize: 1024)
  sram:       20480
  chipid:     0x0410
  descr:      F1xx Medium-density
```
(for some reason it detects 128K flash, even though officially the chip should have 64K, but on this particular chip the extra 64K flash is available).

STM32F103C6T7A:
```
  flash:      32768 (pagesize: 1024)
  sram:       10240
  chipid:     0x0412
  descr:      F1 Low-density device
```

GD32F103CBT6:
```
  flash:      131072 (pagesize: 1024)
  sram:       20480
  chipid:     0x0410
  descr:      F1xx Medium-density
```

GD32F303CCT6 (despite the “F303” name, this chip is mostly a clone of STM32F103, and is one example of a clone in the LQFP48 package that provides a larger flash size than could be available from ST):
```
  flash:      262144 (pagesize: 2048)
  sram:       65536
  chipid:     0x0414
  descr:      F1xx High-density
```

djask, 2022-07-10:
Thanks for the info, judging from the chip ID, looks like I have the last one on your list?

ST32cube says it has 512 on the side, but all UART flashers seem to report 256k. Looks like none of the programs I have used with this have reported the correct chip then...

[image]

sigprof, 2022-07-10:
CH32F103C8T6 (do not buy this clone — it's really bad, the flash is so slow that even `st-flash` needed patching to be able to write anything, and then it turned out that USB is unreliable even in stm32duino):
```
  flash:      65536 (pagesize: 1024)
  sram:       20480
  chipid:     0x0410
  descr:      F1xx Medium-density
```

------

There is also the [Bluepill Diagnostics](https://mecrisp-stellaris-folkdoc.sourceforge.io/bluepill-diags-v1.640.html) program, which you could flash **instead of** stm32duino for testing. However, it claims that my STM32F103C8T6 chip is fake, because it reports 128K flash size:
```
STM32F103C8 authentication FAILED one or more tests:
----------------------------------------------------
FAIL - Declared flash not 65536
PASS - DBGMCU_IDCODE is NOT readable without SWD/Jtag connected
PASS - Second 64KB flash block verified
PASS - JDEC manufacturer id IS STMicroelectronics
```
Other test results for STM32F103C8T6 (`f`, `d`, `j`):
```
131072 flash is declared in the Flash size register at 0x1FFFF7E0

DBGMCU_IDCODE [@ 0xE0042000] = 0x00000000

Jdec Continuation Code: 0x00
Jdec Identity Code: 0x20
JDEC manufacturer id: STMicroelectronics
```
or, when using SWDCOM:
```
DBGMCU_IDCODE [@ 0xE0042000] = 0x20036410
```

Unfortunately, that program is too large to be used on STM32F103C6 (it's about 60K). It also does not seem to work properly on GD32F103CBT6 and GD32F303CCT6 (the USB device is not recognized).

I managed to launch Bluepill Diagnostics on GD32F303CCT6 using SWDCOM, and here is what it prints on `f`, `d`, `j`:
```
262144 flash is declared in the Flash size register at 0x1FFFF7E0

DBGMCU_IDCODE [@ 0xE0042000] = 0x17010414

Jdec Continuation Code: 0x07
Jdec Identity Code: 0x51
GigaDevice Semiconductor
```

sigprof, 2022-07-10:
> ST32cube says it has 512 on the side, but all UART flashers seem to report 256k.

Not sure whether it actually detect something, or just assumes 512K for a high density chip, but as it shows that `0x0414` ID, looks like your chip is definitely not a genuine STM32F103 (as I said, there is no genuine STM32F103 chip in the LQFP48 package with more than 128K flash). Unfortunately, such things can be really common when buying cheap things from shady sellers (and looks like they went as far as actually mislabeling the chip as STM32; or is it really something other written on the chip?).

You may try `qmk flash -kb handwired/onekey/bluepill -km default -e EEPROM_DRIVER=transient` — this code work for me on GD32F303CCT6 even without the other configuration for the different flash size. But you may actually have some other clone that is too incompatible to be supported that way. Without knowing what's the real chip behind those fake markings it's hard to guess what could be done here.

djask, 2022-07-10:
All good. I tried with the transient EEPROM driver which didn't work. I also tried flashing the diagnostics program, and it gives a "USB malfunction" after I reload it to try run it. So it does seem like this chip is a hit or miss. 

Also, yes I double checked the marking (as well as that 3 dimples thing some people mention) and it does have STM32 markings printed on it. 

Anyways the information here is good to know for future. Thanks.

sigprof, 2022-07-10:
If you can, try to contact the seller or marketplace and get your money back, because the chips are definitely not what was advertised. In some other cases of mislabeling it could be possible to get something useful out of the chips (e.g., someone got what was probably GD32F303 mislabeled as STM32F303 from AliExpress — they at least worked with the code compiled for STM32F103xC), but your case looks hopeless (no idea what the chips actually are).

closed 2022-07-10 by sigprof

xisang233, 2023-01-23:
STM32F103C6T6 is not a good choice for qmk firmware. But you can check this issue 
https://github.com/qmk/qmk_firmware/issues/15297 and change something to make it work.