qmk/qmk_firmware #17618: How to use the encoder whose pulse number and positioning number are 1:2?

jeromeli002, 2022-07-11:
这种脉冲数和定位数是1：2的编码器怎么用？直接使用会是转2位才会输出一次，要怎么改让他转一位输出一次。qmk内有参考的键盘吗？
How to use the encoder whose pulse number and positioning number are 1:2? If it is used directly, it will only output once by turning 2 bits. How to change it to turn one bit and output once. Is there a reference keyboard in qmk?

[image]

fauxpark, 2022-07-11:
Change the encoder resolution. This is in the docs.

closed 2022-07-11 by fauxpark

jeromeli002, 2022-07-11:
#define ENCODER_RESOLUTION 4   ？？
this? ? It didn't work, tried it

jeromeli002, 2022-07-11:
I didn't change it right? ? how much to change

fauxpark, 2022-07-11:
It is usually either 2 or 4.

jeromeli002, 2022-07-11:
1~6 have been tried, it seems useless, when the volume is changed, the volume is 2 or 4 once

fauxpark, 2022-07-11:
If changing the encoder resolution isn't doing anything at all, then it's not being applied.

Otherwise, check that it's the same for other keycodes (eg. `KC_A` and `KC_B`). If that works, you might need to add `#define TAP_CODE_DELAY 10`.

einalex, 2024-01-27:
This question is not solved.

The docs say:
```
Additionally, the resolution, which defines how many pulses the encoder registers between each detent, can be defined with:
#define ENCODER_RESOLUTION 4
```

This implies that for 1:2 pulse:detent encoders a setting of 0.5 is appropriate, as there are 0.5 pulses per detent. This however makes no sense, since that variable is saved as uint8_t and can therefore not store floats.

The current code is unable to interpret wave patterns like these:
https://tech.alpsalpine.com/cms.media/product_detail_fig_ec11_c_21_en_0959753c2b.gif

This limitation should either be fixed or made clear in the documentation.

sigprof, 2024-01-27:
> `Additionally, the resolution, which defines how many pulses the encoder registers between each detent, can be defined with: #define ENCODER_RESOLUTION 4`

That description is wrong — it should say “transitions” instead of “pulses” (the encoder data sheets consider the “pulse” to be the whole cycle on both data lines — so 4 transitions).

For your Alps encoder the proper value of `ENCODER_RESOLUTION` should be 2; most cheap clones need 4.  For some cheap encoders setting `#define ENCODER_DEFAULT_POS 3` may be useful (that option can make the encoder work slightly more reliably even if some transitions are not recognized properly due to excessive bounce), but that option can work only with `#define ENCODER_RESOLUTION 4`.

And if you have problems, you really should map the encoder rotation to some regular keys and see what actually happens, because the media keys may have other problems:
- In some cases media keys need a larger value of `TAP_CODE_DELAY` than normal keys due to how the OS handles them.  (Also note that the encoder code has a separate `ENCODER_MAP_KEY_DELAY` setting that you can use if you don't want to touch `TAP_CODE_DELAY`.)  Although the usual problem with the delay being too short is that the tap is not registered at all.
- At least Windows apparently hardcodes the volume step size for media keys to 2%, not 1%; you can map some regular keys to `KC_VOLU` and `KC_VOLD` to check that behavior without using the encoder.

einalex, 2024-01-27:
I realized. I run my encoder with ENCODER_RESOLUTION 2, using KC_LEFT, KC_RIGHT as output. Unfortunately, that does not produce output.
If I run it with ENCODER_RESOLUTION 1, I get alternating left, right keys, regardless of direction.

That seems to me like it's associating some of the H/L L/H transitions to the wrong direction.

sigprof, 2024-01-27:
You may need to check the hardware — the behavior that you describe may happen if one of A/B encoder pins is stuck for any reason instead of changing the state properly.

einalex, 2024-01-27:
yes, I did after debugging, found the culprit and fixed it.