qmk/qmk_firmware #17623: TT(layer) momentary working. toggle not working

Abh15h3k, 2022-07-11:
I'm trying to get TT (toggle) working on my keymap. but only the momentary layer is working. i've tried setting TAPPING_TOGGLE in both `config.h` files but still toggle does not work.

Here are the relevant files.
let me know what stupid thing i did (or missed)

[keyboard/rules.mk](https://pastebin.com/XZDzKug3)
[keyboard/config.h](https://pastebin.com/FgP6WKMg)
[keyboard/ttc.h](https://pastebin.com/245w3060)
[keyboard/ttc.c](https://pastebin.com/ePubgUKN)
[keymap/keymap.json](https://pastebin.com/DAWT2rpg)
[keymap/config.h](https://pastebin.com/itwrvYTM)

`qmk doctor`
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /home/abhishek/Documents/git/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.17.5
⚠ Git has unstashed/uncommitted changes.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.1.0
Ψ Found avr-gcc version 12.1.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

ozelentok, 2022-07-15:
I have had this issue for a few weeks as well, TT acting like MO and ignoring TAPPING_TOGGLE/TAPPING_TERM.
my qmk setup and versions are the same as yours.
I ended up enabling the console and added a print before the following line:
https://github.com/qmk/qmk_firmware/blob/09b0d3e0ed3a9ad897558421d13f59f40831c44c/quantum/action.c#L596
suddenly the issue disappeared and tapping worked so I suspect it's an optimization related bug.
changing the optimization level from "s" to "2" also solved it.
https://github.com/qmk/qmk_firmware/blob/09b0d3e0ed3a9ad897558421d13f59f40831c44c/builddefs/common_rules.mk#L49
I haven't gotten the chance to downgrade avr-gcc to 8.x but it's probably the correct solution for now

Abh15h3k, 2022-07-15:
yeah. using avr-gcc 9 resolves the issues.

marking as closed as avr-gcc 8 is the recommended compiler

closed 2022-07-15 by Abh15h3k

jypma, 2022-08-03:
Can confirm that `OPT ?= 2` also fixed tapping toggle (TT) functionality for me (On Arch linux, avr-gcc 12.1.0-1) Is there a related bug in avr-gcc we can track for this apparent regression?

jypma, 2022-08-03:
Unfortunately, `OPT ?= 2` seems to break split functionality, at least in my case...

kevinstadler, 2022-12-04:
I am also having troubles getting TT functionality to work, the key simply acts like a MO key but ignores multiple tapping. I've tried with both `avr-gcc (Homebrew AVR GCC 9.4.0) 9.4.0` and `avr-gcc (Homebrew AVR GCC 8.5.0) 8.5.0`, with and without `OPT ?= 2`, but neither one works. My `qmk doctor`:
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /Users/k/qmk_firmware
Ψ Detected macOS 10.15.7 (Intel).
Ψ Git branch: master
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2022-11-05 10:55:42 -0700 (e3a97eb1c8) -- [Keyboard] Add Work Louder Numpad (#18912)
Ψ - Latest upstream/master: 2022-12-05 02:53:08 +0800 (4d1b1bc534) -- [Keyboard] add kbdpadmk3 (#18795)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2022-11-05 10:55:42 -0700 (e3a97eb1c8) -- [Keyboard] Add Work Louder Numpad (#18912)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```