qmk/qmk_firmware #17676: I am having a problem compiling on a new computer. How can I solve this problem?

1219008126, 2022-07-14:
☒ Invalid API data: kb: keyboard_folder: 'kb' is not valid under any of the given schemas

tzarc, 2022-07-14:
Don't call your keyboard 'kb'.

1219008126, 2022-07-14:
I've tried other names like that

tzarc, 2022-07-14:
Then I'd imagine something else is wrong.
Try `qmk new-keyboard` and follow the prompts.

1219008126, 2022-07-14:
[Gorilla65.zip](https://github.com/qmk/qmk_firmware/files/9108319/Gorilla65.zip)
I compiled it with a reference to a keyboard I made before, so I think it should be right

tzarc, 2022-07-14:
Uppercase names aren't allowed.

dvDevel, 2023-04-27:
@tzarc Absolutely right.
I have a source having uppercase.
I encountered an error as an Invalid keyboard while I had to try to compile.
After renaming the folder the disappeared.