qmk/qmk_firmware #17689: smt32 F103 failed to flash qmk

jeromeli002, 2022-07-15:
https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/bootloader_only_binaries/generic_boot20_pc13.bin
[image]
[image]
[image]
The first time I used stm32f103, I flashed the bootloader provided in qmk, and the device displayed maple003, but I couldn't flash the qmk firmware. Is there any modification?

sigprof, 2022-07-15:
Apparently the flashing tool detected “Target: STM32F1_Low-density_32K”, which means that you have the STM32F103x6 chip (or maybe even some non-ST clone of that chip) with only 32K flash. The stm32duino bootloader reserves 8K for itself, leaving only 24K (24576 bytes) for the firmware, but apparently your firmware is larger than that, therefore the flashing process fails after crossing the 24K boundary.

If you really want to use this chip, you would need to get the firmware size down, probably by disabling some features; but you may end up wasting a lot of time on it without getting a usable result in the end.  Note that the support for STM32F103x6 chips may not be 100% functional in current QMK versions (e.g., see #15297).  Although the ChibiOS fix for USB support should be included now, the EEPROM emulation code would probably need to be turned off (`EEPROM_DRIVER = transient`); it could be possible to reconfigure it, but then the usable amount of flash will decrease even more.  Also the firmware size check for STM32 chips does not take the emulated EEPROM area into account yet, so you may get firmware corruption when the emulated EEPROM overwrites a part of firmware code.

myst729, 2022-07-22:
How about extended flash, say, W25Q128. Does that possibly work?

sigprof, 2022-07-30:
Can you test the STM32F103C6 support code from #17853?

Note that you need to use at least `board.h` and the bootloader-related section of `rules.mk` from the `handwired/onekey/bluepill_f103c6` keyboard in your custom code, because the bootloader parameters also specify `MCU_LDSCRIPT = STM32F103x6_stm32duino_bootloader`. Also use `EEPROM_DRIVER = transient` for now, because the proper firmware size check which takes the EEPROM backing store into account is not implemented yet, and with only 24 KiB of flash available for QMK it would be easy to get the backing store overlapping the firmware code.

myst729, 2022-10-23:
It works! Though I have to carefully calculate how many bytes each feature takes, and turn off some heavy ones, e.g. RGB matrix. VIA support is the heaviest among them.

closed 2023-02-21 by github-actions[bot]