qmk/qmk_firmware #17720: [Bug] OLED over I2C on RP2040 causes (split?) keyboard slowdown and crash

CodyMathis123, 2022-07-19:
## Describe the Bug

When OLED screens are enabled and used on a KB2040 (Assuming this affects other RP2040 as well), the keyboard becomes slow, missing keystrokes, and eventually unresponsive. 

## System Information

**Keyboard:** Corne / CRKBD
**Revision (if applicable):** v3
**Operating system:** Windows 10
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: C:/Users/<username>/source/keebs/rp2040/qmk_firmware
Ψ Detected Windows 10 (10.0.19043).
Ψ Git branch: develop
Ψ Repo version: 0.11.53
⚠ Git has unstashed/uncommitted changes.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

I am not sure if this only impacts split keyboards or not.

CodyMathis123, 2022-07-19:
@daskygit 

https://github.com/CodyMathis123/qmk_firmware/tree/develop/keyboards/crkbd/keymaps/codymathis123

Here you go!

Jpe230, 2022-07-19:
I have experienced this behavior in boards (CRKBDs and TidBit) with either the KB2040 and PM-RP2040.

In my case, it was due to i2c errors thanks to some poorly solder joints. 

I suggest you to enable debug and see if any failed transaction occur :)

CodyMathis123, 2022-07-19:
I can reproduce this across two separate keyboards, and it does not happen on an elite-c, for example. 

I will look at enabling debug though!

CodyMathis123, 2022-07-19:
Set 

CONSOLE_ENABLE = yes

in rules.mk 

and added the below to my keymap.c

```
void keyboard_post_init_user(void) {
  // Customise these values to desired behaviour
  debug_enable=true;
  debug_matrix=true;
  //debug_keyboard=true;
  //debug_mouse=true;
}
```

The attached file is the output from QMK toolbox. 

[crkbd-debug.txt](https://github.com/qmk/qmk_firmware/files/9142696/crkbd-debug.txt)

Jpe230, 2022-07-19:
Interesting...

Does the bug present itself when using the other half as master? Or even when only using just 1 half without the other connected?

CodyMathis123, 2022-07-19:
It is always the secondary side that stops responding to keypresses. So I can plug in either side, and the secondary will have the issue. 

The 'main' half always types as expected.

amadea-system, 2022-07-19:
I have also experienced this issue on my crkbd w/ KB2040.
And just like @CodyMathis123, it only ever happens on the secondary side. 
When I swap which side is Primary/Secondary, the issue always moves over to the 'new' Secondary side.
The issue does not seem to happen immediately upon power up, but a random amount of time later. Anywhere from a minute to an hour I would say.

When it happens on my keyboard I notice the following symptoms, all exclusively on the Secondary side.
- Keypresses are still recognized, but not well.
- The OLED stops updating.
- RGB Matrix updates at 1-2 fps.

Jpe230, 2022-07-20:
Just tested with both PM-RP2040 and KB2040, either as a slave or master, and in all cases the slave dies the same.

I'm using this to reliably crash my slave half:
```
void render_slave_oled() {
	static int current_block = 0;

	current_block = (current_block + 1) % 528;

	char white_bg[528] = {0};
	memset(white_bg, 0, 528);
	memset(white_bg, 0xff, current_block);
	oled_write_raw(white_bg, 528);
}
```

@CodyMathis123  @amadea-system, can you confirm that this is the exact behavior as yours?

https://user-images.githubusercontent.com/6202305/179901668-a6dc436c-df44-47ab-a602-b18801d1d01f.mp4

(KB2040 as Master and PM-RP2040 as slave)

tzarc, 2022-07-20:
Shouldn't they be 512 bytes? `(128*32/8)=512`?
Also perhaps try making the `white_bg` a `static` -- if it's on the stack instead of the heap the compiler sometimes has a fit.

Jpe230, 2022-07-20:
Oh shoot! My maths did wrong to me.

But sadly even by changing the size and making it static still locks it up hmmm

KarlK90, 2022-07-20:
@Jpe230 @CodyMathis123 @daskygit I'm not sure on the root cause of the problem (yet), but I was able to reproduce the lockup easily. The cause was an I2C IRQ tail chaining of the TX FIFO EMPTY event, after a transmission was done. My fix is to disable all IRQ sources after a successful transmission (just like in the [zephyr driver](https://elixir.bootlin.com/zephyr/latest/source/drivers/i2c/i2c_dw.c#L188)).

Changes are already in a PR -> https://github.com/ChibiOS/ChibiOS-Contrib/pull/329

jaygreco, 2022-07-20:
I was going to open an issue after discovering and digging into the same thing -- I just wanted to add my anecdata to the mix. I have seen in most frequently on the slave side while testing the SNAP, but have also seen the same lockups on the master side albeit with a lower rate of occurrence. The problem seems specific to RP2040 as I haven't seen any I2C errors with AVR on the same board.

KarlK90, 2022-07-21:
@jaygreco could you test the ChibiOS-contrib PR and see if this fixes your lockups as well?

jaygreco, 2022-07-21:
> @jaygreco could you test the ChibiOS-contrib PR and see if this fixes your lockups as well?

No problem. I'll grab your PR and test with my setup and share results.

jaygreco, 2022-07-21:
As of this afternoon, I was able to reproduce the issue on both master and slave of my test board with your ChibiOS-contrib patch pulled in. Is there any additional debug info I can pull during the failure that might help?

Jpe230, 2022-07-22:
Chiming in...the bug is still reproducible with the linked PR.

I now managed to reliably crash the master half by spamming I2C commands to the OLED display

KarlK90, 2022-07-22:
Interesting, could You both post your configs and the code to make it crash?

Jpe230, 2022-07-22:
Sure!

My config is here: https://github.com/Jpe230/Jpe230sKeebs/blob/main/keyboards/crkbd/keymaps/jpe230_rp/config.h

And to crash the board I use this:
```
diff --git a/drivers/oled/ssd1306_sh1106.c b/drivers/oled/ssd1306_sh1106.c
index 30cfeb5648..1c1453e9a2 100644
--- a/drivers/oled/ssd1306_sh1106.c
+++ b/drivers/oled/ssd1306_sh1106.c
@@ -295,17 +295,8 @@ void oled_render(void) {
         return;
     }
 
-    // Do we have work to do?
-    oled_dirty &= OLED_ALL_BLOCKS_MASK;
-    if (!oled_dirty || oled_scrolling) {
-        return;
-    }
-
     // Find first dirty block
     uint8_t update_start = 0;
-    while (!(oled_dirty & ((OLED_BLOCK_TYPE)1 << update_start))) {
-        ++update_start;
-    }
 
     // Set column & page position
     static uint8_t display_start[] = {I2C_CMD, COLUMN_ADDR, 0, OLED_DISPLAY_WIDTH - 1, PAGE_ADDR, 0, OLED_DISPLAY_HEIGHT / 8 - 1};
@@ -319,6 +310,8 @@ void oled_render(void) {
     if (I2C_TRANSMIT(display_start) != I2C_STATUS_SUCCESS) {
         print("oled_render offset command failed\n");
         return;
+    }else{
+        print("oled_render offset command succeded\n");
     }
 
     if (!HAS_FLAGS(oled_rotation, OLED_ROTATION_90)) {
@@ -346,7 +339,9 @@ void oled_render(void) {
     }
 
     // Turn on display if it is off
-    oled_on();
+    //oled_on();
+    oled_active = true;
+    oled_off();
 
     // Clear dirty flag
     oled_dirty &= ~((OLED_BLOCK_TYPE)1 << update_start);
```

KarlK90, 2022-07-22:
@jaygreco @Jpe230 could you try out this contrib patch on top of my pr? It is hacky atm as I'm away until monday but forces a i2c abort if it can't disable the peripheral. It solved the lockups on my side. 

[i2c_abort.patch.txt](https://github.com/qmk/qmk_firmware/files/9168288/i2c_abort.patch.txt)

Jpe230, 2022-07-22:
45 min with the patch applied and no lockups -- Gonna leave it plugged for the rest of the day but so far it seems that it has resolved the lockups 👯

jaygreco, 2022-07-22:
Thanks @KarlK90 -- happy to test over the weekend. No worries there. I'll post some updates in a few days.

jaygreco, 2022-07-26:
It's looking good on my end -- no OLED hangs so far.

KarlK90, 2022-07-26:
**PRs:**

**ChibiOS-contrib:** https://github.com/ChibiOS/ChibiOS-Contrib/pull/329
**QMK:** https://github.com/qmk/qmk_firmware/pull/17798 

I have mostly rewritten the rp2040 i2c driver in the last couple of days and pushed to current state of affairs to chibios-contrib.

The driver now fully utilizes the `rx`/`tx` fifos for sending and receiving data and has proper error handling now. With this the driver withstands a split keyboard with oleds running a derived torture test by @Jpe230 and can still read out a cirque trackpad connected on the secondary half. QMK needed another update in the handling of i2c errors, which is a continuation of @daskygit work.

Please try out these changes and leave your feedback in the PRs if this fixes your problems.

**Torture test:**

```c
void housekeeping_task_user(void) {
	static int current_block = 0;

	current_block = (current_block + 1) % 512;

	static char white_bg[512] = {0};
	memset(white_bg, 0, 512);
	memset(white_bg, 0xff, current_block);
    oled_write_raw(white_bg, 512);
}
```

0xcharly, 2022-07-27:
Looking good as well on my end: currently running the Dilemma with OLED on the slave side, Cirque trackpad on the master side, and I haven't had any issues in the last 6 hours that the keyboard has been plugged in after being flashed with the patches referenced in the comment above. The same configuration used to lock up in just a few minutes before those changes.

Thanks, @KarlK90!

KarlK90, 2022-07-27:
With https://github.com/qmk/qmk_firmware/pull/17817 merged the fixes should now be available on the latest develop branch.

Jpe230, 2022-07-27:
A late reply:

I torture-tested one extra display (using my old code + rendering it each ms)  in the course of the night and happy to report that I couldn't find any problem with the aforementioned PRs!

Amazing job as always and thanks for fix, @KarlK90!

KarlK90, 2022-07-27:
Nice, thanks for the feedback everyone. I'll close this issue then.

PR #17817 (2022-07-27): [Core] Update ChibiOS Contrib for RP2040 fixes
## Description

Contains:

PWM driver: <https://github.com/ChibiOS/ChibiOS-Contrib/pull/328> 
I2C driver fixes: <https://github.com/ChibiOS/ChibiOS-Contrib/pull/329>

## Types of Changes

- [x] Core
- [x] Bugfix
- [x] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Closes #17720

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Pytho [... cut]
closed 2022-07-27 by KarlK90