qmk/qmk_firmware #17721: [Bug] QMK CLI hangs when I flash hanging on "Trying again every 0.5s ..."

kevinlu1248, 2022-07-19:
QMK CLI hangs when I flash at 
```
Bootloader not found. Make sure the board is in bootloader mode. See https://docs.qmk.fm/#/newbs_flashing
 Trying again every 0.5s (Ctrl+C to cancel)...
```

## Describe the Bug

I am turning the board to bootloader mode. I know this since when I run lsusb I see
```
Bus 003 Device 061: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
```
I tried adding myself to udev by following the instructions at https://docs.qmk.fm/#/faq_build?id=linux-udev-rules although I'm not sure if it's working properly since there's no test listed to verify it.

## System Information

**Keyboard: Cannonkeys bakeneko65**
**Revision (if applicable): rev3 (I think)** 
**Operating system: Ubuntu 21.10** 
**`qmk doctor` output:** 
```
Ψ Found qmk_firmware at /home/kevin/qmk_firmware.
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /home/kevin/qmk_firmware
Ψ Detected Linux.
Ψ Git branch: master
Ψ Repo version: 0.17.5
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

fauxpark, 2022-07-19:
`bakeneko65/rev3` is an AVR board, not ARM: https://github.com/qmk/qmk_firmware/blob/master/keyboards/kkatano/bakeneko65/rev3/rules.mk
So you are trying to flash the wrong firmware for the board you have.

keyboard-magpie, 2022-07-19:
No, rev2 is an AVR board too. You would be better inspecting the PCB and asking your vendor what PCB you actually have.

kevinlu1248, 2022-07-19:
It says Instant65 under lsusb. Does that mean anything? I'm trying to understand it from the Cannonkeys page but it's a bit unclear on their website how to flash.

keyboard-magpie, 2022-07-19:
https://github.com/qmk/qmk_firmware/tree/master/keyboards/cannonkeys/instant65

kevinlu1248, 2022-07-19:
Oh that answers my question. Let me give it a shot.

kevinlu1248, 2022-07-19:
Just flashed and it worked like a charm.