qmk/qmk_firmware #17726: [Bug] DOIO/KB16 firmware doesn't work

caffeinism, 2022-07-20:
## Describe the Bug

The configuration of doio/kb16 seems to use atmega32u4 MCU and atmel-dfu bootloader. However, when I connect the product I bought yesterday, it comes out as a device name of Maple 003. Actually, according to [this page](https://github.com/qmk/qmk_firmware/blob/master/docs/driver_installation_zadig.md), the device name Maple 003 seems to be the stm32duino bootloader. In fact, it doesn't even flash on the keyboard.

## System Information

**Keyboard:** DOIO KB-16-1
**Revision (if applicable):** 
**Operating system:** Windows 11
**`qmk doctor` output:** 

```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: C:/Users/_____/qmk_firmware
Ψ Detected Windows 10 (10.0.22000).
Ψ Git branch: master
Ψ Repo version: 0.17.5
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 6.4
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

tzarc, 2022-07-20:
ping @HorrorTroll

HorrorTroll, 2022-07-20:
Doio released first revision PCB is Atmega32u4, they just recently make second revision and using APM32F103 (probably due Atmega32u4 is shortage and expensive):

Here is the first revision PCB i have:
[image]

caffeinism, 2022-07-20:
Sadly the MCU has changed. How did you measure the pins of row and col? Is there a software way? Or did you manually check the location of the pin? If there is a way I can check it myself, I would like to proceed.

HorrorTroll, 2022-07-20:
You would need multimeter and use continuity measurement mode for that

atommarvel, 2022-07-27:
Hello! I ran into this issue as well and wanted to comment here for anyone looking for a way forward to making their macropad at least usable/customizable. Even though this comment doesn't solve this issue, it should at least help people who find this github issue like I did when trying to understand why the macropad wouldn't work. Please feel free to remove this comment if you don't find it appropriate for this thread.

I used [Vial](https://get.vial.today/) with a sideloaded JSON that I found on [a random product page for the kb16](https://naughtyboards.com/products/doio-kb16-01-16-keys-mini-keyboard-gasket-knob-led-screen) and I'm able to customize the macropad!

Here's the exact json in case that product page goes down:

```json
{
    "name": "DOIO",
    "vendorId": "0xFEED",
    "productId": "0x6060",
    "lighting": "qmk_rgblight",
    "matrix": { "rows": 4, "cols": 8 },
    "layouts": {
      "keymap":[
        [
          "0,0",
          "0,1",
          "0,2",
          "0,3",
          "0,4",
          "0,5",
          "0,6"
        ],
        [
          "1,0",
          "1,1",
          "1,2",
          "1,3",
          "1,4",
          "1,5",
          "1,6"
        ],
        [
          "2,0",
          "2,1",
          "2,2",
          "2,3",
          "2,4",
          "2,5",
          "2,6"
        ],
        [
          "3,0",
          "3,1",
          "3,2",
          "3,3"
        ]
      ]
    }
  }
```

thompson-vii, 2022-08-16:
how do i enter bootloader mode ?

seven-jinxin, 2022-08-26:
Keyboard: DOIO KB-16-1
Revision (if applicable):0201
Operating system: Windows 11
Vendor ID: 0x1EAF
Product ID: 0x0003
Name: Maple 003
BOOTLOADER: stm32duino

I'm facing same issue. The current source code not compatible with this hardware.
After flash the compiled firmware into hardware. it doesn't works any more.
It will keep in DFU mode always.

seven-jinxin, 2022-08-28:
> 

To press the top left key on keyboard, and plug usb cable. It should be in DFU Bootloader mode.
Unplug the usb cable and replug it, the keyboard will switch to normal mode.

frozenkitten, 2022-08-30:
I got the firmware for this revision from their custom support: https://github.com/frozenkitten/DOIO_KB16-3_Archive/blob/main/kb16-3.bin
I have tested it and it works.

OZsGit, 2022-08-31:
How did you contact their customer support? Are they able to provide QMK source files as well?

LuckyTrixx, 2022-09-09:
Hay, I am new to the Game. I bought that Board and Got it Working with the Keamap in VIA, but the Knobs are handled like normal Buttons, and Macros won't work correctly (Every Macro i Bind to a knob, returns O or P).

What does the Firmware from @frozenkitten do?

bradschwyzer, 2022-10-03:
> You would need multimeter and use continuity measurement mode for that

Any instructions on this i have that board and a multimeter. I could get the details.

ivanprea, 2022-10-07:
Hi,
like manyof you, i've recently bought a DOIO macropad. I don't know anything about boards but i'm trying to do my best. I understood that the board has to be reset and then flash whit is firmaware trought QMK Toolbox before connecting it to VIA to custumize the macropad. 

So this i  how i have proceed:

When i opend QMK Toolbox, before the reset, i had:  "HID console connected: qmkbuilder DOIO (FEED:6060:0001)"
so thought that was already good, so i ran VIA (later on i tried also VIAL) webapp. I clicked on "Auhorize device+" and  i got the pop-up window displaying the name device "DOIO".      YEEEEE
NOPE nothing happened.

So i reset the board by pressing the reset button on the board.
On QMK Toolbox i had: 
"HID console connected: qmkbuilder DOIO (FEED:6060:0001)
STM32Duino device connected (WinUSB): (Undefined Vendor) Maple 003 (1EAF:0003:0201)
HID console disconnected: qmkbuilder DOIO (FEED:6060:0001)"

To my knowledge that was a good sign so i flashed it with the firmware that @frozenkitten  provided (because the firmware on VIA website it's in .HEX and QMK Toolbox reads .BIN files). I hit "Flash" got the line "Flash complete"

YEEEEAH

NOPE...AGAIN!

Now when i try to use the VIA webapp and i click on "Auhorize device+" the pop-up to select the board does not appear.
So i click again the reset button and we are back to square one. 

@frozenkitten  and @atommarvel  seems that only one that have figured it out. Could you plase, one of you, try to explain better how you make it work? 

I followed this video to flash the board: 
https://www.youtube.com/watch?v=fuBJbdCFF0Q&t=197s&ab_channel=MechMerlin

I'm sorry for my english if something it's not clear, tell me and i'll try to explain myself better. 

Hope someone can help me out :(

Thanks in advance

bradschwyzer, 2022-10-08:
@ivanprea what firmware did you load? this is a request to add support for the newer revision of the DOIO for qmk.

djhazlett, 2022-10-08:
I am looking for the DOIO 38% firmware. I can confirm the one I just received has an ATmega32U4.

bradschwyzer, 2022-10-08:
@djhazlett that is already supported. This is a request for the new hardware revision.

thompson-vii, 2022-10-08:
https://wiki.keebmonkey.com/en/products/mmp

using the pad with the vendor's firmware requires sideloading the json file on every configuration session.

thompson-vii, 2022-10-08:
https://user-images.githubusercontent.com/31672176/194691357-cba4b358-da8a-49b5-861c-0bb995058761.mp4

ivanprea, 2022-10-08:
Hey everyone,
thanks for the help, mostly to @thompson-vii  with your video was much clear.
The board now appear by loading the "kb16VIA". 

Now i'm struggling to undestand how to switch from layout. This macropad can store 4 layouts but i don't find a way to change it. 

Someone with any suggestions?

JohnnyPrimus, 2022-10-08:
@bradschwyzer when you say the 38% is already supported, what do you mean? are you saying someone has produced QMK keyboard source for that board?

djhazlett, 2022-10-08:
@bradschwyzer When you say its supported, do you mean the ATmega32U4? I dont see DOIO 38%, unless I am looking in the wrong spot. And I am new at building/flashing firmware, so I very well could have missed it. Is this something I need to submit a different topic, or is this not the correct place in general?

RatzzFatzz, 2022-10-08:
@djhazlett as the title says this thread is about doio/kb16. DOIO 38% is a different board/firmware.
The first revision of this macroboard with Atmega32u4 is supported by qmk. The newer versions (like APM32F103) not yet.

JohnnyPrimus, 2022-10-08:
@RatzzFatzz sure, thats understood, but the other poster specifically said the 38 was already supported. since that comes as news to me obviously we'd like to find out more

RatzzFatzz, 2022-10-08:
> I can confirm the one I just received has an ATmega32U4.

@JohnnyPrimus I think @bradschwyzer only looked at the mcu @djhazlett mentioned. 
Judging from the qmk configurator doio 38% is not supported.

ivanprea, 2022-10-08:
Hi,
after spending all the afternoon on VIA i came across some "problems" (maybe it's just me):

- i cannot give other keys to the little knobs exept for "PageUp" "PageDwn" etc, when i rotate the knob. I can map only the press button of it.
- to use the macro pad i got to open all the time VIA, I set it up that it opens when i light the pc but it takes a while, which is weird as i got a very good pc. Plus does not open minimize.
- i can't open app like : Chrome, Discord, Phostoshop, etc. by setting the macro like so: {KC_LGUI} DISCORD {KC_ENT}. The only way it's to send a copy of ech softaware on the desktop and then  set a shortcut. Yes it works but it's annoying to have a lot of icons on the desktop. 

Any of you are having this kind of "problems"?

Thanks

thompson-vii, 2022-10-08:
This is not via support forum, please take your questions to qmk discord via channel https://discord.com/invite/qmk-440868230475677696

PHSC138, 2022-10-10:
> You would need multimeter and use continuity measurement mode for that

I have a multimeter and the new board, what data can I provide to help?
- Does continuity of rows and cols just mean the top pin on the left side of the chip has continuity of the leftmost pin of the top side? Then repeat with each pin?

HorrorTroll, 2022-10-13:
Thanks @PHSC138 for helping out finding I/O pin, so that I could rewrite QMK firmware for this new revision PCB. More details is on the mentioned above

thompson-vii, 2022-10-17:
the encoder has 20 dedents, the oem firmware emits 27 pulses per revolution. Does the port default to 20 pules per revolution?

HorrorTroll, 2022-10-17:
yes, the port default is 20 for both 2 rev

thompson-vii, 2022-10-18:
[image]

barebone layout json

```json
{
    "name": "KB16-01",
    "vendorId": "0xD010",
    "productId": "0x1601",
    "lighting": "none",
    "matrix": {
        "rows": 4,
        "cols": 5
    },
    "layouts": {
        "keymap": [
			["0,0","0,1","0,2","0,3",{"x":0.25},"0,4\n\n\n\n\n\n\n\n\ne0",{"x":0.75},"1,4\n\n\n\n\n\n\n\n\ne1"],
			["1,0","1,1","1,2","1,3"],
			["2,0","2,1","2,2","2,3",{"x":0.75,"w":1.75,"h":1.75},"2,4\n\n\n\n\n\n\n\n\ne2"],
			["3,0","3,1","3,2","3,3"]
        ]
    }
}

```

thompson-vii, 2022-10-18:
sometime the board doens't get recognized after sideloading, i don't know why. reconnect the cable might work.

thompson-vii, 2022-10-18:
i have ported the board to vial 

[image]

thompson-vii, 2022-10-22:
do you know the dimension of the oled display ?

HorrorTroll, 2022-10-22:
it is 128x32 SSD1306

HorrorTroll, 2022-10-30:
This issue has been resolved, can anyone closed this?

closed 2022-12-03 by zvecr