qmk/qmk_firmware #17738: [Feature Request] One Shot Next Key functionality

Bertie2011, 2022-07-21:
We have one shot keys for modifiers and layers, but it would be nice to have a generic "One Shot" keycode that will hold the next key pressed until another key is pressed. Kind of like key lock keycode, but without having to disable the lock afterwards.

Example: Pressing in order `OS`, `F3`, `T` -> Generates `F3 + T`

## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

jasonm23, 2022-07-21:
This is known as a modal keymap.  AFAIK there are several implementations of this, but I'll need to find something specific to be more help...

jasonm23, 2022-07-21:
Meanwhile... essentially, you create a layer for the keys that you want to use for the modal keymap (i.e. what keys you'd want to hit after the "one shot" key that activates the layer.)

Then each key in that layer does their thing and switches the keymap layer back to base.

jasonm23, 2022-07-21:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodox_ez/keymaps/vim/vim.h

Step one... see how someone is doing it.

jasonm23, 2022-07-21:
First thing to notice is that any function which is bound to a key can run any sequence of commands or keys.

So you should be able to work out that, switching back to a base layer is easy, and possible from any key in an "upper" layer.

Bertie2011, 2022-07-22:
Yeahhhh, I figured you could probably do it using functions or some other workaround like layers.... but I would rather just be able to insert a "basic keycode" into my keymap and have it work right away😅 (without it being specific to a first key or a set of second keys).

jasonm23, 2022-07-22:
.... so it should just kinda figure out what you want.

Wait for the AI QMK.

(no that's not a thing.)

closed 2022-11-21 by github-actions[bot]