qmk/qmk_firmware #17744: [Feature Request] RP2040 UART driver

Kriechi, 2022-07-21:
Feature request to add UART driver for the RP2040.

ChibiOS seems to be already ready for it to a large degree.

## Feature Request Type

- [x] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

* add UART driver for RP2040 to enable communication with serial devices using a simple write/read API using uint8 or string commands.
* document necessary settings and options in https://docs.qmk.fm/#/uart_driver
* Split keyboards should be able to use both PIO/SIO (for split) and UART (for other devices) at the same time, and any limitation /restriction documented.

karosc, 2022-11-16:
I got a uart driver working using the pico sdk submodule. I added a [uart_vendor](https://github.com/karosc/qmk_firmware/tree/rp_test/platforms/chibios/drivers/vendor/RP/RP2040) driver that you can select using `UART_DRIVER = vendor` in rules.mk. The uart drivers had to be refactors a bit to accommodate the second interface in addition to the hal_serial one. I have a redox_w clone that uses the uart driver for communicating with a bluetooth receiver and recently upgraded from ATMega to RP2040. So far it's working great (although I had to knock the baud rate down from 1M to get stable IO)

JohnAZoidberg, 2023-01-06:
Pretty cool @karosc! I just tried your branch on a raspberry pico and it works if I lower the baud rate like `init_uart(9600);`.

closed 2023-05-07 by github-actions[bot]