qmk/qmk_firmware #17749: [Feature Request] Add "XF86AudioMicMute" keycode?

nazarewk, 2022-07-22:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

On my laptop i have a function key dedicated to muting Microphone (as opposed to muting audio output), but i don't see the mic button anywhere in Oryx/documentation.

In my Sway config it is called `XF86AudioMicMute` (and  `XF86AudioMute` for audio mute)

https://github.com/nazarewk-iac/nix-configs/blob/23c4b1dddfdfea8b07a447ca21f7977a2f6604c9/users/nazarewk/sway/config#L217-L218

It would be nice to add the missing keycode to QMK

fauxpark, 2022-07-22:
`XF86AudioMicMute` is not an HID keyboard keycode, it is an X keysym.

The Linux kernel defines its own set of internal event codes, one of which is [`KEY_MIC_MUTE`](https://elixir.bootlin.com/linux/latest/source/include/uapi/linux/input-event-codes.h#L339). These event codes are mapped from various input methods such as PS/2, USB HID, and ACPI, which is what most laptops use for this functionality. Xorg further maps these event codes to keysyms, but there is a rather hard limit on the number of possible values.

Over HID, it is technically possible using the [Telephony usage page](https://elixir.bootlin.com/linux/latest/source/drivers/hid/hid-input.c#L981), but it is not as simple as just adding the keycode, and it would only work on Linux, as far as I'm aware.

sigprof, 2022-07-22:
Actually xkeyboard-config has two mappings for `XF86AudioMicMute`:
https://gitlab.freedesktop.org/xkeyboard-config/xkeyboard-config/-/blob/xkeyboard-config-2.36/symbols/inet#L108
```
    key <FK20>   {      [ XF86AudioMicMute      ]       };
```
https://gitlab.freedesktop.org/xkeyboard-config/xkeyboard-config/-/blob/xkeyboard-config-2.36/symbols/inet#L225
```
   key <I256>   {       [ XF86AudioMicMute               ]      }; // KEY_MICMUTE
```
`<I256>` is this XKB keycode:
https://gitlab.freedesktop.org/xkeyboard-config/xkeyboard-config/-/blob/xkeyboard-config-2.36/keycodes/evdev#L324
```
	<I256> = 256;		// #define KEY_MICMUTE             248
```
(The `evdev` and `libinput` drivers have a simple keycode mapping: `xkb_keycode = linux_input_keycode + 8`)

However, if you look two lines above in `keycodes/evdev`, you will find [this comment](https://gitlab.freedesktop.org/xkeyboard-config/xkeyboard-config/-/blob/xkeyboard-config-2.36/keycodes/evdev#L322):
```
	// Key codes below cannot be used in X
```
The X11 protocol uses only 1 byte for the keycode field, therefore any keycodes above 255 cannot be used with it — apparently everything above that is Wayland-only. But even if you use Wayland, there does not seem to be a way to generate a HID report which would be converted to `KEY_MICMUTE` from QMK — [drivers/hid/hid-input.c](https://github.com/torvalds/linux/blob/v5.18/drivers/hid/hid-input.c) has two mappings which result in this code:
- `HID_UP_TELEPHONY`, `0x002f` — from the Telephony page, which is currently not advertised by QMK in any HID report;
- `HID_UP_HPVENDOR2`, `0x0001` — some vendor-specific usage from a HP device, which is not emulated by QMK.

However, you can use the other XKB symbol mapping which results in `XF86AudioMicMute` — `<FK20>`; it is `KC_F20` in QMK. So you may try to add `KC_F20` into your keymap, and most Linux versions will detect that key as `XF86AudioMicMute`.

BTW, there are some more keycodes in the F13…F24 range that have XKB symbol mappings by default:
```
    key <FK21>   {      [ XF86TouchpadToggle    ]       };
    key <FK22>   {      [ XF86TouchpadOn        ]       };
    key <FK23>   {      [ XF86TouchpadOff       ]       };
```

nazarewk, 2022-07-22:
Note: I am pretty sure the key also works on Windows (probably by some other means than in Linux) as I originaly started with windows installation on this laptop.

fauxpark, 2022-07-22:
Probably only through ACPI and/or an OEM driver; I just bodged the Telephony/Phone Mute usage into a test firmware and I can see it being sent in Wireshark, but it does not seem to do anything on macOS or Windows.

sigprof, 2022-07-22:
The laptop keys are often special; you may try using `sudo evtest` to investigate what events are actually sent by that key in Linux (you will need to guess which device to watch though). The laptop manufacturer may also supply some special Windows drivers (which then may be automatically installed from Microsoft update services), so the key may appear to work out of the box.

nazarewk, 2022-07-22:
> The laptop keys are often special; you may try using `sudo evtest` to investigate what events are actually sent by that key in Linux (you will need to guess which device to watch though). The laptop [...]

Looks like this for me, coming from `/dev/input/event17:	Dell WMI hotkeys` (as opposed to standard keys at `/dev/input/event1:	AT Translated Set 2 keyboard`):
```
Event: time 1658500002.130421, type 4 (EV_MSC), code 4 (MSC_SCAN), value 100150
Event: time 1658500002.130421, type 1 (EV_KEY), code 190 (KEY_F20), value 1
Event: time 1658500002.130421, -------------- SYN_REPORT ------------
Event: time 1658500002.130463, type 1 (EV_KEY), code 190 (KEY_F20), value 0
Event: time 1658500002.130463, -------------- SYN_REPORT ------------
Event: time 1658500004.315680, type 4 (EV_MSC), code 4 (MSC_SCAN), value 100150
Event: time 1658500004.315680, type 1 (EV_KEY), code 190 (KEY_F20), value 1
Event: time 1658500004.315680, -------------- SYN_REPORT ------------
Event: time 1658500004.315718, type 1 (EV_KEY), code 190 (KEY_F20), value 0
Event: time 1658500004.315718, -------------- SYN_REPORT ------------
```

nazarewk, 2022-07-22:
so it seems i could just use `KC_F20` mapping to get an equivalent.

sigprof, 2022-07-22:
Yes, but now I wonder why you get F20 from there :-)

The code in https://github.com/torvalds/linux/tree/master/drivers/platform/x86/dell definitely uses `KEY_MICMUTE` in multiple places; however, there were patches like https://github.com/torvalds/linux/commit/617103246cfd19af837e4cb614ba9f877c4f7779 replacing `KEY_MICMUTE` with `KEY_F20` to make it actually work under X11 — but that was for a completely different driver.

Hmm, looks like it might be coming from the keymap override included in systemd — https://cgit.freedesktop.org/systemd/systemd/tree/hwdb/60-keyboard.hwdb?id=aa75494ad5cdf7bede947212ad8c8356d78580fa#n273

nazarewk, 2022-07-25:
> Hmm, looks like it might be coming from the keymap override included in systemd — [cgit.freedesktop.org/systemd/systemd/tree/hwdb/60-keyboard.hwdb?id=aa75494ad5cdf7bede947212ad8c8356d78580fa#n273](https://cgit.freedesktop.org/systemd/systemd/tree/hwdb/60-keyboard.hwdb?id=aa75494ad5cdf7bede947212ad8c8356d78580fa#n273)
Sounds about right, I am on Dell Latitude E5470, there is a [rule for Dell Latitude line](https://cgit.freedesktop.org/systemd/systemd/tree/hwdb/60-keyboard.hwdb?id=aa75494ad5cdf7bede947212ad8c8356d78580fa#n275)

closed 2022-11-24 by github-actions[bot]