qmk/qmk_firmware #17756: [Bug] Boardsource/Beiwagon keymap displayed wrong

AzGoalie, 2022-07-23:
### Describe the Bug

The Boardsource/Beiwagon keymap is displayed as a 3x3 when the keyboard is a 4x3
It seems to correctly load the layout, as it loads the 2 layers each with 12 keys, and just a graphical issue with the configurator.

What is displayed in the configurator currently:
[image]

What the keyboard looks like:
[image]

yanfali, 2022-07-23:
@0x6A73@pm.me looks like your commit b99ccd4f064e15c7ea94184986e396603bfa1dc2 needs some love.

waffle87, 2022-07-24:
oops, that's my bad. thanks @noroadsleft!

PR #17771 (2022-07-24): boardsource/beiwagon: correct QMK Configurator layout data
## Description

[title]

cc:
@boardsource, @daysgobye (keyboard maintainers)
@waffle87 (last editor)

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #17756 

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python) [... cut]
closed 2022-07-24 by fauxpark