qmk/qmk_firmware #17759: [Bug] [develop] VIA encoder support does not populate default values on keymap reset

vinorodrigues, 2022-07-23:
## Describe the Bug

Recent PR [#17734] added support for rotary encoders using the `ENCODER_MAP_ENABLE` functionality - however I'm finding that on EEPROM reset the default values encoded into `PROGMEM encoder_map` array are not copied onto the EEPROM.  *(rather all the CW and CCW values are nulled to `KC_NO` )*

My assumption is that since the `dynamic_keymap_set_encoder()` function does some shifting and masking [L140-L141](https://github.com/qmk/qmk_firmware/blob/2202efaf0cdae766b76de1f87dc3c440480e5280/quantum/dynamic_keymap.c#L140-L141), but `dynamic_keymap_reset()` does not [L157-L158](https://github.com/qmk/qmk_firmware/blob/2202efaf0cdae766b76de1f87dc3c440480e5280/quantum/dynamic_keymap.c#L157-L158), that the issue lies there?

See: https://discord.com/channels/673730324047462449/673736911839428641/1000188112292552715 *(on Keychron discord)*

Jpe230, 2022-07-23:
I couldn't reproduce the bug using the Adafruit MacroPad with the [VIA keymap](https://github.com/qmk/qmk_firmware/blob/develop/keyboards/adafruit/macropad/keymaps/via/keymap.c)

For me it works as expected, I tried with BootMagic Lite and `QK_CLEAR_EEPROM` to reset the EEPROM and it both cases the keymap was populated correctly into the EEPROM (In the video I used BootMagic Lite)

https://user-images.githubusercontent.com/6202305/180626029-f8c1ec67-9b73-4ac2-b8e8-892cc0a433a1.mp4

vinorodrigues, 2022-07-24:
darn - i can confirm this is so.  must have been residual eeprom data that was not initialised when i coded in the encoder bits *(and didn't call `QK_CLEAR_EEPROM`)*

sorry to have bothered :( 

closing *(but may be good to leave as a lessons learned to others that may try to port the code)*

closed 2022-07-24 by vinorodrigues