qmk/qmk_firmware #17762: Atmel32U4  failed to flash qmk

mynameiswutong, 2022-07-23:
When  I  try to compile a **via** keymap I created, qmk system prompts that there is an error （As shown in Figure 1）；Then I add `EEPROM_DRIVER = transient` code in my `rules.mk` , and define `TRANSIENT_EEPROM_SIZE` in my `config.h` . I compile a `.hex` firmware file, It compiles successfully。Then I use `QMK toolbox` to flash my keyboards , and i did it ; But when I try to test my new firmware , it doesn't work as expected. Strictly speaking, it doesn't work , and I can't even find it on my computer. If I delete `EEPROM_DRIVER = transient`、`TRANSIENT_EEPROM_SIZE` and remove `VIA` related codes , compile and flash again ,it  works.

Please help me

[image]

fauxpark, 2022-07-23:
You have too many layers. VIA uses the EEPROM to store the keymap, but this is only 1024 bytes on the ATmega32U4. It's actually less because QMK itself uses about 33 bytes, and the VIA dynamic macros require at least 100 bytes. The formula for calculating how much space the keymap will take up is `rows * cols * layers * 2`. If that is more than about 890 bytes, you will need to remove layers from the `via` keymap and add `#define DYNAMIC_KEYMAP_LAYER_COUNT x` to config.h.

mynameiswutong, 2022-07-23:
> You have too many layers. VIA uses the EEPROM to store the keymap, but this is only 1024 bytes on the ATmega32U4. It's actually less because QMK itself uses about 33 bytes, and the VIA dynamic [...]

Thank you very much , I  did it

zvecr, 2022-08-03:
Feel free to repoen if you are still having issues.

closed 2022-08-03 by zvecr