qmk/qmk_firmware #17775: Variable declaration lettercase in rules.mk shouldn't matter (or a warning should be issued if it does and the wrong lettercase is used)

sandragem, 2022-07-24:
Ran into a problem where VIA_ENABLE = YES (with "YES" being uppercase) in rules.mk will not enable VIA. Board cannot be recognized or configured unless "yes" is in lowercase.

The enabled firmware seemed to have otherwise compiled successfully as there was no error message or other indication alerting about the wrong lettercase being used. All keyboard and matrix testing worked normally, but no changes could be made to the default keymap until revising the rules.mk variable to lowercase, which got the board properly recognized.

This seems like an easy enough mistake to make that turns into a nightmare to spot, so not really ideal. It got me wondering: is there any particular reason why the variables set in rules.mk must be set in lowercase? After all, it seems like if we're just trying to match the "yes" or "no" string when declaring the variable, lettercase shouldn't really matter.

Alternatively, if lettercase _is_ somehow important for the thing to work properly, then some kind of warning alerting of the potential issue would be useful. So if variables are declared but anything other than "yes" or "no" in lowercase is used then it should spark a warning of the error.

fauxpark, 2022-07-24:
make is unfortunately not that smart. There is no type system to speak of, Makefile variables are just text, so in the context of QMK anything other than `yes` is assumed to be `no`. For example: https://github.com/qmk/qmk_firmware/blob/master/builddefs/common_features.mk#L696-L700

This is for simplicity and readability; changing this logic to be case insensitive seems nontrivial in make from what I can tell, and it would need to be done in many, many places.

sandragem, 2022-07-24:
I see. In that case, perhaps in that case adding something to the documentation as something to look out for if having this issue might prove useful, then.

fauxpark, 2022-07-24:
It is in the documentation. Everything that shows a rules.mk variable will have the correct case you should use.

zvecr, 2022-07-24:
We are moving to features being enabled the data driven way (within `info.json` and `keymap.json` ). Schema validation will ensure you cannot specify the wrong case.

closed 2022-08-29 by tzarc