qmk/qmk_firmware #17779: [Feature Request] AT Keyboard Converter

Muirium, 2022-07-24:
Hi guys, longtime [Soarer’s Converter](https://deskthority.net/viewtopic.php?f=7&t=25309)  user here, first time QMK. (All my QMK so far has been via [Pandrew’s capsense port](https://deskthority.net/viewtopic.php?p=480235#p480235) for IBM Model F and beamspring.)

I’d like to run QMK on my Teensy _converted_ AT/PS2 keyboards. (Some have an internal Teensy, some hook up via their original cable to a Teensy in a box; in either case it’s always an AT protocol keyboard being converted to USB via a 32u4.) Ideally, I’d like to be able to select the basic, generic full-size AT layout in QMK Configurator, and define layers there to my heart’s content, then compile, download and flash the hex file to the Teensy. But I see no AT converters in QMK Configurator’s menu. Really? There’s ADB support but not the ubiquitous AT/PS2? Woah!

Here’s a thread I wrote about this on Deskthority. I’m not alone in desiring this:

https://deskthority.net/viewtopic.php?f=7&t=27412

I know my way around [TMK Unimap](https://deskthority.net/viewtopic.php?p=505819#p505819) and QMK Configurator a bit, as a user, but I don’t know how to define new targets. An AT converter would get a lot of use!

keyboard-magpie, 2022-08-03:
I _think_ [this](https://github.com/qmk/qmk_firmware/tree/master/keyboards/converter/ibm_terminal) might be the QMK conversion, but I am by no means a SME in this area. The relevant config.qmk.fm entry would to match would be [this](https://config.qmk.fm/#/converter/ibm_terminal/LAYOUT)

snacksthecat, 2022-08-03:
It looks like that may be only for code set 3 (terminal), but worth a shot. Give it a go, @Muirium.

_"It supports PS/2 Scan Code Set 3 and runs on USB AVR chips such like PJRC Teensy."_

Thanks @keyboard-magpie

sigprof, 2022-08-03:
From looking at https://www.win.tue.nl/~aeb/linux/kbd/scancodes-10.html, `converter/ibm_terminal` has a high chance to fail with many PC keyboards:

> (vii) The [Microsoft Keyboard Scan Code Specification](http://www.microsoft.com/hwdev/download/desinit/scancode.zip) writes: _In the very early days of Windows NT, an attempt was made to use the much more orthogonal Scan Code Set 3, but due to bugs in the implementation of this Scan Code Set on numerous OEM keyboards, the idea was abandoned._ And also: _Scan Code Set 3 is not used or required for operation of Microsoft operating systems._

(As usual, any feature that is not used by Windows is almost guaranteed to be either not implemented or broken.)

There was an attempt to add a converter suitable for PC keyboards (#12821), but looks like it was abandoned.

Muirium, 2022-08-18:
> There was an attempt to add a converter suitable for PC keyboards (#12821), but looks like it was abandoned.

I'd like to see that completed and brought to QMK Configurator. I can vouch for TMK's ibmpc_usb converter myself, as that's what I'm using now. QMK Configurator is an improvement on Hasu's Unimap, however, so I'm still missing that. Besides, I'm sure many people have this exact use case.

closed 2022-12-18 by github-actions[bot]

ifohancroft, 2023-01-02:
I'd love that feature as well by the way.

rljacobson, 2023-05-25:
> There was an attempt to add a converter suitable for PC keyboards (https://github.com/qmk/qmk_firmware/pull/12821), but looks like it was abandoned.

It looks to me more like it wasn't accepted because the author added mouse support in his second to last commit that was written in C++, and someone said it should be written in C. A week later he removed the C++ mouse support he had added ("for now") and then closed the PR himself without explanation, so I don't really know. It's not clear what actually happened there. It looks complete to me. Maybe he couldn't get anyone interested in it, or maybe he was frustrated that his hard work on the mouse stuff was met with a negative comment. (It's all too easy to infer a tone in words written in text on the internet that isn't intended by the author.)

Maybe it's worth opening an issue to discuss if that PR could be re-opened, rebased, and merged. I'm not super familiar with how QMK development works w.r.t. code maintenance "ownership," for lack of a better phrase, but I would imagine that responsibility requires almost no work to meet.

marfrit, 2024-08-03:
At the time being I just made the TMK stuff work somehow with QMK.

There's a vial fork out there with a bit more polished version. 

I myself have moved to RP2040 and PIO based code. There's a mouse/ keyboard version in my repo (pio_m4_converter iirc) which is also wip, since my offsprings are quite demanding over keyboard hobbying ;-)

marfrit, 2025-08-05:
Awesome! The mentioned pio_m4_converter has an additional trick to get the sequence of a host -> device -> reply right by flipping a bit switch (I had problems with that sometimes also when flipping the indicator leds). 

On merging upstream - I am not against anyone else persuing this. For the time being, I do have a pluggable design in mind supporting avr, stm32 and rp2040 - but as often with "almost perfect designs", they're hard to actually implement, given my aforementioned situation has not changed much ;)