qmk/qmk_firmware #17802: stuck on handwired split right not registering keystrokes

evils, 2022-07-26:
Hi, i handwired a TBK mini with some raspberry pi pico boards

after a few days of help from discord i managed to get to a point where my setup mostly works
the left hand registers keystrokes, and the keys correctly map to the keymap
it also works independent on which half i connect to USB
(so the right half also works and serial communication works)

however none of my right hand's keys result in a keycode
(directly shorting a row and a column pin also does nothing, so i don't think i somehow miswired just the right)
this feels like i'm just missing something obvious, but so far neither me or the discord folks haven't found it
so i'm hoping this issue gets the right eyes on my work

https://github.com/evils/qmk_firmware/tree/tbkpico/keyboards/handwired/tbkpico

evils, 2022-07-30:
so far i've tried to switch the handedness of the halves to show the right half's wiring is ok
found that side detection isn't supported in data driven keyboards yet
so added the support of that that i needed

had it briefly work

cleaned up my code
and now it not only doesn't work, but my left half's keymap is scrambled again

evils, 2022-07-31:
turns out i'm an idiot who was flashing a different file than the one that was just built

everything seems to work, except layer switching across the split

drashna, 2022-08-07:
This may help: 
https://docs.splitkb.com/hc/en-us/articles/360010588860-Only-one-half-of-my-keyboard-works-at-a-time-but-not-when-they-are-both-connected

evils, 2022-08-08:
nice article, thanks
my symptoms are different though, both halves work perfectly on their own
and connecting the TRRS cable has no effect on that
i will double check the continuity up to the pico boards instead of just to the connectors

but the change from both halves working and communication no longer working happened when i figured out i was a stale firmware file

side note, i'm doing the rx/tx switching in my wiring
seems like it's possible now that the 2 halves know they're halves, one attempts to do the switching in firmware
but i don't think i defined that anywhere

evils, 2022-08-08:
turns out my `GP0` is shorted to GND somewhere
currently i'm assuming this is ESD damage in the RP2040 as i can't find any shorts
the pico board is glued down to some proto board so there could be a short on the bottom...

for now i'm failing to do a bitbanged half duplex on the remaining pin...

evils, 2022-08-09:
didn't get bitbanging half duplex going on the pico
but i had a spare, so i swapped it out and now everything works :D

working on opening a PR for my data driven additions now

closed 2022-08-09 by evils