qmk/qmk_firmware #17804: Compilation error on MacBook Pro (13-inch, M2, 2022)

s4ms0n, 2022-07-26:
Hi, I am running:
`qmk compile -kb input_club/ergodox_infinity -km samsons_layout
`
where samsons_layout is my own layout that I was able to build with a docker image.
Now I do not have this option and I am trying to build it directly on my macOS 12.4 (M2 chip)

I am getting:
`Ψ Compiling keymap with gmake --jobs=1 input_club/ergodox_infinity:samsons_layout

QMK Firmware 0.17.5
WARNING: Some git submodules are out of date or modified.
 Please consider running make git-submodule.

Making input_club/ergodox_infinity with keymap samsons_layout

arm-none-eabi-gcc (GCC) 12.1.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: layouts/community/ergodox/samsons_layout/keymap.c                                        [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/send_string.c                                                                    [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/logging/print.c                                                                  [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/split_common/split_util.c                                                        [OK]
Compiling: quantum/split_common/transport.c                                                         [OK]
Compiling: quantum/split_common/transactions.c                                                      [OK]
Compiling: quantum/main.c                                                                           [OK]
Compiling: lib/printf/printf.c                                                                      [OK]
Compiling: quantum/mousekey.c                                                                       [OK]
Compiling: platforms/chibios/eeprom_teensy.c                                                        [OK]
Compiling: quantum/process_keycode/process_backlight.c                                              [OK]
Compiling: quantum/led_matrix/led_matrix.c                                                          [OK]
Compiling: quantum/led_matrix/led_matrix_drivers.c                                                  [OK]
Compiling: lib/lib8tion/lib8tion.c                                                                  [OK]
Compiling: drivers/led/issi/is31fl3731-simple.c                                                     [OK]
Compiling: platforms/chibios/sleep_led.c                                                            [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: quantum/crc.c                                                                            [OK]
Compiling: drivers/lcd/st7565.c                                                                     [OK]
Compiling: quantum/process_keycode/process_unicode.c                                                [OK]
Compiling: quantum/process_keycode/process_unicode_common.c                                         [OK]
Compiling: quantum/utf8.c                                                                           [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/command.c                                                                        [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: platforms/chibios/drivers/i2c_master.c                                                   [OK]
Archiving: .build/obj_input_club_ergodox_infinity_samsons_layout/i2c_master.o                       [OK]
Compiling: platforms/chibios/drivers/serial_usart.c                                                platforms/chibios/drivers/serial_usart.c: In function ‘usart_init’:
platforms/chibios/drivers/serial_usart.c:140:17: note: ‘#pragma message: usart_init: MCU Familiy not supported by default, please supply your own init code by implementing usart_init() in your keyboard files.’
  140 | #        pragma message "usart_init: MCU Familiy not supported by default, please supply your own init code by implementing usart_init() in your keyboard files."
      |                 ^~~~~~~
[... 117 lines cut ...]
Compiling: lib/chibios-contrib/os/hal/src/usbh/hal_usbh_ftdi.c                                      [OK]
Compiling: lib/chibios-contrib/os/hal/src/usbh/hal_usbh_aoa.c                                       [OK]
Compiling: lib/chibios-contrib/os/hal/src/usbh/hal_usbh_hid.c                                       [OK]
Compiling: lib/chibios-contrib/os/hal/src/usbh/hal_usbh_uvc.c                                       [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_ee24xx.c                                              [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_ee25xx.c                                              [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_eeprom.c                                              [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_timcap.c                                              [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_qei.c                                                 [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_usb_hid.c                                             [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_usb_msd.c                                             [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_comp.c                                                [OK]
Compiling: lib/chibios-contrib/os/hal/src/hal_opamp.c                                               [OK]
Linking: .build/input_club_ergodox_infinity_samsons_layout.elf                                      [ERRORS]
 |
 | /opt/local/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: cannot find -lg_nano
 | /opt/local/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: cannot find -lc_nano
 | collect2: error: ld returned 1 exit status
 |
gmake[1]: *** [builddefs/common_rules.mk:364: .build/input_club_ergodox_infinity_samsons_layout.elf] Error 1
gmake: *** [Makefile:413: input_club/ergodox_infinity:samsons_layout] Error 1
Make finished with errors
`

Any help will be appreciated .

s4ms0n, 2022-08-01:
Anyone?

keyboard-magpie, 2022-08-01:
`Please consider running make git-submodule.` in the warnings may be  of some help to you.

s4ms0n, 2022-08-01:
make git-submodule
QMK Firmware 0.17.5
Synchronizing submodule url for 'lib/chibios'
Synchronizing submodule url for 'lib/chibios-contrib'
Synchronizing submodule url for 'lib/googletest'
Synchronizing submodule url for 'lib/lufa'
Synchronizing submodule url for 'lib/printf'
Synchronizing submodule url for 'lib/vusb'
Submodule 'ext/mcux-sdk' (https://github.com/NXPmicro/mcux-sdk) registered for path 'lib/chibios-contrib/ext/mcux-sdk'
Cloning into '/Users/andriysamsonyuk/Repositories/qmk_firmware/lib/chibios-contrib/ext/mcux-sdk'...
remote: Enumerating objects: 37615, done.
remote: Counting objects: 100% (2486/2486), done.
remote: Compressing objects: 100% (1151/1151), done.
remote: Total 37615 (delta 1328), reused 2486 (delta 1328), pack-reused 35129
Receiving objects: 100% (37615/37615), 74.07 MiB | 11.01 MiB/s, done.
Resolving deltas: 100% (25754/25754), done.
Submodule path 'lib/chibios-contrib/ext/mcux-sdk': checked out '102c27d7cc2aaac87c133702e7829a105e0cb433'

s4ms0n, 2022-08-01:
the error is the same:

qmk compile -kb input_club/ergodox_infinity -km samsons_layout
Ψ Compiling keymap with gmake --jobs=1 input_club/ergodox_infinity:samsons_layout

QMK Firmware 0.17.5
Making input_club/ergodox_infinity with keymap samsons_layout

arm-none-eabi-gcc (GCC) 12.1.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: layouts/community/ergodox/samsons_layout/keymap.c                                        [OK]
Compiling: quantum/command.c                                                                        [OK]
Linking: .build/input_club_ergodox_infinity_samsons_layout.elf                                      [ERRORS]
 |
 | /opt/local/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: cannot find -lg_nano
 | /opt/local/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: cannot find -lc_nano
 | collect2: error: ld returned 1 exit status
 |
gmake[1]: *** [builddefs/common_rules.mk:364: .build/input_club_ergodox_infinity_samsons_layout.elf] Error 1
gmake: *** [Makefile:413: input_club/ergodox_infinity:samsons_layout] Error 1
Make finished with errors

fauxpark, 2022-08-01:
Your arm-gcc is not the version that the QMK setup script installs (`osx-cross/arm/arm-gcc-bin@8`). You will need to remove it and run `qmk setup`.

s4ms0n, 2022-08-01:
> 

It helped, thanks a lot!

closed 2022-08-01 by s4ms0n