qmk/qmk_firmware #17825: Linking error when using JOYSTICK_DRIVER = digital

greaka, 2022-07-28:
Trying to compile qmk with only digital joystick enabled doesn't seem to work. `analogReadPin` is defined in `analog.h`, which I don't need otherwise. Is this a bug or am I missing something?

```
Linking: .build/gmmk_pro_rev1_iso_greaka.elf                                                        [ERRORS]
 |
 | /usr/lib/gcc/arm-none-eabi/12.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_gmmk_pro_rev1_iso_greaka/quantum/process_keycode/process_joystick.o: in function `process_joystick_analogread_quantum':
 | /git/qmk_firmware/quantum/process_keycode/process_joystick.c:110: undefined reference to `analogReadPin'
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [builddefs/common_rules.mk:364: .build/gmmk_pro_rev1_iso_greaka.elf] Error 1
Make finished with errors
make: *** [Makefile:413: gmmk/pro/rev1/iso:greaka:flash] Error 1
```

greaka, 2022-07-28:
If anyone wants to reproduce this, here is a patch:
<details>
<summary>patch</summary>

Apply this on top of `e7affd05417a48b1e5c0948a24034227384a5d13` (current master at time of writing)

```
From 05699244410a537d77c91d1daa8f292ff7d185a5 Mon Sep 17 00:00:00 2001
From: Greaka <git@greaka.de>
Date: Thu, 28 Jul 2022 10:45:16 +0200
Subject: [PATCH] custom keybinds

---
 keyboards/gmmk/pro/rev1/iso/config.h          |  3 ++
 .../gmmk/pro/rev1/iso/keymaps/greaka/keymap.c | 37 +++++++++++++++++++
 .../gmmk/pro/rev1/iso/keymaps/greaka/rules.mk |  5 +++
 3 files changed, 45 insertions(+)
 create mode 100644 keyboards/gmmk/pro/rev1/iso/keymaps/greaka/keymap.c
 create mode 100644 keyboards/gmmk/pro/rev1/iso/keymaps/greaka/rules.mk

diff --git a/keyboards/gmmk/pro/rev1/iso/config.h b/keyboards/gmmk/pro/rev1/iso/config.h
index 26626ccfc1..bfcac18389 100644
--- a/keyboards/gmmk/pro/rev1/iso/config.h
+++ b/keyboards/gmmk/pro/rev1/iso/config.h
@@ -21,3 +21,6 @@
 #define DRIVER_1_LED_TOTAL 66
 #define DRIVER_2_LED_TOTAL 33
 #define DRIVER_LED_TOTAL (DRIVER_1_LED_TOTAL + DRIVER_2_LED_TOTAL)
+#define UNICODE_SELECTED_MODES UC_LNX
+#define JOYSTICK_BUTTON_COUNT 0
+#define JOYSTICK_AXES_COUNT 1
diff --git a/keyboards/gmmk/pro/rev1/iso/keymaps/greaka/keymap.c b/keyboards/gmmk/pro/rev1/iso/keymaps/greaka/keymap.c
new file mode 100644
index 0000000000..731f25f3ac
--- /dev/null
+++ b/keyboards/gmmk/pro/rev1/iso/keymaps/greaka/keymap.c
@@ -0,0 +1,37 @@
+#include QMK_KEYBOARD_H
+#ifdef JOYSTICK_ENABLE
+  #include "joystick.h"
+#endif
+
+const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
+	[0] = LAYOUT(KC_ESC,  KC_F1,   KC_F2,   KC_F3,   KC_F4,   KC_F5,   KC_F6,   KC_F7,   KC_F8,   KC_F9,   KC_F10, KC_F11, KC_F12, KC_DEL, KC_PSCR, KC_GRV,  KC_1,    KC_2,    KC_3,    KC_4,    KC_5,    KC_6,    KC_7,    KC_8,    KC_9,    KC_0,    KC_MINS, KC_EQL,  KC_BSPC, KC_HOME, KC_TAB,  KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P,    KC_LBRC, KC_RBRC, KC_PGUP, MO(1),   KC_A,    KC_S,     KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L,    KC_SCLN, KC_QUOT, KC_BSLS, KC_ENT,  KC_PGDN, KC_LSFT, KC_BSLS, KC_Z,    KC_X,   KC_C,    KC_V,    KC_B,    KC_N,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH, KC_RSFT, KC_UP,   KC_END,  KC_LCTL, KC_LGUI, KC_LALT, KC_SPC,  KC_RALT, KC_APP,  KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT),
+	[1] = LAYOUT(DM_RSTP, DM_PLY1, DM_PLY2, DM_REC1, DM_REC2, KC_MPLY, KC_MPRV, KC_MNXT, KC_MSTP, BL_TOGG, BL_INC, BL_DEC, BL_OFF, RESET,  KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_AGIN, KC_TRNS, KC_TRNS, RGB_TOG, RGB_M_P, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, UC(0xA7), KC_TRNS, KC_TRNS, KC_TRNS, RGB_SAD, RGB_VAD, RGB_VAI, RGB_SAI, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_UNDO, KC_CUT, KC_COPY, KC_PSTE, KC_TRNS, KC_TRNS, KC_TRNS, RGB_HUD, RGB_HUI, KC_TRNS, KC_TRNS, KC_VOLU, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_MPRV, KC_VOLD, KC_MNXT)
+};
+
+#ifdef JOYSTICK_ENABLE
+  joystick_config_t joystick_axes[JOYSTICK_AXES_COUNT] = {
+      [0] = JOYSTICK_AXIS_VIRTUAL
+  };
+
+  int16_t joystick_position = 0;
+#endif
+
+bool encoder_update_user(uint8_t index, bool clockwise) {
+#ifdef JOYSTICK_ENABLE
+    joystick_position += (clockwise ? 1 : -1);
+    if (joystick_position < 0) joystick_position = 0;
+    if (joystick_position > 1000) joystick_position = 1000;
+    if (joystick_position != joystick_status.axes[0]) {
+        joystick_status.axes[0] = joystick_position;
+        joystick_status.status |= JS_UPDATED;
+    }
+#endif
+
+    if (clockwise) {
+        tap_code(KC_WH_R);
+    } else {
+        tap_code(KC_WH_L);
+    }
+    return true;
+}
+
diff --git a/keyboards/gmmk/pro/rev1/iso/keymaps/greaka/rules.mk b/keyboards/gmmk/pro/rev1/iso/keymaps/greaka/rules.mk
new file mode 100644
index 0000000000..c8a33150cf
--- /dev/null
+++ b/keyboards/gmmk/pro/rev1/iso/keymaps/greaka/rules.mk
@@ -0,0 +1,5 @@
+UNICODE_ENABLE = yes
+DYNAMIC_MACRO_ENABLE = yes
+DEBOUNCE_TYPE = asym_eager_defer_pk
+JOYSTICK_ENABLE = yes
+JOYSTICK_DRIVER = digital
-- 
2.37.1
```

</details>

Then build qmk for keyboard `gmmk/pro/rev1/iso` with keymap `greaka`

drashna, 2022-07-29:
Looks like it's looking for the analog code, even though it's not enabled.  

That's ... not supposed to happen.

PR #17844 (2022-07-30): Fix check when compiling HID Joystick with digital driver
## Description

Analog read is not hidden behind a pre-processor, so will try to run even if analog isn't enabled/set as the driver.

## Types of Changes

- [x] Core
- [x] Bugfix

## Issues Fixed or Closed by This PR

* fixes #17825

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk.fm/#/pr_checklist) and have made the appropriate changes.
 [... cut]
closed 2022-07-30 by zvecr