qmk/qmk_firmware #17876: QMK keyboards (Moonlander, Planck) not detected after OS wakeup on OSX Ventura

readywater, 2022-08-02:
Posting this here as advised on [a reddit thread.](https://www.reddit.com/r/ergodox/comments/wdqjbf/moonlander_and_planck_not_working_well_with_new/)

**Impacts**: Intel and M1 Macs with OSX Ventura Beta (1 and 2)

## Description

The ZSA Moonlander and Planck (which are running QMK) won't be registered as keyboards if left plugged into a mac machine with the new OSX Ventura beta installed that has slept. The easy fix is just replug the keyboard. The keyboard WILL trigger a wake from sleep, but keyboard events won't be registered after that. The keyboard is receiving power the whole time and was recently flashed. It works as expected on other computers and on an IOS device.

This doesn't seem to be the case for the standard apple keyboards as an aside (I ran a small test), so wondering if it has something to do with the OSX configuration stage (the left shift identification).

The problem is consistent across two macs: a 2018 mac mini (with a hub and plugged in directly via usb-a) and an m1 mac air (plugged in directly via usb-c)

## System Information

**Keyboard:** ZSA Moonlander, ZSA Planck
**Revision (if applicable):** 
**Operating system:** OSX Ventura Beta 1 and 2
**`qmk doctor` output:**  Wasn't able to generate but will edit if I can get it set up
```
(Paste output here)
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other: 

## Additional Context
Mac Mini and Air are both totally fresh installs

readywater, 2022-08-02:
I should mention: I don't think this is a qmk bug — rather just an FYI on Ventura. My guess is it has something to do with the usb peripheral permission revamp they've implemented.

drashna, 2022-08-08:
It might help to add `#define NO_USB_STARTUP_CHECK` in your config.h file.  This disables some of the startup checks, and may allow the board to work better. But it won't sleep/idle. 

Also, I get this sometimes, even on Big Sur.

paulds-explore, 2022-09-20:
I am using a ymdk/sp64 on an M1 (12.6). `#define NO_USB_STARTUP_CHECK` and `#define USB_SUSPEND_WAKEUP_DELAY 500` are in my config. Wake from sleep works if the PC is woken using the mouse or built-in keyboard/trackpad. However, if you press a key on the sp64 to wake the machine, it seems to lock up - requiring the USB cable to be unplugged and reconnected again before keypresses are registered again.

I think this issue will impact many boards.

KenVanHoeylandt, 2022-10-15:
> Wake from sleep works if the PC is woken using the mouse or built-in keyboard/trackpad. However, if you press a key on the sp64 to wake the machine, it seems to lock up - requiring the USB cable to be unplugged and reconnected again before keypresses are registered again.

I have the same behaviour on an M1 Pro with macOS Monterey (12.6):
When using the touchpad, I can safely wake up the laptop and then use my keyboard.
If I use the keyboard to wake it up, the keyboard hangs and even the reset button has no effect. Only USB re-plugging fixes it.

I didn't have this issue on an Intel i9 MacBook Pro from 2019 with Monterey.

andrebrait, 2022-10-17:
I can work my M1 Pro MacBook with my GMMK Pro just fine, using QMK.
Currently on macOS 12.6.

@KenVanHoeylandt one note, 12.6 introduced some sleep-related bugs, including ones with the USB-C port. If you're using a USB-C monitor, try switching it to the HDMI port on the MacBook itself and see if that works and you can wake the MacBook from the QMK keyboard again.

KenVanHoeylandt, 2022-10-19:
@andrebrait THanks for the tip. My monitor is connected through the HDMI port directly on my laptop. The Sofle keyboard is connected through USB C though.

LSChyi, 2022-11-24:
@paulds-explore If I understand correctly if you define `NO_USB_STARTUP_CHECK`, then `USB_SUSPEND_WAKEUP_DELAY` actually takes no effect, for ChibiOS-based keyboard, the wakeup logic is [here](https://github.com/qmk/qmk_firmware/blob/558ab18611665c3b447c27bac82fac2224eadaef/tmk_core/protocol/chibios/chibios.c#L186), and only if you **not** defining `NO_USB_STARTUP_CHECK`, the keyboard could enter this section, calling the [`restart_usb_driver`](https://github.com/qmk/qmk_firmware/blob/558ab18611665c3b447c27bac82fac2224eadaef/tmk_core/protocol/chibios/chibios.c#L195), then the [`USB_SUSPEND_WAKEUP_DELAY`](https://github.com/qmk/qmk_firmware/blob/558ab18611665c3b447c27bac82fac2224eadaef/tmk_core/protocol/chibios/usb_main.c#L748) could take effects.

Does this issue is the same as #9962?

paulds-explore, 2022-11-24:
For some reason I don’t see this comment on the thread anymore, but to
test it I removed NO_USB_STARTUP_CHECK and only left
USB_SUSPECT_WAKEUP_DELAY. Now, if I attempt to wake from sleep by pressing
a key, it works as expected pressing a key on the left half of the
keyboard, however pressing a key on the right side doesn’t wake it up and
locks it up (the existing behaviour this bug is about), requiring me to
unplug the USB.

benfrain, 2023-09-23:
Same issue with a ZSA Voyager running 13.5.1