qmk/qmk_firmware #17887: [Bug] Flashing the right half of the ergodox infinity does not work

s4ms0n, 2022-08-03:
Previously I was building and flashing my ergodox infinity via:
`util/docker_build.sh ergodox_infinity:samsons_layout:"dfu-util MASTER=right"` for the right half and
`util/docker_build.sh ergodox_infinity:samsons_layout:dfu-util` for the left half.

Since I have changed my OS I am working directly with the qmk command:
`qmk compile -kb input_club/ergodox_infinity -km samsons_layout`
to compile the code.

I can also flash the compiled code to the keyboard via:
```
qmk flash -kb input_club/ergodox_infinity -km samsons_layout -bl dfu-util
qmk flash -kb input_club/ergodox_infinity -km samsons_layout -bl dfu-util -e MASTER=right
qmk flash -kb input_club/ergodox_infinity -km samsons_layout -bl dfu-util-split-right
qmk flash -kb input_club/ergodox_infinity -km samsons_layout -bl dfu-util-split-left
```

I have also used make commands with the same result
```
 make input_club/ergodox_infinity:samsons_layout
 make input_club/ergodox_infinity:samsons_layout:dfu-util-split-left
 make input_club/ergodox_infinity:samsons_layout:dfu-util-split-right
 make input_club/ergodox_infinity:samsons_layout:"dfu-util MASTER=right"
 make input_club/ergodox_infinity:samsons_layout:dfu-util
```

All of the commands mentioned above flash the left side of the keyboard.
Am I missing an option, or using something wrong?

Help would be very appreciated.

## System Information

**Keyboard: Ergodox infinity** 
**Revision (if applicable): 0.17.6 ** 
**Operating system: MacOS 12.4** 
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /Users/andriysamsonyuk/Repositories/qmk_firmware
Ψ Detected macOS 12.4.
Ψ Git branch: samsons_layout
Ψ Repo version: 0.17.6
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest samsons_layout: 2022-08-02 17:48:21 +0200 (e06a9cbf82) -- split config into left and right layouts, since dfu-util split was not working
Ψ - Latest upstream/master: 2022-08-01 06:00:31 -0700 (3b40ded512) -- ryanbaekr rb18 Configurator Data Fixes (#17872)
Ψ - Latest upstream/develop: 2022-08-01 06:04:26 -0700 (b1eac5c671) -- [Keyboard] Overhaul uzu42 (#17868)
Ψ - Common ancestor with upstream/master: 2022-08-01 06:00:31 -0700 (3b40ded512) -- ryanbaekr rb18 Configurator Data Fixes (#17872)
Ψ - Common ancestor with upstream/develop: 2022-08-01 06:00:31 -0700 (3b40ded512) -- ryanbaekr rb18 Configurator Data Fixes (#17872)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2021-12-08 11:37:03 +0000 --  (257302333)
Ψ - lib/chibios-contrib: 2022-04-18 12:52:47 +0200 --  (2a6b73ff)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2021-07-31 15:44:44 +1000 --  (19a5d533f)
Ψ - lib/printf: 2020-01-01 18:16:48 +0100 --  (d3b9846)
Ψ - lib/vusb: 2020-08-16 01:07:21 +0100 --  (bdb53e4)
Ψ QMK is ready to go, but minor problems were found
```

keyboard-magpie, 2022-08-03:
The readme for the infinity should provide some clues- https://github.com/qmk/qmk_firmware/tree/master/keyboards/input_club/ergodox_infinity#ergodox-infinity

You likely need to specify the handedness in your config.h i.e. `#define MASTER_RIGHT` or `#define EE_HANDS` (the latter of which requires the `dfu-util-split-left/right` arguments

s4ms0n, 2022-08-17:
I have tried to follow the Readme before posting this issue here and tried it again now.
I might be doing smth. wrong but I just don't see it :(

daskygit, 2022-10-06:
The legacy code for `MASTER=right` was removed some time ago now, what do you mean by `does not work?`

s4ms0n, 2022-10-06:
> The legacy code for `MASTER=right` was removed some time ago now, what do you mean by `does not work?`
> 
> As in you can flash the right side and it doesn't work or you're unable to flash the right side?

Hi, thanks for the answer.
I can flash only the left side of the keyboard, when I try to flash the right side, it is flashed with the left side layout.
As a workaround, I have mirrored my layout and flashed the right side independently.
This way I have 2 independent keyboards. It is annoying, but I was not able to compile the keyboard using 
https://github.com/qmk/qmk_firmware/tree/master/keyboards/input_club/ergodox_infinity#ergodox-infinity

I have tried to follow the EE_HANDS initialization step, but it seems not to work for me, or I am doing smth wrong.

Any help will be very appreciated.
Cheers

daskygit, 2022-10-06:
EE_HANDS is handled differently on arm based boards, the handedness is only set when the eeprom is initialised. It's likely you need to clear eeprom for dfu-util-split-right to take effect. 

Looking at the default keymap for the ergox there is an EEP_RST button you can access on the bottom left.

Failing that you could switch to develop branch, as EE_HANDS has been changed so dfu-util-split-right will take immediate effect after flashing.

s4ms0n, 2022-10-07:
Hm, I have upgraded to the latest master branch. Now I am getting a different error (see below)
Am I missing smth. in the configuration?

`Ψ Compiling keymap with gmake --jobs=1 input_club/ergodox_infinity:samsons_layout_left:dfu-util-split-left

QMK Firmware 0.18.13
Making input_club/ergodox_infinity with keymap samsons_layout_left and target dfu-util-split-left

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  44652	      0	  44652	   ae6c	input_club_ergodox_infinity_samsons_layout_left.bin

Compiling: keyboards/input_club/ergodox_infinity/ergodox_infinity.c                                 [OK]
Compiling: .build/obj_input_club_ergodox_infinity/src/default_keyboard.c                            [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                      quantum/eeconfig.c: In function 'eeconfig_init_quantum':
quantum/eeconfig.c:63:13: note: #pragma message: Faking EE_HANDS for left hand
 #    pragma message "Faking EE_HANDS for left hand"
             ^~~~~~~
 [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/logging/print.c                                                                  [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/split_common/split_util.c                                                        [OK]
Compiling: quantum/split_common/transport.c                                                         [OK]
Compiling: quantum/split_common/transactions.c                                                      [OK]
Compiling: quantum/main.c                                                                           [OK]
Compiling: lib/printf/src/printf/printf.c                                                           [OK]
Compiling: quantum/mousekey.c                                                                       [OK]
gmake[1]: *** No rule to make target 'platforms/chibios/eeprom_teensy.c', needed by '.build/obj_input_club_ergodox_infinity_samsons_layout_left/eeprom_teensy.o'.  Stop.
Make finished with errors
gmake: *** [Makefile:414: input_club/ergodox_infinity:samsons_layout_left:dfu-util-split-left] Error 1`

s4ms0n, 2022-10-07:
It worked now, my keymap was too old, after some changes and updates it worked. Thank you for your help!

closed 2022-10-07 by s4ms0n