qmk/qmk_firmware #17890: STM32F411 MCUs do not work with QMK

doomducker, 2022-08-03:
I have found that the STM32F411 MCUs do not work with QMK because of a bug in the silicon that causes dfu-util to not be able to flash code meaning qmk cant flash the code with qmk msys or qmk toolbox.

I have talked to Tormod Volden from the dfu util forums in my bug report here https://sourceforge.net/p/dfu-util/tickets/141/ 
they said _"It is a bug in the built-in bootloader on the chip. It would have been nice if ST could document their silicon bugs. The question is if it is possible to make a workaround without corrupting anything for proper devices. "_

I have tested flashing the same bin file on my 3 stm32f401 dev boards and all of them flash 100% fine and all 3 of my stm32f411 dev boards failed in the same way. so it is 100% not just one chip not working its an issue with qmk/dfu-util

**detailed explanation of the issue** 

im building a keyboard with QMK and am using 3 brand new stm32f411CEU6 WeAct studio V3.0 development boards (Freq. 100MHZ ROM:128KB RAM:512KB)
i also have 3 stm32f401CEU6s that i am using as well
and i cannot write to it as i get
`dfuse_download: libusb_control_transfer returned -9 (LIBUSB_ERROR_PIPE)`

about 3 weeks ago i was using an stm32f401 and it was working fine and i didn't have any issues i haven't updated anything since then i have changed nothing in the code other than the appropriate things to change the MCU.
i have tried this on 3 different windows 10 computers on a AMD ryzen 7 3700x, intel i7-1165g7 and an intel i5 4440. all give the same error i believe they are all on slightly diffrent versions of windows and dfu
i have tried uploading code to the stm32f411 on all of the computers i have also made a basic hello world in visual studio code using platformIO with Arduino framework and i have rebuilt the code and played with the code but i allways get the same error messages.
i have tryed using both the USB-C connector on multiple USB port and also the pins 11 & 12 which correspond to the USB D+ & D-

the qmk toolbox output when I try to flash the code is

```
STM32 DFU device connected (WinUSB): STMicroelectronics STM32  BOOTLOADER (0483:DF11:2200)
Attempting to flash, please don't remove device
> dfu-util.exe -a 0 -d 0483:DF11 -s 0x08000000:leave -D "C:\Users\doomducker\qmk_firmware\doomducker_default.bin"
> dfu-util 0.11
> 
> Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
> Copyright 2010-2021 Tormod Volden and Stefan Schmidt
> This program is Free Software and has ABSOLUTELY NO WARRANTY
> Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
> 
> Opening DFU capable USB device...
> Device ID 0483:df11
> Device DFU version 011a
> Claiming USB DFU Interface...
> Setting Alternate Interface #0 ...
> Determining device status...
> DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
> Clearing status
> Determining device status...
> DFU state(2) = dfuIDLE, status(0) = No error condition is present
> DFU mode device DFU version 011a
> Device returned transfer size 2048
> DfuSe interface name: "Internal Flash  "
> Downloading element to address = 0x08000000, size = 69644
> 
> Erase     [                         ]   0%            0 bytes
> Erase     [=                        ]   5%         4096 bytes
> Erase     [==                       ]   8%         6144 bytes
> Erase     [===                      ]  14%        10240 bytes
> Erase     [====                     ]  17%        12288 bytes
> Erase     [=====                    ]  20%        14336 bytes
> Erase     [=====                    ]  23%        16384 bytes
> Erase     [======                   ]  26%        18432 bytes
> Erase     [=======                  ]  29%        20480 bytes
> Erase     [========                 ]  32%        22528 bytes
> Erase     [=========                ]  38%        26624 bytes
> Erase     [==========               ]  41%        28672 bytes
> Erase     [===========              ]  44%        30720 bytes
> Erase     [============             ]  49%        34816 bytes
> Erase     [=============            ]  52%        36864 bytes
> Erase     [==============           ]  58%        40960 bytes
> Erase     [===============          ]  61%        43008 bytes
> Erase     [================         ]  64%        45056 bytes
> Erase     [=================        ]  70%        49152 bytes
> Erase     [==================       ]  73%        51200 bytes
> Erase     [===================      ]  76%        53248 bytes
> Erase     [====================     ]  82%        57344 bytes
> Erase     [=====================    ]  85%        59392 bytes
> Erase     [======================   ]  88%        61440 bytes
> Erase     [=======================  ]  94%        65536 bytes
> Erase     [======================== ]  97%        67584 bytes
> Erase     [=========================] 100%        69644 bytes
> Erase    done.
> 
> Download  [                         ]   0%            0 bytes
> Download  [                         ]   2%         2048 bytes
> Download  [=                        ]   5%         4096 bytes
> Download  [==                       ]   8%         6144 bytes
> Download  [===                      ]  14%        10240 bytes
> Download  [====                     ]  17%        12288 bytes
> Download  [=====                    ]  20%        14336 bytes
> Download  [======                   ]  26%        18432 bytes
> Download  [=======                  ]  29%        20480 bytes
> Download  [========                 ]  32%        22528 bytes
> Download  [=========                ]  38%        26624 bytes
> Download  [==========               ]  41%        28672 bytes
> Warning: Invalid DFU suffix signature
> Download  [===========              ]  44%        30720 bytes
> A valid DFU suffix will be required in a future dfu-util release
> dfuse_download: libusb_control_transfer returned -9 (LIBUSB_ERROR_PIPE)
> Error during download
Flash complete
```
this does this every time but the final download percentage ranges from 0% to ~75% most of the time it stops around 40%-45% but basically every time the total bytes downloaded changes so it's not that it getting stuck in the .bin file at the same place every time

MICRO CONTROLLER that is the problem
stm32f411CEU6 WeAct studio V3.0 development board

is it something you can find a fix for? or is it up to the people doing dfu util? either way, I hope you can help! :)

also you should remove it from this https://github.com/qmk/qmk_firmware/blob/master/docs/compatible_microcontrollers.md untill you resolve this  so people dont go and buy it for a keyboard

thanks 

[logfile.txt](https://github.com/qmk/qmk_firmware/files/9249963/logfile.txt)

tzarc, 2022-08-03:
STM32F411CEU6 absolutely works fine with QMK.

There are a fair number of boards that target them already -- the only known issues are that there were bad manufacturing batches that were potentially sold on the grey market that ended up in people's hands due to them being cheap.

The STM32F4's are known to have potential issues with DFU when running with a higher-speed HSE -- ST's AppNote AN2606 states that a lower speed HSE should be used such as 8MHz -- however the WeAct Blackpill boards use a 25MHz crystal.

At this stage it's likely not a problem with QMK or dfu-util -- there are a fair few issues with the Blackpill boards such that we had to create a documentation page to deal with people designing boards and incorrectly assuming that all pins and functions are compatible for general use -- see [our Blackpill docs](https://docs.qmk.fm/#/platformdev_blackpill_f411).

closed 2022-08-03 by tzarc

zvecr, 2022-08-03:
Closing as we would not be responsible for any dfu-util fix, and with plenty of working F411 examples, not planning to remove it from docs. 

If theres a fix available, do let us know as there are plenty of us who would be available for testing.