qmk/qmk_firmware #17895: ISO-variant support for Ducky One 2 SF

ollipasanen, 2022-08-03:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

#17260 added support for the keyboard, but only for the ANSI variant.
I would like to use this firmware on my ISO (Nordic)-variant, but I am not sure what this would require.

1) Is anyone planning on implementing this anytime soon? @f7urry @fauxpark 
2) What modifications need to be made to get it to work?

I could try tinkering with it on my own but I am unsure whether the current firmware will even function on my variant and don't want to brick it.

keyboard-magpie, 2022-08-03:
Implementing it will be in the hands of volunteers, as i presume f7urry was when they did the initial commit. It takes more volunteers willing to experiment who actually have the hardware on hand to make progress, I'm afraid.

ollipasanen, 2022-08-03:
That's what I thought. I am willing to help try to speed up the adaptation with my hardware but I've got little experience with developing QMK.

ollipasanen, 2022-08-04:
I compiled the ANSI firmware and flashed it onto my Ducky One 2 SF. Surprisingly, it worked out of the box 🥳. The only thing that failed was the nuvoton-isp flashing software so I had to use the other suggested one (nu-isp-cli).

Using the default keymap, all buttons seem to work as expected (even the nordic ÅÄÖ).

For others coming across this, I've attached the keymap I configured for mine below. It's a rough copy of the one from Vortex POK3R:

TAB = FN (Only 1 layer)
TAB + IJKL = Arrow keys
TAB + Number row = FN row
TAB + U = PGUP
TAB + O = PGDWN
TAB + P = PRTSCR
TAB + Y = CALC
TAB + H = HOME
TAB + N = END
TAB + ^ = PAUSE
TAB + Ö = INS

[ducky_one2sf_1967st_layout_nordic.json.txt](https://github.com/qmk/qmk_firmware/files/9261913/ducky_one2sf_1967st_layout_nordic.json.txt)

closed 2022-12-04 by github-actions[bot]