qmk/qmk_firmware #17922: [Bug] Mousekeys in Constant Speed Mode only move one step per keypress

navidJadid, 2022-08-06:
I have a problem when using the constant speed mode with mouse keys in that when I press and hold one of the mouse direction buttons the cursor only moves one step and stops afterwards. Other mouse modes work without issue though. Other mouse modes work without issue.

For the setup I just followed the instructions on the [docs page for mouse keys](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_mouse_keys.md).

I use a Lily58 rev1 on Ubuntu 20.04., but the issue also persists when I connect the keyboard to my Laptop which runs MacOs.

qmk version: 1.1.0

drashna, 2022-08-08:
could you add the output of qmk doctor?

navidJadid, 2022-08-10:
Sure! Here you go:

```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: <obscured>/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp <obscured>/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp <obscured>/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32' boards. Run 'sudo cp <obscured>/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp <obscured>/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp <obscured>/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'massdrop' boards. Run 'sudo cp <obscured>/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp <obscured>/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp <obscured>/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.17.0
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```