qmk/qmk_firmware #17937: Can't the knob switch layers? For example setting TO(1) output is N TO(2) output is O. The buttons are used to switch layers normally, the knobs cannot

jeromeli002, 2022-08-06:
Can't the knob switch layers? For example setting TO(1) output is N TO(2) output is O. The buttons are used to switch layers normally, the knobs cannot

waffle87, 2022-08-07:
You want the rotation to change layers? You either need to use [encoder map](https://docs.qmk.fm/#/feature_encoders?id=encoder-map) where you're able to use keycodes like `TO(2)`. Or if you're using `encoder_update_user()`, you need to call the applicable [layer function](https://docs.qmk.fm/#/feature_layers?id=functions).

jeromeli002, 2022-08-07:
> 你想让旋转改变图层吗？[您要么](https://docs.qmk.fm/#/feature_encoders?id=encoder-map)需要在可以使用`TO(2)`. 或者如果你正在使用`encoder_update_user()`，你需要调用适用的[图层函数](https://docs.qmk.fm/#/feature_layers?id=functions)。

if (clockwise) {
tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 1, 0));
} else {
tap_code16(dynamic_keymap_get_keycode(biton32(layer_state), 1, 1));
}

In this way, you can customize via mapping, you cannot switch layers, and some key combinations cannot be implemented.

zvecr, 2022-08-07:
That is the legacy method of doing things. As waffle mentions, [encoder map](https://docs.qmk.fm/#/feature_encoders?id=encoder-map) is the way to configure support.

<https://github.com/qmk/qmk_firmware/pull/17734>

drashna, 2022-08-08:
`tap_code16(dynamic_keymap_get_keycode` isn't the right way to do this anyways.  tap_code only supports basic and modded keycodes.  It does not support layer keycodes or any feature codes, etc. 

But encoder map is the correct way to do this, and supports dynamic keymap

jeromeli002, 2022-08-08:
> `tap_code16(dynamic_keymap_get_keycode` isn't the right way to do this anyways. tap_code only supports basic and modded keycodes. It does not support layer keycodes or any feature codes, etc.
> 
> But encoder map is the correct way to do this, and supports dynamic keymap

Is there an example reference that can accurately implement all keys and key combination mappings? It supports via/vial customization, and now some key combinations cannot be implemented.

zvecr, 2022-08-08:
Example has already been provided,

> That is the legacy method of doing things. As waffle mentions, [encoder map](https://docs.qmk.fm/#/feature_encoders?id=encoder-map) is the way to configure support
> 
> https://github.com/qmk/qmk_firmware/pull/17734

For examples on currently on the `develop` branch:
```sh
$ rg ENCODER_MAP keyboards | grep keymaps/via
keyboards/protozoa/p01/keymaps/via/keymap.c:#if defined(ENCODER_MAP_ENABLE)
keyboards/protozoa/p01/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes 
keyboards/mechwild/mercutio/keymaps/via/keymap.c:#ifdef ENCODER_MAP_ENABLE
keyboards/mechwild/mercutio/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes
keyboards/mechwild/mokulua/standard/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes
keyboards/mechwild/mokulua/standard/keymaps/via/keymap.c:#ifdef ENCODER_MAP_ENABLE
keyboards/mechwild/mokulua/mirrored/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes
keyboards/mechwild/mokulua/mirrored/keymaps/via/keymap.c:#ifdef ENCODER_MAP_ENABLE
keyboards/mechwild/obe/keymaps/via/keymap.c:#ifdef ENCODER_MAP_ENABLE
keyboards/mechwild/obe/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes
keyboards/mechwild/waka60/keymaps/via/keymap.c:#ifdef ENCODER_MAP_ENABLE
keyboards/mechwild/waka60/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes
keyboards/mechwild/murphpad/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes
keyboards/mechwild/murphpad/keymaps/via/keymap.c:#ifdef ENCODER_MAP_ENABLE
keyboards/work_louder/micro/keymaps/via/keymap.c:#if defined(ENCODER_MAP_ENABLE)
keyboards/work_louder/micro/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes
keyboards/star75/keymaps/via/keymap.c:#if defined(ENCODER_MAP_ENABLE)
keyboards/star75/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes
keyboards/adafruit/macropad/keymaps/via/rules.mk:ENCODER_MAP_ENABLE = yes
keyboards/adafruit/macropad/keymaps/via/keymap.c:#ifdef ENCODER_MAP_ENABLE
```

closed 2022-08-08 by zvecr